openscad/openscad #3544: Exporting STL triggers a coredump ( filedialog could not be displayed )

jpagliaccio, 2020-12-17:
On CentOS 7 exporting to a STL causes a core dump. Running as root (sudo) the file launcher is displayed and the STL exports correctly. This is the output when running as a normal user:  

```
kfilemodule(5144): couldn't create slave: "Unable to create io-slave:
klauncher said: Unknown protocol ''.
" 
kfilemodule(5144): couldn't create slave: "Unable to create io-slave:
klauncher said: Unknown protocol ''.
" 
Cannot mix incompatible Qt library (version 0x40807) with this library (version 0x40802)
kioslave: ####### CRASH ###### protocol = trash pid = 5157 signal = 6
Cannot mix incompatible Qt library (version 0x40807) with this library (version 0x40802)
Aborted (core dumped)

```
The file explorer does not open like issue #3523 but the cause appears to be different. 
Keep up the great work. Let me know if you want me to test.

thehans, 2020-12-17:
Can you paste the contents of Help -> Library Info please?

jpagliaccio, 2020-12-18:
**WHEN RUNNING AS A NORMAL USER (jpagliaccio) 
WHEN RUNNING AS ROOT IS BELOW 
Note: all of the ownership and permissions in /home/jpagliaccio/.local look okay.** 

OpenSCAD Version: 2014.03
Compiler, build date: GCC "4.7.2", Mar  9 2014
Boost version: 1_53
Eigen version: 3.1.1
CGAL version, kernels: 4.0.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.2
MingW build: No
GLib version: 2.38.2
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/jpagliaccio/.local/share/OpenSCAD/libraries
  /opt/openscad-2014.03/lib/openscad/../../libraries
GLEW version: 1.9.0
OpenGL Version: 1.4 (3.3 (Compatibility Profile) Mesa 18.3.4)
GL Renderer: NVA8
GL Vendor: nouveau
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: no
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: no

Using QGLWidget

GL Extensions:
GL_APPLE_packed_pixels
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_shadow
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_transpose_matrix
GL_ARB_vertex_program
GL_ARB_window_pos
GL_ATIX_texture_env_combine3
GL_ATI_draw_buffers
GL_ATI_texture_env_combine3
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_copy_texture
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_object
GL_EXT_multi_draw_arrays
GL_EXT_packed_pixels
GL_EXT_point_parameters
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_vertex_array
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_MESA_pack_invert
GL_NV_blend_square
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

**WHEN RUNNING AS ROOT** 

OpenSCAD Version: 2014.03
Compiler, build date: GCC "4.7.2", Mar  9 2014
Boost version: 1_53
Eigen version: 3.1.1
CGAL version, kernels: 4.0.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.3.2
Qt version: 4.8.2
MingW build: No
GLib version: 2.38.2
OPENSCADPATH: <not set>
OpenSCAD library path:
  /root/.local/share/OpenSCAD/libraries
  /opt/openscad-2014.03/lib/openscad/../../libraries
GLEW version: 1.9.0
OpenGL Version: 1.4 (3.3 (Compatibility Profile) Mesa 18.3.4)
GL Renderer: NVA8
GL Vendor: nouveau
RGBA(8880), depth(24), stencil(8)
GL_ARB_framebuffer_object: no
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: no

Using QGLWidget

GL Extensions:
GL_APPLE_packed_pixels
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_shadow
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_transpose_matrix
GL_ARB_vertex_program
GL_ARB_window_pos
GL_ATIX_texture_env_combine3
GL_ATI_draw_buffers
GL_ATI_texture_env_combine3
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_copy_texture
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_object
GL_EXT_multi_draw_arrays
GL_EXT_packed_pixels
GL_EXT_point_parameters
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_vertex_array
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_MESA_pack_invert
GL_NV_blend_square
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

jpagliaccio, 2020-12-18:
Ps: I tried on 2 other CentOS 7 workstations and I get the same crash when running as a normal user as soon as the file dialog tries to open (eg: export or save as).

t-paul, 2020-12-18:
Looks like the package does not match the CentOS environment. The library info shows for OpenSCAD that it was build with:
```
 Qt version: 4.8.2
```
So that error message would indicate that version 4.8.7 is installed and is not compatible.
```
Cannot mix incompatible Qt library (version 0x40807) with this library (version 0x40802)
```
I see only 2 options:
1) Raise the issue with the owner of the CentOS package, maybe it can be rebuilt to match the Qt version used now
2) Check if one of the options from the [download page](https://www.openscad.org/downloads.html#linux) works (AppImage, Flatpak, Snap). This would also update to a more recent release.

thehans, 2020-12-18:
Oh wow, 2014 OpenSCAD is quite ancient at this point.

t-paul, 2020-12-22:
Closing as this needs to be solved on distro side.

closed 2020-12-22 by t-paul