openscad/openscad #3547: Hull bug in RC3

revarbat, 2020-12-24:
I’ve found a bug that appears in OpenSCAD RC3 and 2020.12.21.

The code below incorrectly previews AND renders a hull.  The code looks silly with all sorts of unecessary `translate([0,0,0])`s and `mirror()`s, overridden `$fn=4` and `convexity=` args, but if you remove or change any one of those, then the hull will be suddenly preview and render correctly.

```openscad
hull() {
  translate([-4,-1,1]) cube(2, center=true);
  translate([3,0,0]) {
    intersection() {
      rotate([90,0,0]) {
        linear_extrude(height=4, center=true) {
          circle(r=2, $fn=4);
        }
      }
      translate([0,0,0]) {
        translate([0,0,0]) {
          mirror([0,0,1]) {
            translate([0,0,0]) {
              linear_extrude(height=4, center=true, convexity=2, $fn=4) {
                circle(r=2, $fn=4);
              }
            }
          }
        }
      }
    }
  }
}
```

The correct hulling, as shown by OpenSCAD 2019.05 is:
[image]

The INCORRECT hulling that RC3 exhibits is:
[image]

revarbat, 2020-12-24:
This code posted was pared down from the CSG tree of a shape in the BOSL2 library that ran afoul of this bug.

nophead, 2020-12-24:
It looks correct for me with RC3 on Windows.

[image: image.png]

revarbat, 2020-12-24:
If it helps any, I'm using OSX OpenSCAD 2020.12.21.  The bug was found by someone running Windows OpenSCAD 2020.12-RC3.  The original code that exhibited the behavior was using the BOSL2 library (github.com/revarbat/BOSL2):

```openscad

revarbat, 2020-12-24:
The full untrimmed CSG tree is:
```openscad
group() {
  group() {
    group() {
      group();
      group();
      group();
    }
    group();
    group() {
      multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
        group() {
          group() {
            group() {
              group() {
                group() {
                  hull() {
                    group() {
                      multmatrix([[1, 0, 0, -3], [0, 1, 0, -3], [0, 0, 1, 3], [0, 0, 0, 1]]) {
                        group() {
                          multmatrix([[1, 0, 0, -1], [0, 1, 0, -1], [0, 0, 1, 1], [0, 0, 0, 1]]) {
                            group() {
                              group() {
                                multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
                                  group() {
                                    group() {
                                      group() {
                                        linear_extrude(height = 2, center = true, convexity = 2, scale = [1, 1], $fn = 4, $fa = 12, $fs = 2) {
                                          square(size = [2, 2], center = true);
                                        }
                                      }
                                    }
                                  }
                                  group();
                                }
                              }
                            }
                          }
                        }
                      }
                    }
                    group() {
                      multmatrix([[1, 0, 0, 3], [0, 1, 0, -3], [0, 0, 1, 3], [0, 0, 0, 1]]) {
                        group() {
                          group() {
                            group() {
                              group() {
                                intersection() {
                                  group() {
                                    group() {
                                      multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
                                        group() {
                                          group() {
                                            group() {
                                              group() {
                                                multmatrix([[1, 0, 0, 0], [0, 0, 1, 0], [0, -1, 0, 0], [0, 0, 0, 1]]) {
                                                  group() {
                                                    group() {
                                                      group() {
                                                        group() {
                                                          group() {
                                                            group() {
                                                              group() {
                                                                group() {
                                                                  multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
                                                                    group() {
                                                                      group() {
                                                                        group() {
                                                                          group() {
[... 82 lines cut ...]
                                            group();
                                          }
                                        }
                                      }
                                    }
                                  }
                                }
                              }
                            }
                          }
                        }
                      }
                    }
                  }
                }
              }
            }
          }
        }
        group();
      }
    }
  }
}
```

revarbat, 2020-12-24:
That BOSL2 cuboid code, in OpenSCAD 2019.05, results in this:
[image]

But in 2020.12.21 and RC3, results in this:
[image]

nophead, 2020-12-24:
Is it due to the change in semantics of intersection with lazy union?

revarbat, 2020-12-24:
I do not have Lazy Unions enabled.
[image]

nophead, 2020-12-24:
I don't even have a Features tab on my version of RC3 but your example
works for me.

adrianVmariano, 2020-12-24:
I encountered this bug under Linux.  Doesn't seem to affect the Windows version.  

Note that the RC versions are not supposed to have that features tab and lazy union is supposed to be disabled in those versions.  In RC2 there was a bug where you could inherit setting from a dev version you had run.

nophead, 2020-12-24:
Here is something that is different between RC3 and the last release.

difference() {
    if(1)
        echo("hello");
    cube();
    sphere();
}

With RC3 the if() echo() produces an empty group() regardless of the if
being true or false and it becomes the first argument of difference().

difference() {

group();

cube(size = [1, 1, 1], center = false);

sphere($fn = 0, $fa = 12, $fs = 2, r = 1);

}

With the last release  the if() echo() produces an empty group() when false
and a nested one when true but both are ignored by the difference().

difference() {

group() {

group();

}

cube(size = [1, 1, 1], center = false);

sphere($fn = 0, $fa = 12, $fs = 2, r = 1);

}

So I expect the problem is caused by empty groups in the intersection.

thehans, 2020-12-25:
I was not able to reproduce this issue using the original posted scirpt, on Linux Mint 20, running RC3 appimage, nor with the nightly .deb install.

But testing on the csg which was pasted, I can reproduce a difference.  I simplified it down as much as I could (display AST helped a lot to put nodes with single child inline), and got down to this:
```
hull() {
  intersection() {
    multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
      multmatrix([[1, 0, 0, 0], [0, 0, 1, 0], [0, -1, 0, 0], [0, 0, 0, 1]]) {
        multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, -1, 0], [0, 0, 0, 1]]) {
          linear_extrude(height = 4, center = true, convexity = 2) circle($fn = 4, r = 2);
          group();
        }
        group();
      }
    }
    multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
      multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, -1, 0], [0, 0, 0, 1]]) {
        linear_extrude(height = 4, center = true, convexity = 2) circle($fn = 4, r = 2);
        group();
      }
      group();
    }
  }
}
```
Removing any of the remaining empty group nodes fixes it (flush caches between any edits).

Not sure exactly what's happening, but it seems that previously cached nodes are being re-used without proper extra transforms applied.

adrianVmariano, 2020-12-25:
For the record I encountered the problem in the appimage RC3 under Ubuntu.

nophead, 2020-12-25:
Why does one need to flush the cache when changing code?

I had the same issue with if() echo() adding an empty group, breaking my
difference. I had to flush the cache after fixing it to  get the correct
result.

Seems like empty groups were ignored in the release but now are not, and
even get remembered. So even the ghost of an empty group will break
difference() and intersection().

nophead, 2020-12-25:
Perhaps empty groups affect geometry now but aren't included in the cache
signature.

I think the distinction between no geometry and empty geometry is now
broken.

If echo() produces no geometry, why does if(true) echo() produce empty
geometry?

nophead, 2020-12-25:
Also echo() wrapped in a module produces an empty group.

I think there needs to be a clear distinction between no geometry and empty
geometry and if(false) should return no geometry, as it used to do, and
if(true) should return its child. If the child returns no geometry, like
echo, or an empty module, then it should return no geometry, not empty
geometry. If one wants to return empty geometry it should something like
union();

So

if(something) cube(); else union(); // To return cube or empty geometry.
For the first argument to difference();
if(something) cube();  // To return cube or no geometry.  For arguments of
intersection.

adrianVmariano, 2020-12-25:
I agree with you that if (false) ... should return no geometry.  This bug report is not the right place to talk about that if you want something to happen, though.  Maybe you should file a separate bug report, highlighting the change from 2019 version, as in "it broke my old code".   And/or (re)start the conversation on the forum about it.  In lists if (false) returns nothing, not an empty list.

nophead, 2020-12-25:
I think this bug is due to the same basic cause though. Empty groups used
to not count as geometry and now they do.

jordanbrown0, 2020-12-25:
Critically, empty groups (from `echo`, `if(false)`, `for(never)`) **are** geometry for `children()` purposes.

nophead, 2020-12-25:
But even for children it is arguable that if(false) is not a child rather
than an empty geometry child.

adrianVmariano, 2020-12-26:
I think that empty groups from echo are children in RC3 but they are not geometry.   So they get passed as a child.  The echo runs only if the corresponding child is "executed".  But that child doesn't produce a group() in the CSG.  

Why should if(false) be geometry?  If I use if to construct a list like

[1,2,if(false) 3, 4, 5] 

the 3 is simply not on the list.  It seems like if (false) ... shouldn't be a child at all.  If you want a child, add an else.

revarbat, 2020-12-27:
Okay, I went step by step and flushed my cache at every step.  This code will hull incorrectly on a freshly run OS X OpenSCAD 2020.12.21:
```openscad
hull() {
  translate([-3,-3,3]) {
    intersection() {
      rotate([90,0,0]) {
        translate([0,0,0]) {
          mirror([0,0,1]) {
            translate([0,0,0]) {
              linear_extrude(height=4, center=true, convexity=2, $fn=4) {
                circle(r=2, $fn=4);
              }
            }
          }
          group();
        }
        group();
      }
      translate([0,0,0]) {
        mirror([0,0,1]) {
          translate([0,0,0]) {
            linear_extrude(height=4, center=true, convexity=2, $fn=4) {
              circle(r=2, $fn=4);
            }
          }
        }
      }
    }
  }
  translate([4,-4,4]) {
    linear_extrude(height=2, center=true) {
      square([2,2], center=true);
    }
  }
}
```

MichaelAtOz, 2020-12-27:
@revarbat Perhaps a picture of what incorrectly looks like.
Looks normal to me on Debian & Windows.

revarbat, 2020-12-27:
Based on my most recent code to reproduce, this isn't a matter of empty geometry in the `intersect()`.  The empty `group()`s are in `rotate()` and `translate()`.

revarbat, 2020-12-27:
Okay, here's a screen video recording of what I am seeing:

https://user-images.githubusercontent.com/350001/103165918-52a23b00-47d2-11eb-86aa-96b52ad8a192.mp4

thehans, 2020-12-29:
I am still looking into this, and after a `git bisect` it appears that the empty geometry changes from https://github.com/openscad/openscad/commit/ebe46ab9c are where the regression was introduced.  

The RC seems to act as if the second child of the CSG operation is identical to the first, but I haven't been able to figure out how that is happening.

Here is my minimized version of the problem that I've been testing:
```
hull() {
  // intersection here can also be union or difference,
  // which generate different expected geometries,
  // but all correct with 2019 and incorrect with RC
  intersection() { 
    rotate([90,0,0]) {
      translate([0,0,0]) {
        cylinder(r=2, h=4, $fn=4, center=true);
        group();
      }
      group();
    }
    translate([0,0,0]) {
      cylinder(r=2, h=4, $fn=4, center=true);
    }
  }
}
```

**Output Screenshots**
RC:
[image]

2019 release:
[image]

nophead, 2021-01-01:
I have a similar problem where half of an object made with mirror disappears when I remove the last cylinder  from the top of stack.

[image]

[image]

I.e. when remainder is less than zero in this code:
```
module screw_polysink(type, h = 100, alt = false) { //! A countersink hole made from stacked polyholes for printed parts
    head_depth = screw_head_depth(type);
    assert(head_depth, "Not a countersunk screw");
    layers = ceil(head_depth / layer_height);
    rmin = screw_clearance_radius(type);
    sides = sides(rmin);
    lh = layer_height + eps;
    render(convexity = 5)
        for(side = [0, 1]) mirror([0, 0, side]) {
            for(i = [0 : layers - 1])
                translate_z(i * layer_height) {
                    r = screw_polysink_r(type, i * layer_height + layer_height / 2);
                    if(alt)
                        rotate(i % 2 == layers % 2 ? 180 / sides : 0)
                            poly_cylinder(r = r, h = lh, center = false, sides = sides);
                    else
                        poly_cylinder(r = r, h = lh, center = false);
                }

            remainder = h / 2 - layers * layer_height;
            if(remainder > 0)
                translate_z(layers * layer_height)
                    poly_cylinder(r = rmin, h = remainder, center = false);
        }
}
```

It works as expected in the last release. Is this likely to be the same bug?

nophead, 2021-01-01:
If I remove the render() then preview is correct but F6 is always missing the mirror.

thehans, 2021-01-01:
Sounds like the same issue to me.  I just created a PR which should fix this issue.  If you post a full running script I can test on my build and tell you if its also fixed, or you can try downloading one of the CI server's build artifacts from #3568 and see for yourself.

nophead, 2021-01-01:
Yes I used to be able to easily download the artefacts and test PRs. The only hard bit was spelling artefacts but now is seems to be impossible.

nophead, 2021-01-01:
After going round a few circles I manged to download #3568 and it does seem to fix my bug as well.

MichaelAtOz, 2021-01-01:
@nophead I had to subscribe to CircleCI (uses GitHub authentication), then when you click on the 'details' link next to e.g. 'ci/circleci: openscad-mxe-64bit' in the PR you have to 'Login with GitHub' (if you haven't been there recently) 
[with login, it doesn't remember where you wanted to go, shows you everything - easiest to go back to GitHub, click on 'Details' again]
Then click on the 'Artifacts' tab, Bobs your uncle.
Let me know if you want me to email it to you, just this once.

nophead, 2021-01-01:
Yes that is exactly what I did. One has to go around in circles to use CircleCI!

t-paul, 2021-01-02:
Fixed via #3568.

closed 2021-01-02 by t-paul

revarbat, 2021-01-03:
Should this be fixed in MacOS X OpenSCAD nightly 2021.01.02?

Because this: 
[image]

Is getting hulled like:
[image]

While MacOS X OpenSCAD 2019.05 renders it correctly like:
[image]

revarbat, 2021-01-03:
Looking at history, the merged code might just have not made it into nightlies yet.  I'll try again tomorrow.

t-paul, 2021-01-03:
OpenSCAD-2021.01.02.dmg is e94f5b4 which does not include fix #3568.

OpenSCAD-2021.01.03.dmg is a980a3a which is latest master right now.

I've triggered the download scripts so latest snapshots are available via [snapshot download](https://www.openscad.org/downloads.html#snapshots) page.

revarbat, 2021-01-04:
Okay, I concur that this is fixed in MacOS nightlies 2021.01.03.  Thanks!