openscad/openscad #3550: Export to DXF format via command line

dvvord, 2020-12-26:
Hi, I'm having an issue when trying to export open scad 3d object to DXF format via command line.

Here is the command line:
openscad.exe --o example.dxf --autocenter --viewall example.scad
or
openscad.exe --o example.dxf example.scad
Output error message I'm getting: "Current top level object is not a 2D object."

File example.scad contains simple instruction: cube([2,3,4]);

And I can export to DXF via GUI. 
It is the same for Win10 and Ubuntu 18, OpenSCAD version 2019.05.

MichaelAtOz, 2020-12-26:
A cube() is a 3D object.

If I try to export a cube() to DXF via GUI I get `Current top level object is not a 2D object. location unknown`.

thehans, 2020-12-26:
OpenSCAD only supports 2D DXF export.

If you want 2D export you can convert 3D to 2D using projection.
If you want 3D export, you'll have to use another file format.

MichaelAtOz, 2020-12-26:
That is with 2020.12-RC3, 2019.05 gets 'UI-ERROR: Current top level object is not a 2D object.'

dvvord, 2020-12-26:
@thehans  but I can export 3D object to dxf via GUI, why it can't be done via CLI ?

MichaelAtOz, 2020-12-26:
@dvvord as I said I get errors when trying to export cube() to DXF in GUI.
Show what code you are using in the GUI that exports.

dvvord, 2020-12-26:
@MichaelAtOz @thehans my bad, both of you right, it seems I was using other tool to convert one of scada formats to 3d dxf. (returning to old project and missed this part).

closed 2020-12-26 by dvvord