openscad/openscad #3552: Circle/cylinder inner/middle/outer - again

fancsali, 2020-12-26:
I am pretty new to the world of 3D printing, and by pure chance, I stumbled upon this problem **before** actually printing something and scratching my head and pulling my hair for days on end.

I am, of course, talking about the issue of the approximation of circles and cylinders as described in PR #109 and PR #599.

I know, I am sort of necro-bumping here, but I reckon, if the issue is recurring with pretty much everyone starting out with OpenSCAD it might be worth addressing. So, from a complete newbies (to 3D, not programming, though) perspective:

- Having this option described in the documentation would not be a source of confusion, but rather great help.
- The language syntax being locked down doesn't mean too much in the world of evolving programming languages IMO; backwards compatibility does: so a meaningful default could solve that issue.
- On the other hand, user space code is always slower than compiled one; so - again coming from a programming background - I'd expect often used features (i.e. boilerplate) being moved into the core code-base.

**NOTE**: Apologies for being a outsider smarta**  here, what I am trying to say, is that from a user perspective the evolution of the language is just as much appreciated as the backwards compatibility; and I think, many of my fellow users are sharing that thought in this case.

rcolyer, 2020-12-26:
This phenomenon is already described in some good detail, along with examples of the mathematical solutions, in the user manual under the "undersized holes" section of cylinder:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#cylinder

Perhaps there are some other places it could be described, or that could reference this section to redirect people, if there are other areas people might be reading where this needs to be considered.  The manual is open for editing as a wiki, and it would be low-hanging fruit for anyone with ideas on how to improve this description to include a bit more detail.

On the linked PRs, it seems that was by consensus considered not the right solution to the issue.  The proposed syntax was pretty non-standard for OpenSCAD, and the $ variables come with complications that don't seem appropriate for the issue of how to define a radius for a specific circular object.

I had to deal with this issue in some detail while creating my threading library https://www.thingiverse.com/thing:1686322 and attempting to make holes correctly sized, threads correctly sized to fit inside of each other, and using cylinder to make properly sized hex bolts heads.  My conclusion is that for fixed polygon objects like hex bolt heads, one needs to be very attentive to the issue, but at this point it is a geometry problem like many other in OpenSCAD, and one can easily insert the needed corrections (which can be derived or grabbed from the manual).  For smooth round holes (such as with proper settings for $fa and $fs in place), the reason printed parts don't fit is actually not due to this issue at all, but due more to the printing issues with plastics adhesion, printing jitter, and shrinkage properties.  To handle this effectively it is absolutely necessary to treat it in user space with correction tolerance factors.  Perhaps there is merit to describing this problem as a side note in the manual as well to be helpful, but it is notably less an OpenSCAD issue and more of a general 3D printing design issue.

jordanbrown0, 2020-12-26:
+1

I have to apply this correction on a regular basis.  (Probably the most common case is to land one side of an octagonal cylinder on the print bed.)

I can do the trig, and I have done the trig, and it's easy trig.  I have an adjustment function in one of my libraries.  Yet I think this would be an improvement.

Even with the adjustment function, I don't use it *quite* often enough to remember the sense of the function.  Does it tell me the length of the apothem of a polygon with a specified radius?  Does it tell me the radius I need to specify to get a particular apothem?  Does it tell me the ratio for a particular number of sides, or does it accept a radius or diameter and tell me the "adjusted" radius/diameter?  It's almost easier to redo the trig every time I need it, than to look up the function.

I dislike the original PR#109/PR#599 constants.  I dislike @t-paul's literal-string variation less.  My favorite is @hairykiwi's percentage answer, as modified by @GilesBathgate to be a fraction rather than a percentage.  The only question in my mind is what the name of the parameter should be, though almost anything would be OK.  (They suggest `circumscribed`, and that's sensible but I didn't immediately remember that it meant.  My own bike-shed contribution would be "inside", so that inside=1 means that 100% of the the resulting figure is inside the theoretical circle, and inside=0 would mean that 0% of the the resulting figure is inside; it is all outside.)

t-paul, 2020-12-27:
I'm not seeing any new information or suggestions, it's just the same discussion with lots of different opinions again. So instead of discussing as very narrow feature, my suggestion would be to look at more general strategies for solving this kind of issue. The first thing coming to mind is full support for customizing built-in module so user space code can really extend behavior of for example `circle()` while still being able to call built-in `circle()`.

Improved documentation and/or a well documented mini-library would be quite welcome, both are out of scope for this issue tracker though.

jordanbrown0, 2020-12-27:
Indeed... looks like PR#599 kind of ran out of steam, rather than being decided one way or the other.

There seem to be quite a few people who are interested, and a few who say "well, you don't really *need* it".  But I don't think there was anybody saying "it's a bad idea", except in the philosophical sense that if you don't *need* it then it shouldn't be in the core.

I have some thoughts on userspace extensibility of core modules, but I'll take them to the mailing list.

nophead, 2020-12-27:
A discussion about circles going round in circles!

In this case you don't need access to the original because it is easy to
make a circle with polygon, sin and cos. And easy to make a cylinder with
linear_extrude and a circle.

t-paul, 2020-12-27:
That's true for maybe 99.9% of the cases, but when thinking about topics like exports that try to retain circle information as much as possible, it would not work anymore to replace with a dedicated polygon. It's a good solution for now, but it would be great to improve on that.

Anyway, closing the issue as there's nothing actionable at this point and this is not the place for open ended discussions.

closed 2020-12-27 by t-paul

fancsali, 2021-01-04:
I beg to differ, but it's your project, not mine. Also I reckon this not being actionable is a fair point.

Just for the record, to make it clear what my intention with this was; in two simple sentences.

- Indeed one can build anything with sin/cos and some lines. But then also, we can build anything out of triangles...
- I think, the point here is, not that this idea was thought of "not needed" or "wrong"; rather than in the previous thread(s) couldn't agree on the how.