openscad/openscad #3557: issue with C-style for loops, 2020 RC3

Parkinbot, 2020-12-28:
Just switched to 2020 RC3 on Windows 64. 
The following code used to work in older versions of OpenSCAD, but doesn't in 2020 RC3

```
echo(sum([1,2,3])); 
function sum(x) = [for(i=0, s = x[0]; i<len(x);i=i+1, s=s+x[i]) if(i==len(x)-1) s][0]; 
```

I just get:

WARNING: undefined operation (number + undefined) in file 3D Designs, line 2 

to me it looks like *i* gets unknown before *s+x[i]* gets evaluated a last time.

nophead, 2020-12-28:
Isn't it simply use of x[i] before checking i < len(x)?

nophead, 2020-12-28:
The more warnings added the harder it is to code in OpenSCAD IMHO.

nophead, 2020-12-28:
This works:
```
echo(sum([1,2,3])); 
function sum(x) = [for(i=0, s = x[len(x)-1]; i<len(x);s=s+x[i], i=i+1 ) if(i==len(x)-1) s][0]; 
```

Parkinbot, 2020-12-28:
It is not the point to find some construct that works. (I replaced it with *function sum(P, n=0) = n<len(P)-1?P[n]+sum(P, n+1):P[n];* )
The point is that this is a typical C-style for loop and RC3 has a changed behavior.

nophead, 2020-12-28:
Yes it used to be fine to make undefined values as along as you didn't use them. RC3 is a pain in lots of respects as I have to make my code more ugly to work around it.

t-paul, 2020-12-28:
There's no behavior change, result is still the same.

Core of the issue is the execution order which is already discussed in #3227.

closed 2020-12-28 by t-paul

Parkinbot, 2020-12-28:
Call me stupid, but I don't understand your "execution order thing" and why you closed this issue. 

Let's look at the call 
```
echo(sum([1,2,3])); 
function sum(x) = [for(i=0, s = x[0]; i<len(x);i=i+1, s=s+x[i]) if(i==len(x)-1) s][0]
```
and say the execution is about to process the third cycle. 
Given *i* equals 1 then *i<len(x)*  holds. With *i=i+1*, *i* gets the value 2. So for what reason *x[i]* can't be evaluated? 
However x[i-1] can be evaluated in a function like this: 

```
function sum(x) = [for(i=1, s = x[0]; i<len(x);i=i+1, s=s+x[i-1]) if(i==len(x)-1) s][0];

t-paul, 2020-12-28:
As the linked issue and also the documentation shows, the code is basically syntactic sugar for:
```
function sum_(x, s, i) =
    i < len(x)
    ? concat([s], let(i = i + 1, s = s + x[i]) sum_(x, s, i))
    : [];
    
function sum(x) = sum_(x, x[0], 0)[len(x) - 1];

echo(sum([1, 2, 3]));
```
which produces the same warning and maybe clarifies what's going on.

Again, please see issue #3227.

nophead, 2020-12-28:
The issue is the new warning. The original code goes past the end of the
array and reads an undefined value but that doesn't matter because it is
never used. Now it generates a warning.

Parkinbot, 2020-12-28:
Got it, but I must say, the situation is really stupid and will make more people mad. I wonder, why the implementation can't avoid the situation (or at least the warning). 

I changed my code into: 

```
echo(sum([1,2,3])); 
function sum(x) = [for(i=0, s = x[0]; i<len(x); i=i+1, s=i<len(x)?s+x[i]:0) if(i==len(x)-1) s][0]; 
```

nophead, 2020-12-28:
Yes it would be nice if undef was only considered a warning when passed to something that needs a defined value.

Note that you could have put `s=i<len(x)?s+x[i]:undef` or `s=i<len(x)?s+x[i]:1/0`, which illustrates the warning is bonkers.

jordanbrown0, 2020-12-28:
This is the eternal debate around warnings.  Is the program doing something harmless, so the warning is noise, or is it actually doing something unexpected so the warning tells the programmer where to look for the bug?  It's not hard to imagine similar cases where this fencepost error would reflect an actual bug.

This implementation also misbehaves with empty lists, because it always tries to access x[0].

The question is "what kinds of operations should complain when presented with undef as an argument?".  The current answer includes arithmetic.

Arithmetic could just soldier on, with the definition that the sum of undef and anything is undef, but then when you finally get to something that insists on an actual value (comparison?  if?  translate?) it might be difficult to track down where the undef came from.  Somebody had a clever idea of having an undef carry along its original source, so that when you finally had to stop it could still say where the undef came from.  That might yield a reasonable compromise, though you still have to be very careful about what operations try to soldier on. "Equals" comparison seems like a tough question.

thehans, 2020-12-28:
> Yes it would be nice if undef was only considered a warning when passed to something that needs a defined value.

It is, and addition needs a defined value.

nophead, 2020-12-28:
I was think of something that has an effect, like cube().

jordanbrown0, 2020-12-28:
What should this do?

```
a = [0,1,2];
b = a[3];
if (b == 1) {
    cube();
} else {
    sphere();
}
```

nophead, 2020-12-28:
Draw a sphere because undef does not equal 1.

jordanbrown0, 2020-12-28:
So the programming error silently results in something that may or may not be what the programmer really wanted.

As you might be able to tell, I'm personally in the "warning" camp.  They're a pain in the neck, but they force me to make my programs be **correct**.  (Well, more correct.)

nophead, 2020-12-28:
They just make me add more and more code to get around them. I often want to have things undefined because I know they won't be used.

Parkinbot, 2020-12-28:
It looks viable to me to get around the issue by using some protector syntax like an undef->0 map. But if you don't know the issue, you'll start swearing at OpenSCAD.  
```
function sum(x) = [for(i=0, s = x[0]; i<len(x); s=s+protect(x[i+1]), i=i+1) if(i==len(x)-1) s][0]; 
function protect(x) = x==undef?0:x;  
```

jordanbrown0, 2020-12-28:
I think it much more likely that people will swear at OpenSCAD for forcing them to write it this way, instead of the much more obvious
```
for (i = 0, s=0; i < len(x); i++) s = s + x[i];
```
It was a pretty convoluted construct **before** OpenSCAD started detecting arithmetic based on out-of-bounds array references.

nophead, 2020-12-29:
Yes I prefer the tail recursive version. Not sure how they compare speed wise.