openscad/openscad #3558: Ctrl-insert (copy) does not work with 2020.12-RC3

tarhytti, 2020-12-28:
I'm using ctrl + insert button to copy selection but this key combination does not work in with rc3 version.

Of course, there's work-a-round by using ctrl-v but habit is hard to change...

t-paul, 2020-12-28:
Keyboard shortcuts should become configurable at some point, there's not going to be support for that in the next release yet though. At this point only `CTRL+C` / `CTRL+V` is supported

MichaelAtOz, 2020-12-29:
ctrl+insert produces this pop-up
[image]
What is it meant to do? It worked as 'Copy' in 2019.05.

t-paul, 2020-12-29:
Insert text template (code snippet).

MichaelAtOz, 2020-12-29:
Is that documented anywhere?

t-paul, 2020-12-29:
I don't see it in https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP so it's probably only in https://github.com/openscad/openscad/issues/3047

nophead, 2020-12-29:
How does this work? I get a popup but I can't find any way to use it. Is it
just a prompt or I am supposed to be able to get it to complete what I
started typing?

MichaelAtOz, 2020-12-29:
> How does this work?

I'm looking...

nophead, 2020-12-29:
Oh, it seems to be right click to code complete. It isn't normal to right click on a context menu in Windows.

t-paul, 2020-12-29:
Ok, can we *please* go back to the actual issue?

Any suggestions for a different keyboard short-cut to use for the templates?

ofmetal, 2020-12-29:
> Any suggestions for a different keyboard short-cut to use for the templates?

How about just Ctrl+T ?

BTW, Ctrl+Ins is not completely gone in RC3. I've found it still works in about window :)

t-paul, 2020-12-29:
`CTRL+T` is already spoken for although for a feature which is probably rarely used. For reference, see https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts

jordanbrown0, 2020-12-29:
And if we were to reassign Ctrl+T, it should go to New Tab (and Ctrl+N to New Window), to match browsers.

t-paul, 2020-12-29:
IntelliJ IDEA uses `ALT+Insert` for "Generate Code". That might work?

ofmetal, 2020-12-29:
> IntelliJ IDEA uses `ALT+Insert` for "Generate Code". That might work?

Sounds good.

t-paul, 2020-12-30:
Windows build from #3562:
- [64-bit installer](https://6549-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.12.29.ci6549-x86-64_change-template-shortcut-Installer.exe)
- [64-bit zip](https://6549-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2020.12.29.ci6549-x86-64_change-template-shortcut.zip)

MichaelAtOz, 2020-12-30:
In my searching it appears that many new keyboards drop the insert key. 

Not a problem for me, but I'm wondering how wide spread this is, particularly Mac.

 

Aside:

Have to be careful with Alt re menu selection, but Alt-Insert can't be a menu shortcut.

Using another modifier with Alt, can stop conflict with menu selection, e.g. Ctrl+Atl+T.

 

 

  _____  

 

<https://github.com/notifications/beacon/AAPBVYIC7EJL5VVBJD3CM4DSXIZXJA5CNFSM4VMNPBMKYY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFTLA6IA.gif> 

 

This email has been checked for viruses by AVG.
https://www.avg.com

commit 1dfc2bfb (2020-12-30): Fix template action shortcut to not clash with Copy&Paste (fixes #3558).
commit d2c594cf (2020-12-30): Merge pull request #3562 from openscad/change-template-shortcut

Fix template action shortcut to not clash with Copy&Paste (fixes #3558).
commit da2bcc1f (2020-12-30): Fix template action shortcut to not clash with Copy&Paste (fixes #3558).
closed 2020-12-30 by t-paul