openscad/openscad #3559: "mesh is not closed" error in latest Linux nightly and crash with latest stable

menny, 2020-12-29:
OS: Linux
Nightly version: `0+git.52eb8ec`
Stable version: `2019.05`

```
$fn=50;

sphere(d=14);
translate([0, 0, 5]) polyhedron(
        points=[[3,0,0],[0, 1.5, 0], [-3, 0, 0], [0, -1.5, 0], [0, 0, 12]],
        faces=[[0, 1, 2, 3], [0, 1, 4], [1, 2, 4], [2, 3, 4], [3, 0, 4]]);
        
```
This will preview correctly:
[image]

but fails to render the polyhedron:
[image]

I get an error `The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.` in the latest nightly and a crash with the latest stable openscad

menny, 2020-12-29:
The error:
[image]

t-paul, 2020-12-29:
The problem is what the message says. The polygon has most faces flipped looking inwards. Use "Thrown together" view (F12) and change the winding order so there's no faces left with pink-ish color.

See also this [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_appearing_with_F5_but_not_F6?).

closed 2020-12-29 by t-paul

menny, 2020-12-29:
thank you for the quick reply! I didn't even realize that could be an issue.

t-paul, 2020-12-29:
Yeah, this is still a bit obscure but we now have the new error window and looking at the screenshot I'm thinking we should be able to link to FAQ entries here in case of such error messages.