openscad/openscad #3563: 2020.12-RC3 Autocomplete activated by decimal point in string

MichaelAtOz, 2020-12-30:
I presume this is not by design.
[image]

When I typed the `.` the pop-up poped-up.
Also type `a="x.` pops-up, or `a="a.` not `a=".`

donc42, 2021-01-03:
I was working on this and all the solutions I came up with are not optimal. I need some guidance on how you would like this to work.

My first thought was to look at each character as added by the Scintalla Editor. I can look for '"' and keep a inStringLiteral toggle to determine if auto complete is allowed. The problem is in a visual editor things can be deleted, block copied and you can use your mouse to move around inside the document. At this point, the toggle could be incorrect. As an enhancement to this method I was thinking every time I got a ';' I would reset the toggle. At least make it start working again for the next line.

Another possibility is that every time the editor wants to present an auto complete, I could scan the current line up to the cursor position and determine if that point was inside a string. I believe this would work except for strings spanning multiple physical lines. Not sure how common those are.

The only 100% accurate way to make this work correctly is to scan the entire document, every time the editor wants to display auto correct, up to the cursor keeping track of strings. Of course even this would fail if the user had unmatched quotes since the parsing/error detection hasn't been done yet. This would be time consuming and I believe worse than the original bug.

I think if it were my product, I would scan the current line and just live with it not quite working for multi line string literals.

What do you think?

Don

t-paul, 2021-01-03:
Yes, I was thinking along those lines aswell. For the specific issue here, detecting a string on the same line seems a good first step.

There's probably not going to be a perfect solution as we can't rely on the text buffer to be syntactically correct. It would be nice to extend the scanning logic to detect more detailed context information. I'm not sure what the best strategy would be, extending the manual parsing logic, use regular expressions, or maybe having a lexer trying to tokenize and use this information.

My idea would be to go first with the suggested PR for this specific "don't trigger in string" issue and create a new one which could over things like "trigger outside string between `=` and `;`" which probably is an expression so we could limit completions to functions and hide modules.

donc42, 2021-01-03:
Great! That is what I did with my last pull request. 

I had thought it could be enhanced at a later time. 

Do you want me to add a new feature suggestion for "trigger outside string between = and ;" or will you?

t-paul, 2021-01-03:
Sure, maybe we can name the new follow-up issue a bit more general as that was just one of the main contexts I thought of. I guess it makes sense to collect a couple of more potential contexts that could be useful before diving into next coding step. Not necessarily to implement all in one go, but to get some understanding what parts of the code structure to look for when deciding what place the cursor is in.

thehans, 2021-01-04:
It seems a bit odd that it triggers on decimal point *at all* (within or without string).
So typing `a.` outside of a string literal also triggers it, but doesn't make much sense in that case.

I don't know much about how autocomplete works, but OpenSCAD language does technically support `.` as a "member access" operator, but only for in a couple special cases (which maybe even undocumented if I remember correctly?):
```
v = [1,2,3]; // vector "members"
echo(v.x);
echo(v.y);
echo(v.z);
r = [4:5:6]; // range "members"
echo(r.begin);
echo(r.step);
echo(r.end);
```
So it could be kinda neat, if possible for decimal point to trigger a separate autocomplete list containing only those member names.  
Kinda neat... but not very crucial, so failing that I'd rather just not trigger on decimal at all.

The other place where I feel autocomplete is more hindrance than help is when typing inside a comment, since this is likely where one is writing human-language words rather than OpenSCAD.  I had mentioned in #3155 originally, but i think that aspect just got overlooked.  By the way the fix for that by the way was to disable "fillup" characters, which I would think would cover the case of decimal point, but for some reason its still triggering.

thehans, 2021-01-04:
OK, looking back at that previous issue, I think I understand.  The "fillup" characters are for committing to an autocomplete entry after the list has been displayed.  Maybe there is a separate configuration for what characters activate it in the first place?

Another thought I had was if syntax highlighting could be useful to determine the autocompletion context in some way?
For example the editor "knows" in some sense when you are typing within a comment based on the color of the text it displays.  
Even an opened comment `/*` with no closing `*/` will be colored appropriately, but maybe less helpful in the case of strings, since I noticed it won't color unless the end quote is present.

donc42, 2021-01-04:
Well, first I have been contributing for a week and have never used Qt or Scintilla (or the QsciScintalla Qt front end).

This is my rough understanding based on my limited experience....the decision to show autocomplete is totally inside the QsciScintilla editor based on some "start characters" Never really spent time to figure those out since I couldn't change them. When the Scintilla editor wants to show autocomplete it calls back to OpenSCAD to fill in the list. By scanning the current context, OpenSCAD could and does somewhat decide what to show or in the case of string literals returns an empty list.

I thought it was strange that a "." showed so many obviously disallowed syntax items. As you pointed out, if it is a vector a list with "xy,z" would be appropriate and if it is a range a list of "begin, step, end" would be good. It would be nice to track more context and target that list better. Guess it is a trade off on usage versus difficulty tracking the syntax on the fly. 

I guess this semantic processor could either be something new or maybe the parser.y could somehow be hooked to do running parsing in the editor as opposed to a block run when the user hits "preview?"

Sounds like a pretty big project...

commit 8bf6470c (2021-01-05): don't show autocomplete in strings. removed extraneous global variable. (fixes #3563)
commit 9a367473 (2021-01-05): Merge pull request #3578 from donc42/ignore-autocomplete-in-strings

Don't show autocomplete in strings (fixes #3563).
commit b7e8a945 (2021-01-05): don't show autocomplete in strings (fixes #3563)
commit 068d1261 (2021-01-06): Don't show autocomplete in strings (fixes #3563).
closed 2021-01-05 by t-paul