openscad/openscad #3565: Integrate source code formatter (linter) into IDE

alexose, 2020-12-31:
There's a wide variety of styles for writing OpenSCAD code:  Tabs, spaces, spaces before parenthesis, brackets, no brackets, etc.  It would be helpful to add a code formatter (a.k.a. linter) into the IDE in order to standardize the way different projects look.

There's already an effort underway here https://github.com/Maxattax97/openscad-format that might be helpful.

t-paul, 2020-12-31:
Marking as feature request, but external calls are not likely to be accepted. Formatting should be based on the internal AST.

alexose, 2021-01-01:
A style guide (ala [AirBnB's javascript style guide](https://github.com/airbnb/javascript)) might be a useful piece of this as well.

alexose, 2023-06-28:
I came here to suggest this feature, but then I realized that I had already suggested it in 2021.  🤦

jordanbrown0, 2023-06-28:
Note that [linters](https://en.wikipedia.org/wiki/Lint_(software)) are usually *not* style checkers.  Instead, they look for subtle potential errors, patterns where the program passes the compiler but might still have errors.  Variables assigned but never used, used but never assigned, constant expressions in conditionals, type mismatches, that kind of thing.

jordanbrown0, 2023-06-28:
Having a general configurable "feed this selection through an external program" mechanism doesn't seem silly.

jordanbrown0, 2023-11-18:
Duplicate of #2011.

mkatychev, 2025-01-28:
Hi all, just wanted to chime in that  I've written an OpenSCAD formatter that is now ready for general use:
https://github.com/tweag/topiary/pull/845

* You can see the [before](https://github.com/tweag/topiary/blob/7afe496065b18c12dff8631e8e6a5e94979e19f4/topiary-cli/tests/samples/input/openscad.scad) and [after](https://github.com/tweag/topiary/blob/7afe496065b18c12dff8631e8e6a5e94979e19f4/topiary-cli/tests/samples/expected/openscad.scad) formatting pass in the pull request linked above.

* More details on setup and how the parser/lexer is implemented:
  https://lists.openscad.org/empathy/thread/Z4PXFLQM53WXRCYUUKKNOMHJFT4CUMEW
  installation should be simplified once the openscad feature is merged into main

* To address https://github.com/openscad/openscad/issues/3565#issuecomment-753021465: 
  though this formatter uses an external AST, the AST/CST itself is explicitly built for OpenSCAD and covers a [myriad of edge cases](https://github.com/mkatychev/tree-sitter-openscad/blob/master/test/corpus/literals.txt)

P.S.

Curious if anyone could chime in on the indentation issue brought up [here](https://github.com/tweag/topiary/pull/845#discussion_r1932656458), OpenSCAD doesn't seem to have a style guide so I went off of what I saw in the wiki book and popular libraries.