openscad/openscad #3566: Example `logo_and_text` Test Failures

thehans, 2020-12-31:
I just ran `ctest -C All` to see if there were any newly fixed or failing tests since last release, and I saw that these tests are now failing:
 - `cgalpngtest_logo_and_text`
 - `opencsgtest_logo_and_text`
 - `throwntogethertest_logo_and_text`

These aren't reported as errors from regular CI tests, since they are omitted from the default test suite, probably due to time constraints.

The console output of these tests include the warning:
```
WARNING: Viewall and autocenter disabled in favor of $vp*
```

This scad file indeed defines its own $vp* variables, which I guess used to be ignored when run from command line, until this behavior was changed somewhat recently?

Should `examples/Basics/logo_and_text.scad` just have those custom view settings deleted (maybe its desired to keep for exporting merch images etc.)?  

Is it still possible to override the `$vp*` variables when they are explicitly written in the scad file like that, so maybe just a custom test command could also solve it?  

It does get clipped terribly when exported to a square aspect ratio, as seen from test report:
[image]

rcolyer, 2020-12-31:
Setting them to undef appears to remove the warning and restore the viewall/autocenter default behavior.  That would repair the test without altering the example.
```
-D'$vpr=undef;$vpt=undef;$vpd=undef'
```

MichaelAtOz, 2021-01-01:
> The console output of these tests include the warning:
>
>``` 
>WARNING: Viewall and autocenter disabled in favor of $vp*
> ```

Looks like those options are default unless --camera is used.
Does the WARNING contribute to test failure? 

> used to be ignored when run from command line, until this behavior was changed somewhat recently?

Yes, #1797 & #3440.

The example looks bad as an example, so should probably be fixed.
```
$vpt = [300, 0, 80];
$vpd = 1600;
```
Corrects the clipping via command line, at least on my systems.

> (maybe its desired to keep for exporting merch images etc.)?

Nope.

But this version matches the colour, spacing & style of the merch & web site.

[logo_and_text.zip](https://github.com/openscad/openscad/files/5758610/logo_and_text.zip)

[image]

thehans, 2021-01-01:
> Does the WARNING contribute to test failure?

I think only if an "echotest" is run on the script, and it looks like there is no echotest for `logo_and_text` even with `ctest -C All`
Even then, it would just require updating the expected .echo results ... assuming the existence of that WARNING is the expected behavior now.

I'll make a PR with your suggested change and update the expected images.

PR #3571 (2021-01-02): Update examples/logo_and_text.scad and expected test results.
Using updated example provided by @MichaelAtOz

Fixes #3566
commit f66bc9ae (2021-01-02): Update examples/logo_and_text.scad and expected test results. (Fixes #3566)
commit 1f4a9f84 (2021-01-03): Update examples/logo_and_text.scad and expected test results. (Fixes #3566)
closed 2021-01-02 by t-paul