openscad/openscad #3567: linear_extrude-scale-zero-tests fail for CGAL render

thehans, 2020-12-31:
I just ran ctest -C All to see if there were any newly fixed or failing tests since last release, and I saw that these tests are now failing:
 - `cgalpngtest_linear_extrude-scale-zero-tests`
 - `csgpngtest_linear_extrude-scale-zero-tests`

These aren't reported as errors from regular CI tests, since they are omitted from the default test suite due to time constraints.

The problem is that many of the cases in this test generate non-manifold geometry when scaled to 0 with linear_extrude.

And due to #3526, this bad geometry is now detected earlier on and returns empty geometry instead of attempting the implicit top level union on them.
As intended by that PR, the test now emits many messages of:
`ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.` for those objects.

I guess CGAL managed to not crash on them before, possibly since none of the extrudes actually intersect with neighbors, but the question remains what to do with these type of zero scaled extrusions.

There was some previous discussion in PR 3351, regarding what to do with these types of issues caused by linear_extrude with scale=0, starting around here: https://github.com/openscad/openscad/pull/3351#issuecomment-644874848 but a conclusion was not reached.

Here's a screenshot clipped from the test report:
[image]

kintel, 2023-11-04:
@pca006132 An interesting observation here is that this test works with fast-csg, but fails with Manifold and CGAL Nef polyhedrons. I don't really expect Manifold to manage non-manifold objects, but it could be an interesting corner case.

pca006132, 2023-11-04:
Not sure about the linear extrude implementation in OpenSCAD, the constructor in manifold will produce a self-intersecting geometry but the result is still a manifold (topologically speaking...). How does Manifold fail with this? Is it throwing an exception, return some weird stuff or segfault? I would expect one of the first two but not segfault.

kintel, 2023-11-04:
I haven't investigated in detail, but the OpenSCAD symptom is this log output, and no resulting geometry:
```
ERROR: [manifold] Input mesh is not closed!
ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold
```

pca006132, 2023-11-04:
ok, this looks fine

kintel, 2024-03-31:
Status as of 2024-03-24: Still broken, also with Manifold.

Manifold issue is #5046

kintel, 2024-05-01:
Works with Manifold with #5096

stefano2734, 2026-09-20:
There are some improvements in this area at CGAL in the last 2 years. 
Perhaps a new test can show this also here.
https://github.com/CGAL/cgal/pulls?q=is%3Apr+state%3Aclosed+polyhedron