openscad/openscad #3570: Timestamps for console messages

t-paul, 2021-01-01:
In some cases it might be useful to have timestamps for console messages. That should be added in a way that does not that screen space for every message line or so. For initial discussion, see #3540.

Also needs some further discussion to cover both command line and GUI usage.

thehans, 2021-01-01:
Another idea idea for timestamping things, as opposed to the absolute date and time, would be to log seconds since program start.  A bit like how `dmesg` under linux writes seconds since boot for every line in the log (at least on my distro).

jordanbrown0, 2026-02-08:
If format is a preferences item, then it would presumably be controlled by a CLI option, and the default would be "none", preserving output compatibility.