openscad/openscad #3577: Closing app when there are unsaved files: After selection "Save All" app stays open

tarhytti, 2021-01-03:
Hard to write proper title for the issue so here comes steps to reproduce situation:

1) open at least two files
2) edit both files (do not save them)
3) close openSCAD application from File menu or with close window -button
4) select "Save All":
[image]

Expected results:
5) all files are saved and openSCAD is closed

Actual results:
5) all files are saved and openSCAD is still open

tombsar, 2021-01-03:
Duplicate of https://github.com/openscad/openscad/issues/3556 ?

t-paul, 2021-01-03:
Yep, I think it's a duplicate, please re-check once RC5 is out (or with snapshot build >= 2021.01.03).

closed 2021-01-03 by t-paul