openscad/openscad #3582: Failed Save - partially completed

MichaelAtOz, 2021-01-04:
This is probably a matter of taste, but current behaviour is unpalatable to me.

With a new file, File/Save to a location you do not have write permission, give it a proper filename in the save dialogue.
Gets a dialogue box with the following (and in console)
```
Failed to open file for writing /usr/share/openscad-nightly/examples/Basics/sphere-chopped.scad (Permission denied)
```
OK so far.

But it then progresses to give the Tab/Title the new filename.
I would expect it to be an atomic operation, if the save fails, the 'untitled.scad' instance stays as it was.

This also happens with Save-As, the editor takes on the new filename, even though it does not exist.
Also the `*` in the Tab/Title, signifying editor has changes, is no longer shown,
although if you close it, you do get the 'file changed' prompt.

This was on Debian w 2020.12.15.

t-paul, 2021-01-04:
Can you verify with the latest snapshot? I believe that is fixed via #3576.

MichaelAtOz, 2021-01-04:
[image]
Nope, not fixed, 2020.01.03-nightly, File/New, File/Save, select un-writable directory, give it a name 'save-bug.scad', write error, but editor takes on new name. And same with Save-as, including loss of `*`.

I'll give it a go on Windows.

MichaelAtOz, 2021-01-05:
I'm not a Windows security permissions specialist, I just setup a directory and selected Deny-write.

On Windows, something checks whether it can be written at the Save dialogue box level.

[image]

Given the dialogue is still open, I assume that something is Windows not OpenSCAD??

t-paul, 2021-01-05:
Aha, the change I had in mind did cover the recent list, not the editor path update.

commit 10b61c12 (2021-01-05): Fix save error handling (fixes #3582).
commit cac04425 (2021-01-05): Merge pull request #3589 from openscad/fix-save-error-handling

Fix save error handling (fixes #3582).
commit ed245300 (2021-01-06): Fix save error handling (fixes #3582).
closed 2021-01-05 by t-paul