openscad/openscad #3583: Inconsistent rounding

GillesBouissac, 2021-01-04:
A file with only these 2 lines of code gives an error:
```
echo("0.57/0.01        = ", 0.57/0.01);
assert(0.57/0.01==57,        "ERROR 0.57/0.01!=57");
```
The output is:
```
ECHO: "0.57/0.01 = ", 57 
ERROR: Assertion '((0.57 / 0.01) == 57)' failed: "ERROR 0.57/0.01!=57" in file rounding.scad, line 2 
```
This looks like a bug to me.

We can also do this which also gives inconsistent results:
```
echo("floor(0.57/0.01) = ", floor(0.57/0.01));
assert(floor(0.57/0.01)==57, "ERROR floor(0.57/0.01)!=57");
```
The output is:
```
ECHO: "floor(0.57/0.01) = ", 56 
ERROR: Assertion '(floor((0.57 / 0.01)) == 57)' failed: "ERROR floor(0.57/0.01)!=57" in file rounding.scad, line 2
```
Btw floor(57) result is 57 as expected.

Environment:
```
openscad --version
OpenSCAD version 2021.01.03
uname -a
Linux X99 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Same results with this version:
```
openscad --version
OpenSCAD version 2019.05
```

nophead, 2021-01-04:
It is just a consequence that 0.1 cannot be represented exactly in binary floating point. It is a recurring number like 1/3 in decimal.
```
Same in Python
Python 2.7.16 (default, Oct 10 2019, 22:02:15)
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 0.57/0.01==57
False
>>> 0.57/0.01-57
-7.105427357601002e-15
>>>
```

GillesBouissac, 2021-01-04:
Ok i see,
I will have to use some tricks (something like round(0.57*1000000)/round(0.01*1000000)) to have the behavior i need.
I close the issue thank you for the fast answer.

closed 2021-01-04 by GillesBouissac

jordanbrown0, 2021-01-05:
Mostly, if you think you want to compare a computed number for equality, you're doing something wrong.

But if you really must, you probably want something like

```
// Round to six decimal places
function round6(x) = round(x*1000*1000)/1000/1000;
x = round6(0.57/0.01);
echo(x, x-57);
assert(x==57);
```