openscad/openscad #3584: Parsing Error when typing in empty source file.

donc42, 2021-01-04:
if you run OpenSCAD and specify an empty file such as testing.scad on the command line and then type any character as the first character, you will get 

ERROR: Parser error: syntax error in file testing.scad, line 2
Execution aborted

if you complete the line with =1; and hit preview it is fine.

See the screen shot after typing "x"

[image]

MichaelAtOz, 2021-01-05:
That didn't happen initially on Windows.
But with Design/Auto-reload&preview enabled, it does.
Somewhat related to #3539. 
I'm presuming a (bug in) preview is hung or looping due to zero data,
typing anything allows the preview to continue.

Obviously the auto-reload&preview shouldn't happen.

But possibly also related to #3582, as in this case the file-load failed, but
the editor context thinks it is, in Don's case, testing.scad*.

If file load fails, it presumably should go to 'untitled.scad'??

donc42, 2021-01-05:
I didn't mention in my original post but testing.scad was a 0 byte file but I originally discovered it on syntactically empty file that just contained two line feeds.

MichaelAtOz, 2021-01-05:
My test was a non-existent file.

MichaelAtOz, 2021-01-05:
> 
> 
> I didn't mention in my original post but testing.scad was a 0 byte file but I originally discovered it on syntactically empty file that just contained two line feeds.

Interestingly, your error (in the screengrab) was `no such file or directory`, I wouldn't expect that for either type you mentioned.

donc42, 2021-01-05:
good catch. I had reproduced it on an empty file (touch test.scad) but that testing.scad was non-existent. And like I said, it originally happened on a two byte file containing line feeds.

donc42, 2021-01-06:
I was working on the issue of the error when you run OpenSCAD with a non-existent filename. I found the place the file is opened in TabManager::refreshDocument(). Problem is the tab name, the file changed, etc have already been set when this is called and this routine like most of yours is "void" and doesn't throw an exception.

How can this routine indicate that it couldn't open a file? I could easily make it a bool and return true on error and then I could handle this error and you would wind up with a untitled.scad file if the one you specify doesn't exist.

Alternately, I could try to open the file myself before calling the refreshDocument(). This would have less impact on the rest of OpenSCAD but seems wasteful to open the file and then call refreshDocument() to open file file (and read it) again.

What is the OpenSCAD standard?

BTW, I am thinking the file with only newlines is probably a separate issue.Anyway, I can't treat it the same way since the file actually exists. Will have to go after that one next.