openscad/openscad #3593: Feature request: Reflect Release candidates in version()

MichaelAtOz, 2021-01-07:
and maybe version_num().

I've just been testing with various RC's, version() returns the same,
e.g. [2020,12,0] for both RC2 & RC3.

At startup it shows in the console `OpenSCAD 2020.12-RC3` but
that is soon lost. Doing `echo(version());` does not provide the info
to distinguish the console output of two RC's.

`version()` could return a fourth item, e.g. `[2020,12,0,3]'

`version_num()` could place the RC number as a decimal, e.g. 20201200.3

thehans, 2021-01-07:
If this is done, I think we should just take care to ensure that a `version_num()` of a final release is always greater than that of a release candidate.

Are we not using the lower two (day of month?) digits?  I know the date based numbering is only "nominal" and not an exact build date, after all we have passed the end of December 2020 of course.

So I'm thinking we should probably bump those least significant whole digits for a final release (Otherwise we end up with e.g. 20201200.0 release < 20201200.5 RC5).

MichaelAtOz, 2021-01-08:
Also need to cope with point releases, e.g.
```
// OpenSCAD version 2015.03-2 
echo(version(),version_num()/100); // ECHO: [2015, 3, 0], 201503
```

Hmmm, this is getting complex...

Thinking cap Engaged...

t-paul, 2021-01-08:
Maybe adding `version_info()` returning a data structure?
```
i = version_info();
echo(i.year); // 2021
echo(i.full_version); // 2021.01-RC5
echo(i.release); // true
```

MichaelAtOz, 2021-02-01:
How about:

`v=version_info();` returns a structure (initially a vector)
```
v is [ 
        version - vector as below      // 0 - same struc as version() - support deprecation in future
          [year, month, day(zero for release & candidate)],
        number,                        // 1 - same as version_num() - ditto
        major,                         // 2 - text, yyyy.mm
        minor,                         // 3 - text, yyyy.mm[.dd | "-"& release | "-RC"& release ]  e.g 2021.01.15, 2015.03-2,  2021.01-RC6
        snapshot,                      // 4 - text, yyyy.mm[.dd | "-"& release | "-RC"& release] ".ci" & ci "(git " & git ")" 
        release,                       // 5 - release number e.g. 6 (even for RC6)
        is_release,                    // 6 - boolean - true for release and release candidate 
        is_candidate,                  // 7 - boolean
        ci,                            // 8 - as per info in snapshot (include for release)
        git                            // 9 - ditto
      ]
```
Structure element names can be added later, as '.version', '.number', '.snapshot' etc.

Up-to 'minor' would be sufficient for now, allowing easy `echo(v.minor); // 2015.03-2, or 2021.01-RC6`.
You can infer/extract 'release' 'is_release' & 'is_candidate' from there.

As per sort-ability (later is always > earlier), using 'minor' will generally sort so, but
RC causes a problem. Depending on assumptions about how candidates are done
on particular year/month, it may not be a problem. I'm now thinking on this.
As this is linked to development scripts etc, this needs wider consideration, 
adding 'candidate' (a number) could help, or is 'git' sufficient? 

Current/past use. 
Type | Shows | In About* | In console | In Lib-info | In Status-bar | version()
----- | ------- | ---------- | ----------- | ----------- | -------------- | ----------- 
Major Release | `OpenSCAD version 2019.05` | [x] | [x] | [x] | [x] | [2019, 5, 0]
Minor Release | `OpenSCAD version 2015.03-2` | [x] | [x] | [x] | No `-2` | [2015, 3, 0]
Candidate | `OpenSCAD version 2021.01-RC6`  | [x] | [x] | [x] | No `-RC6`| [2021, 1, 0]
Snapshot | `OpenSCAD version 2021.01.15.ci6794 (git 7d8cb1e2d)` | [x] | No `(git...)`| [x] | [x] | [2021, 1, 0]

*Title of 'About' Window shows yyyy.mm, except in Snapshot shows full info.

MichaelAtOz, 2021-12-05:
I'm caught up on my reading here & the mailing-list.
I'm still time constrained by IRL issues, so my contribution will still be limited.

I just read the package manager discussions #3479 
I can see that becoming part of the development process.
```
info: 
  id: openscad
  version: 2021.1.0
  summary: OpenSCAD is a 3D CAD program for the creation of 3D models.
```

So sorting out [versioning nomenclature ](https://en.wikipedia.org/wiki/Software_versioning)will become important.
I haven't used enough package managers to know how the version comparison is handled in the weeds, 
but I suspect basic alpha/num sort order is used.
As mentioned above the current Release Candidate usage could be a problem as `RCn` will always be greater than Minor Releases.

Would it be expected that RC's, Nightlies & the dev Snap builds be excluded from package management, or segregated?

p.s. 
In the table above 'No' means 'does not include' the bit that follows. 
The Snap versions have different behaviour, shows `OpenSCAD Version 2021.11.19` & `version()` includes the `DD`