openscad/openscad #3594: Scales not Visable -  Rotate messes up standard directions

bigrednurse, 2021-01-07:
1.  It would sure be nice if someone would fix this problem so that scales can be displayed on top of everything.  What use are they when the model covers them.  You have a menu item to display the scale but does not work unless the part is below the scale.

2.   Whenever you do a rotate, the translate x,y,z gets shifted.  Really folks, why add this level of confusion.  You should not rotate x,y,z in the translate statement x direction should **ALWAYS** be x direction. Why should you have to think this is rotated now x means z.  This is really stupid in MHO.  You should not have to go back and say oh, I made a rotation so now the directions are wrong.       At least add a menu item to keep directions constant 

These two problems should not exit.  The fact that they do is mind boggling.

3,  A new feature I would like would be to be able to take the mouse and grab the side of a panel and drag it across the part
and have it automatically make a cutaway for you.

Can anybody fix these issues

MichaelAtOz, 2021-01-07:
Not any time soon.

For 1. you can place a `%` in front of you model, e.g. `%example006();`, to make it transparent:
[image]

2. is the way it works. Cacading operations affect each step. If it didn't you couldn't actually build anything.

3. That is not how OpenSCAD works. If you want that you need another CAD package.

closed 2021-01-07 by MichaelAtOz

bigrednurse, 2021-01-07:
Thanks for the tip on the % sign. I have no primer for this cad program I'm just using it. 
 But even tho a rotation causes switching of the axis it should not. 
Why should the user have to keep track of the changing axis. Let the computer keep track. 
Then just enter the direction you actually mean. 
Add menu option to keep standard directions constant.

nophead, 2021-01-07:
If you put the translate before the rotate it moves the rotated object
along the world axes. If you put the translate after the rotate it
translates the object before it is rotated along the local axes of the
object. This is just how nested transformations work.