openscad/openscad #3601: Problem?? comment_parser: ERROR IN PARAMETER: Parser error in comments of file

MichaelAtOz, 2021-01-09:
I happened to have `--debug=all` when testing, running through the examples.
I noticed this in examples/parametric/sign.scad.

```
Parsing design (AST generation)...
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
parameterslider: ParameterSlider: [60.00:1.00:200.00, 82.00] - (0 - 140, 22)
parameterslider: ParameterSlider: [1.00:1.00:10.00, 2.00] - (0 - 9, 1)
Compiling design (CSG Tree generation)...
```
The customizer parameters look OK.
Perhaps it is `/*[ group name ]*/` in the top comments?

MichaelAtOz, 2021-01-09:
&
```
Loaded design 'C:/Program Files/openscad-2021.01-RC5/examples/Parametric/candleStand.scad'.
 
Parsing design (AST generation)...
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
parameterslider: ParameterSlider: [3.00:1.00:14.00, 7.00] - (0 - 11, 4)
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Polygon2d-CGAL: Polygon2d::tessellate(): 1 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 1 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 2 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 2 outlines
Geometries in cache: 8593
Geometry cache size in bytes: 21094816
CGAL Polyhedrons in cache: 102
CGAL cache size in bytes: 209769568
Compiling design (CSG Products normalization)...
Normalized tree has 53 elements!
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
Compile and preview finished.
Total rendering time: 0:00:00.195
```

thehans, 2021-01-14:
It looks like it prints that "error" for every customizer parameter that has no comment after it (on the same line).  There are 10 errors and 10 values with no comment.  

For example if you put `// [1:50]` after all the numeric ones, it goes down to one error (only the checkbox is left, for which I don't think there is an appropriate customizer comment).

It seems to be a harmless error since the customizer apparently works fine with that file, but probably better if the parser can be altered so it doesn't count 'no comment' as an error.

ChrisCoxArt, 2022-01-18:
I cannot reproduce these errors in 2021.01 or the current dev build. Looks like it was fixed in the comment parser at some point.

MichaelAtOz, 2022-01-18:
Just now, candlestand.scad, 2021.01, Windows 7/64

```
OpenSCAD 2021.01
https://www.openscad.org/

Copyright (C) 2009-2021 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

QSettingsCached: QSettings::setValue(): view/showAxes = 'true'
QSettingsCached: QSettings::setValue(): view/showScaleProportional = 'true'
QSettingsCached: QSettings::setValue(): view/orthogonalProjection = 'true'
QSettingsCached: QSettings::setValue(): recentFileList = ''
QSettingsCached: QSettings::setValue(): recentFileList = ''
Loaded design 'C:/Program Files/OpenSCAD/examples/Parametric/candleStand.scad'.
 
Parsing design (AST generation)...
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
parameterslider: ParameterSlider: [3.00:1.00:14.00, 7.00] - (0 - 11, 4)
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Polygon2d-CGAL: Polygon2d::tessellate(): 1 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 1 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 2 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 2 outlines
Geometries in cache: 11
Geometry cache size in bytes: 315376
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 53 elements!
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
Compile and preview finished.
Total rendering time: 0:00:00.219
```
I'll try a later dev, but can't run the latest due to #4003.

thehans, 2022-01-18:
Still reproducible on Linux Mint 20.2 with nightly dev build.  @ChrisCoxArt Did you run with `--debug=all` ? 
Also, the messages display in the OpenSCAD's GUI Console, it can be confusing since some debug messages go to stdout, and some to Console.

MichaelAtOz, 2022-01-18:
OpenSCAD 2021.12.14 via snap on Debian.

```
OpenSCAD 2021.12.14 
https://www.openscad.org/
 
Copyright (C) 2009-2021 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
QSettingsCached: QSettings::setValue(): recentFileList = ''
QSettingsCached: QSettings::setValue(): recentFileList = ''
Loaded design '/snap/openscad-nightly/2990/usr/share/openscad-nightly/examples/Parametric/candleStand.scad'.
QSettingsCached: QSettings::setValue(): view/showAxes = 'true'
QSettingsCached: QSettings::setValue(): view/showScaleProportional = 'true'
QSettingsCached: QSettings::setValue(): view/orthogonalProjection = 'true'
 
Parsing design (AST generation)...
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
comment_parser: ERROR IN PARAMETER: Parser error in comments of file 
 
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Polygon2d-CGAL: Polygon2d::tessellate(): 1 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 1 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 2 outlines
Polygon2d-CGAL: Polygon2d::tessellate(): 2 outlines
Geometries in cache: 11
Geometry cache size in bytes: 315376
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 53 elements!
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
Compile and preview finished.
Total rendering time: 0:00:00.022
renderer: setColorScheme
```

MichaelAtOz, 2022-01-18:
I can't test 'latest' functional Windows dev 2021.10.19 mentioned in #4003, due to #3719.

ChrisCoxArt, 2022-01-18:
OK, why can't I reproduce this on MacOS?
And yes, I know where the errors are displayed in the GUI.

ChrisCoxArt, 2022-01-18:
Yay!  I got it to happen. Still not sure why it wasn't happening earlier.

commit 51e17a6f (2022-01-19): don't try to parse empty comments (fixes #3601)
closed 2022-01-19 by thehans