openscad/openscad #3603: -o doesn't exit on windows command line

rawwerks, 2021-01-10:
i'm using openscad in the windows command prompt to output a file using the -o option. the file writes, but openscad does not exit afterwards. 

a sort of workaround is adding ` && Taskkill /IM openscad.exe /F` at the end, but there has to be a better way to exit after writing the output file...

t-paul, 2021-01-10:
I can't reproduce that on Windows 10 with version 2021.01-RC5 using a simple script `test.scad`
```
cube(10);
sphere(10);
```
and running `openscad.com -o test.stl test.scad`.

Please give more details, including OpenSCAD Version, Windows Version and full command line. If possible (and needed) also provide a sample file.

Note that `openscad.exe` will disassociate from the console but still finish processing and exitting after writing the STL.

rawwerks, 2021-01-11:
aha, that is it. i was explicitly calling "openscad.exe" instead of "openscad.com" (or just openscad).

thank you so much!

closed 2021-01-11 by rawwerks