openscad/openscad #3604: Flag that reports whether current file was included / used

jordanbrown0, 2021-01-11:
Problem:  I have a library that needs to be `include`d, because it defines constants for its callers to use.  I want to have test/demo cases for it.  If it was `use`d, I could just include the test/demo cases right there, but with `include` they would get executed.

The idea would be, say, to have `$include` and `$use` that are true if `include`d or `use`d, respectively.

This would allow me to say
```
if (!$include) {
    do_test_demo_cases();
}
```

It would also let you say things like:

```
assert(!$use, "this library is only useful when included, not used");
assert($use || $include, "this is a library, don't try to execute it by itself");
```

MichaelAtOz, 2021-01-11:
```
module pingTest() echo("ping");

*!pingTest(); // remove '*' to test
```

jordanbrown0, 2021-01-11:
Sure.  Or
```
// !pingTest();   // uncomment to test
```
But it would be nice (far, far from critical) to be able to do it "automatically".

jordanbrown0, 2021-01-11:
It would be particularly convenient when flipping back and forth between the library and a consuming program, where you want to make a change, exercise the test case, exercise the real case, make a change, et cetera.

t-paul, 2021-01-12:
This seems like a very low level solution. Is there maybe a way to solve the workflow problem without exposing internal processing details?

At any rate, it can't be a `$` variable as that is dynamically scoped. I think this feature would need lexical scoping to reflect the static code structure.

jordanbrown0, 2025-03-09:
I agree that a `$` variable can't work.

My only idea would be to have the lexer (or parser?) expand a magic token like `__FILE__` into something interesting, but I don't think I want to propose such a mechanism.  And the issue hasn't popped up again for me in several years, so...

closed 2025-03-09 by jordanbrown0