openscad/openscad #3606: Feature request:  Ability to apply transformations to vector lists

tnm23, 2021-01-11:
I would like to request a (hopefully) simple modification to the language to allow geometry transformations (scale, rotate, mirror, etc) to operate on a list of 3D position vectors, as well as the actual geometric primitives they seem currently restricted to, e.g.

[[Xb1,Yb1,Zb1], ..., [Xbn,Ybn,Zbn]] = rotate([x,y,z]) [[Xa1,Ya1,Za1], ..., [Xan,Yan,Zan]]

Can this be added to OpenSCAD without inordinate difficulty?  It would be tremendously useful in writing succinct constraint calculations for parametric models without having to tediously derive elaborate and error-prone trigonometric functions in three dimensions.

Perhaps this could be most elegantly achieved by simply creating non-rendering equivalents to the polygon() and polyhedron() primitives, that have only the [points] or [paths] arguments, respectively, and a utility function to extract these lists of vertices from their transformed versions so that all the rest of the code would simply operate on them as if they were standard geometry?  (perhaps this might later allow for the possibility of using 3D acceleration hardware to solve geometric constraints?)

t-paul, 2021-01-11:
No, exposing all internals directly into the language will make it impossible to change things without breaking compatibility with existing scripts. See #3087 for a discussion of how a feature like this may be done while still unifying the language part visible to the user.

closed 2021-01-11 by t-paul

nophead, 2021-01-11:
Unless I misunderstand the request it is trivial to apply a transform to a list of points in user land using matrix multiplication.
```
function vec3(v) = [v.x, v.y, v.z]; //! Return a 3 vector with the first three elements of `v`
function transform(v, m) = vec3(m * [v.x, v.y, v.z, 1]); //! Apply 4x4 transform to a 3 vector by extending it and cropping it again
function transform_points(path, m) = [for(p = path) transform(p, m)]; //! Apply transform to a path
```

To make the transform matrix just use functions with the familiar names.
```
function translate(v) = let(u = is_list(v) ? len(v) == 2 ? [v.x, v.y, 0] //! Generate a 4x4 translation matrix, `v` can be `[x, y]`, `[x, y, z]` or `z`
                                                         : v
                                           : [0, 0, v])
                          [ [1, 0, 0, u.x],
                            [0, 1, 0, u.y],
                            [0, 0, 1, u.z],
                            [0, 0, 0,   1] ];

function rotate(a, v) = //! Generate a 4x4 rotation matrix, `a` can be a vector of three angles or a single angle around `z`, or around axis `v`
    is_undef(v) ? let(av = is_list(a) ? a : [0, 0, a],
                      cx = cos(av[0]),
                      cy = cos(av[1]),
                      cz = cos(av[2]),
                      sx = sin(av[0]),
                      sy = sin(av[1]),
                      sz = sin(av[2]))
                          [
                              [ cy * cz, sx * sy * cz - cx * sz, cx * sy * cz + sx * sz, 0],
                              [ cy * sz, sx * sy * sz + cx * cz, cx * sy * sz - sx * cz, 0],
                              [-sy,      sx * cy,                cx * cy,                0],
                              [ 0,       0,                      0,                      1]
                          ]
                : let(s = sin(a),
                      c = cos(a),
                      C = 1 - c,
                      m = sqr(v.x) + sqr(v.y) + sqr(v.z), // m used instead of norm to avoid irrational roots as much as possible
                      u = v / sqrt(m))
                          [
                              [ C * v.x * v.x / m + c,       C * v.x * v.y / m - u.z * s, C * v.x * v.z / m + u.y * s, 0],
                              [ C * v.y * v.x / m + u.z * s, C * v.y * v.y / m + c,       C * v.y * v.z / m - u.x * s, 0],
                              [ C * v.z * v.x / m - u.y * s, C * v.z * v.y / m + u.x * s, C * v.z * v.z / m + c,       0],
                              [ 0,                           0,                           0,                           1]
                         ];

function scale(v) = let(s = is_list(v) ? v : [v, v, v]) //!  Generate a 4x4 matrix that scales by `v`, which can be a vector of xyz factors or a scalar to scale all axes equally
                        [
                          [s.x, 0,   0,   0],
                          [0,   s.y, 0,   0],
                          [0,   0,   s.z, 0],
                          [0,   0,   0,   1]
                        ];
```