openscad/openscad #3607: Feature request: Splines and Bezier curves

variostudio, 2021-01-12:
It would be nice to have such functionality as splines and bezier curves.

I'm considering OpenScad for airfoil design and splines is pretty requested feature in this area.

Thanks!

t-paul, 2021-01-12:
I guess this would be an extension to #508. For the time being, make sure to look at existing airfoil libraries like https://www.thingiverse.com/thing:898554

variostudio, 2021-01-12:
Thanks for update.
Please let me know if I can sponsor thus feature for quicker progress on it

Regards,
Dmytro.

t-paul, 2021-01-12:
@variostudio you can try adding a bounty, but looking back that seems to work better for GUI topics and not so well for more complex or language changes which tend to need more time to discuss, design and implement.

jordanbrown0, 2021-01-12:
It seems like what the OP is asking for may be satisfied by libraries that return the coordinates for splines and Bézier curves, to contribute to polygons and polyhedra.  I'm not clear on the relationship between the two, but BOSL2 has support for Bézier curves and surfaces and a Google search for "openscad spline" has a number of relevant-looking hits.

lf94, 2021-03-31:
I wrote my own bezier curve function which takes 4 points: start, control point 1, control point 2, and end; it then generates as many points it needs to satisfy $fn.

```
/home/lee/Code/Mine/3D/prelude/bezier4.scad:1.1,18.61
// P = (1−t)³P1 + 3(1−t)²tP2 +3(1−t)t²P3 + t³P4
function bezier4(points) =
let (s = 1.0 / $fn)
[for(t = 0, i = 0; i < $fn + 1; t = t + s, i = i + 1)
       (pow(1 - t, 3)             * points[0])
+  (3 * pow(1 - t, 2) * pow(t, 1) * points[1])
+  (3 * pow(1 - t, 1) * pow(t, 2) * points[2])
+  (                    pow(t, 3) * points[3])
];

module plot2d(points, size) {
  for(point = points) {
    translate(concat(point, 0))
      circle(r = size);
  }
}
```

In the near future I think an arbitrary bezier function will be available. If someone puts up a bounty I may get it done even sooner.

`plot2d` is just a function to see where the points are.

stefano2734, 2021-06-28:
See Levien phd with better spiro splines with less errors in contours.
Optimized bezier splines are also available in this PhD.
This work is introduced in Inkscape.

https://levien.com/phd/phd.html

An optimization of curves or surfaces with FEM is also faster and more simple.

jordanbrown0, 2025-03-09:
BOSL2 has Bézier and NURBS support.
https://github.com/BelfrySCAD/BOSL2/wiki/beziers.scad
https://github.com/BelfrySCAD/BOSL2/wiki/nurbs.scad