openscad/openscad #3610: double free or corruption (!prev)

fermino, 2021-01-13:
Okay, so I managed to break openscad. I dind't find any guidelines to open an issue so if I'm missing anything please let me know!

I'm trying to make a napkin holder based on an STL model I found online, and when I try to render it, OpenScad crashes with a `double free or corruption (!prev)` error. 
I'm trying several formats but I guess it has to do with the render process and not the exporting. 
I'm using version 2019.05 and during the day I'll try to compile from master and see what happens. 

Command: `openscad -o test.stl test.scad`
The code: 
`
napkinHolder("SLC_TEMPLE_revised.stl", 5, 5, 50, 189.6, 118.0, 188.7, 0, 104.1, 0, 18);

module napkinHolder(model, bottom_t, wall_t, width, mx, my, mz, mxoff, myoff, mzoff, off, rot = false)
{
	clearance = 2;
	// TODO: Add chamfer
	union() {
		// The base
		translate([0, 0, bottom_t/2])
		cube([width, my+clearance, bottom_t], center=true);

		// Load the model in both sides
		translate([0, 0, bottom_t]) {
			disp = width/2 - wall_t-clearance;
			translate([-disp, 0, 0])
			napkinHolderSide(model, mx, my, mz, mxoff, myoff, mzoff, off, wall_t, false, rot);

			translate([disp, 0, 0])
			napkinHolderSide(model, mx, my, mz, mxoff, myoff, mzoff, off, wall_t, true, rot);
		}
	}
}

module napkinHolderSide(model, mx, my, mz, mxoff, myoff, mzoff, off, final_thickness, flip = false, rot = false)
{
	// To do: rotate => rotate the model 90 degrees on load
	rotate([0, 0, (flip ? 180 : 0)])
	// We'll scale the thickness of the model
	scale([final_thickness / off, 1, 1])
	difference() {
		// Center it and offset it by off
		translate([-off, -my/2, 0])
		// Move the model to 0, 0, 0
		translate([-mxoff, -myoff, -mzoff])
		import(model);

		// Take a slice
		translate([0, -my/2, -1])
		cube([mx-off, my, mz+2]);
	}
}`

Are there any debugging options I should try?

t-paul, 2021-01-13:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Reporting_bugs,_Requesting_features

I guess you covered most of it already. However it would be useful to know which OS/Distribution.

t-paul, 2021-01-13:
Oh, please add the call that triggers the issue. The code above only has the module definitions.

fermino, 2021-01-13:
Thanks!
I just updated the post, the module call was inlined with the code tag so it didn't display, I added the system info and the info dump too. 

I'll try this out using the windows build to see if the same thing happens. If so, I'll move on to compiling from master.

thehans, 2021-01-13:
I was able to reproduce the crash on latest build.  

As is often the case with CGAL related errors, the STL you are using appears to have some bad geometry in it.

I imported just the STL file by itself, previewed with F12 "Thrown Together" mode (this highlights backwards faces which is useful for diagnosing mesh problems).
Everything looked reasonable from the outside, but when I positioned the camera inside the model (only possible with perspective view), I see what looks like coincident back/front faces on the center towers: 
[image]

Regardless, we do try to avoid crashing even on invalid input, so I ran a debug build and got a backtrace:
```C++
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff4c3f859 in __GI_abort () at abort.c:79
#2  0x00007ffff4caa3ee in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4dd4285 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff4cb247c in malloc_printerr (str=str@entry=0x7ffff4dd6690 "double free or corruption (!prev)") at malloc.c:5347
#4  0x00007ffff4cb412c in _int_free (av=0x7fffdc000020, p=0x7fffb78a6b80, have_lock=<optimized out>) at malloc.c:4317
#5  0x00000000017cf650 in __gnu_cxx::new_allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > > >::deallocate(CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >*, unsigned long) (this=0x7fffe171b960, __p=0x7fffb78a6b90, __t=479)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/10/../../../../include/c++/10/ext/new_allocator.h:133
#6  0x00000000017cf628 in std::allocator_traits<std::allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > > > >::deallocate(std::allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > > >&, CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >*, unsigned long) (__a=..., __p=0x7fffb78a6b90, __n=479) at /usr/bin/../lib/gcc/x86_64-linux-gnu/10/../../../../include/c++/10/bits/alloc_traits.h:492
#7  0x00000000017cf2eb in std::_Vector_base<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >, std::allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > > > >::_M_deallocate(CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >*, unsigned long)
    (this=0x7fffe171b960, __p=0x7fffb78a6b90, __n=479) at /usr/bin/../lib/gcc/x86_64-linux-gnu/10/../../../../include/c++/10/bits/stl_vector.h:354
#8  0x00000000017cf94d in std::_Vector_base<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >, std::allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedg [... cut ...]
#36 0x00007ffff52135ae in QThread::started(QThread::QPrivateSignal) (this=this@entry=0x2fc6b50, _t1=...) at .moc/moc_qthread.cpp:162
#37 0x00007ffff52159ba in QThreadPrivate::start(void*) (arg=0x2fc6b50) at thread/qthread_unix.cpp:356
#38 0x00007ffff4c00609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#39 0x00007ffff4d3c293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

From my library info: `CGAL version, kernels: 5.0.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck`

MichaelAtOz, 2021-01-13:
The STL has self-intersections (red), and duplicate triangles (purple), also has 91 shells, so lots

of the bits may not be attached, although the other problems may cause that count to be wrong.

 

 

  _____  

 

 
<https://user-images.githubusercontent.com/252233/104512155-652eb980-55b3-11eb-854a-49e353ba1e61.pn
g> issue3610

<https://github.com/openscad/openscad/issues/3610#issuecomment-759756755>  it on GitHub, or
unsubscribe
<https://github.com/notifications/beacon/AAPBVYKJOI33NGDECYYHDVLSZYHTJA5CNFSM4WAXQWR2YY3PNVWWK3TUL5
2HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFVEPPUY.gif> 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2021-01-14:
Definitely something dodgy in my emails, it drops the images.
This was in my last post:
The STL has self-intersections (red), and duplicate triangles (purple)
[image]

fermino, 2021-01-15:
Thank you guys for the info! I'll see if there's any online fixer that works (already tried a couple and no luck there). I guess I could try to do it manually but quite honestly I would know where to start. 

Any hints?

Aside from that, should I close the issue or keep it open so maybe this error is somehow catched in a future release?

MichaelAtOz, 2021-01-15:
Leave this open, as thehans said, it shouldn't crash on such STL issues; 
when fixed it will still either generate an error message or bad export tho.

I've never had much luck fixing self-intersections manually, I don't have the
patience. I'd look for another model.

kintel, 2024-03-31:
Status as of 2024-03-24: Works with both CGAL and Manifold

closed 2024-03-31 by kintel