openscad/openscad #3613: Feature request: User-defined named list Indices

TLC123, 2021-01-14:
When working with object like data in lists it is very easy to end up with cryptic code.

I  here suggest feature to make clean code the natural outcome.
 
Currently there are three named indices with  .x .y .z
These are extremely clear and makes code almost self explaining 
a=[1,2,3];
echo(a.x)
:1

Globally index names doesn't scale well.
first=0; 
last=4;
a=[1,2,3,4,5];
echo ( a[ first ]  );
Quickly gets messy, out of scope, and intent likely collides at some point.

If such named indices where user defined PER list, code could be equally clear to read as .x .y .z.
Some ting like:
a=[10,2,5] named <red,green,blue>;
and then accessed as
echo(a.red);
:10

if named index is undefined return undefined. Used to identify user defined list type.

a=["point" ,1,2,5] named <point,x,y,z>;
b=["vertex",1,2,5] named <vertex,x,y,z>;
echo( is_undef(a.point) );
:false
echo( is_undef(b.point) );
:true

Some consideration needs to be taken and defined at concatenation and copying of lists.
a=["point" ,1,2,5] named <point,x,y,z>;
b=["vertex",1,2,5] named <vertex,x,y,z>;
c=concat (a,b);
If unique then its no trouble but if names overlap there needs to be a predictable resolution defined. (i.e use first index)

Partially named indices
list=[1,1,1,1,1,1,1];
a=["data",1,2,list] named <head,x,y>;

Named functions stored in lists
a=[function(x) x,function(x) x\*x,function(x) x\*x\*x] named< id, square, cube>;
echo ( a.square(10) );
:100

b=[ 1 , 2 , 3 ];
a=[ function(x) concat( x , x ) , function(x) x[0] , function(x) [x] ] named< first, mid, last>;
echo( a.first( b) );
:[ 1 , 2 , 3 , 1 , 2 , 3 ]

Alternativ implementation as a list local scope if such thing is possible.
a=["point",1,2,5,list] let( point=[0],x=[1],y=[2],z=[3], last=[4] );

Maybe a opportunity to define swizzling as named ranges. 
a=[1,2,3] let( x=[0],y=[1],z=[2], xyz=[0,1,2], zyx=[2,1,0]);
This part would rely on the following functionality

And finally maybe a too tangential feature to this last proposal.
But allow for ranges in index access.
a=[0,1,2,3,4];
echo( a[ 1 : 2 ] );
:[ 1 , 2 ]
echo( a[ 0 : 2 : 4 ] );
:[ 0 , 2 , 4 ]
somerange=[2:3];
echo( a.somerange );
:[ 2 , 3 ]

Anyway i think a feature, or a subset, with this or similar properties greatly could enhance readability of code that even become self explanatory an quit heavily rely on maintaining commenting.  I hope my description was clear.

t-paul, 2021-01-14:
I think the first part would be covered by the object literals proposal #3087 in a way that feels a bit more natural.

The second part may need a slight change as ranges are not a good fit. Using array slicing as shown in #3115 with slightly different interpretation (see [python array slicing](https://stackoverflow.com/questions/509211/understanding-slice-notation)) of the values give much more flexibility and expressivenes.

I'll set it to discussion for now, but I believe the features requested here are covered by those 2 other solutions.

TLC123, 2021-01-15:
Thank you. The object literals proposal is very fine. I failed to emphasize the difference. I will argue that the greater merit of named indexes is not captured by the object literals proposal. Namely by having overlap of listness and named access list and matrices can be composed in a natural mathematical manner and clearly read out by name.
 objects literals lack the power list have in being constructed/reconstructed/iterated over by list comprehension.
objects literals has not the same arithmetic power that vector and matrices carries. 
I will reconsider my proposal to see if it can fit as extension under the object literal proposals. Tank you again.

MichaelAtOz, 2021-01-15:
No offence, but I think the proposal is very convolved and not at all a natural extension of the OpenSCAD language.
Having it associated with assignment is not at all meaningful.
If you want the same for multiple usage you should not need to recreate it every time you assign.

Yes, there is a need for some pseudo-ENUM, but it needs to be much more accessible to the 'introductory' OpenSCAD user.

t-paul, 2021-12-23:
Added vector swizzling via #4010, I don't see custom / user defined extension on that. At some point objects will serve as general data structures.

closed 2021-12-23 by t-paul