openscad/openscad #3614: Release Candidate 2021.01-RC5 Hi-DPI Linux Text Issue

mholiv, 2021-01-15:
In Release Candidate 2021.01-RC5 the text appears blurry and looks bad overall on Linux with Hi-DPI displays. 

This is regression from OpenSCAD version 2019.05 where things look good.

My system:
```
X version number:    11.0
X.Org version: 1.20.10
Kernel: 5.4.85-1-MANJARO
Screen: dimensions: 7843x2160 pixels (2073x571 millimeters) (2 4K Displays)
Global Scale: 137.5%
```
[image]

rcolyer, 2021-01-15:
You have a lot more pixels to work with there than I do, but I currently don't see any difference at a pixel to pixel level between the fonts between the releases then and now.  Could you also add two text files with the contents of Help / Library Info for both versions on your system?  The top and the long sections at the bottom (e.g. font path) might be important.

t-paul, 2021-01-16:
I don't see that issue on my system (Dell XPS 13 with 4K built-in display 3840x2160 @ 200% scale + Full HD external display 1920x1080 @ 100% scale).

[image]

Maybe Qt/QScintilla don't support fractional scale?

mholiv, 2021-01-16:
@t-paul I tested it at 100% 200% and 125%. Yah it looks like non integer scaling is the root cause here.

The strange part is that in version 2019.05 things where find and looked good.

In both the release candidate and the 2019.05 build the non-text-editor UI scales really well. It's just the text editor that has the issues. This leads me to believe it is a QScintilla issue.

I found a thread about a similar issue here.
https://github.com/flathub/org.octave.Octave/issues/13

I guess the root question is, "Why can 2019.05 handle Hi-DPI Non integer scaling but RC5 can't?" Is is a pretty noticeable regression for me. :(

rcolyer, 2021-01-17:
@mholiv Are you by any chance able to do a compile and test of the PR above that contains a potential fix?  So far none of us are able to reproduce this issue, so we're not confident it actually fixes anything.  While I don't think disabling the buffering as done there causes any serious issues, I'd be hesitant to merge it without knowing that it actually fixes at least one problem.

mholiv, 2021-01-17:
@rcolyer Just compiled and can confirm that your PR works. Text looks great on HI-DPI displays using non-integer scaling. 

Thanks for helping with this.

Side note When compiling the dependencies building the build script I ran into an error while compiling glib2. I ran the dependency check script and it said everything was good so I moved forward. Just FYI.

t-paul, 2021-01-17:
Nice! I'm not seeing a difference on my system, so crossing fingers it's not causing issues on other platforms. I'll add the patch to the release.

t-paul, 2021-01-17:
Fixed via #3620.

closed 2021-01-17 by t-paul