openscad/openscad #3615: Release Candidate 2021.01-RC5 Save Diologue

mholiv, 2021-01-15:
In Release Candidate 2021.01-RC5 the save dialogue is non native and is annoying to use.

This is regression from OpenSCAD version 2019.05 where the native save  dialogue is used.

My System:
```
Operating System: Manjaro Linux
KDE Plasma Version: 5.20.4

KDE Frameworks Version: 5.77.0
Qt Version: 5.15.2
Kernel Version: 5.4.85-1-MANJARO
OS Type: 64-bit
Processors: 12 × AMD Ryzen 5 1600X Six-Core Processor
Memory: 31.4 GiB of RAM
Graphics Processor: Radeon RX 580 Series
```

[image]
[image]

t-paul, 2021-01-16:
What installation package is that? OpenSCAD does request the native dialog and that works fine for me on Debian/Testing running the nightly build debian package on Gnome. For example the Snap package might not be able to get the native dialog as it's isolated from the running OS.

mholiv, 2021-01-16:
@t-paul This is via the linux appimage

mholiv, 2021-01-17:
When I build from source it is good. This is an app image thing. Thanks. Closing.

t-paul, 2021-01-17:
Thanks for validating, lets keep the issue open for the packaging part. Maybe there's some library missing in the AppImage or something like that and this could be even fixed after the release by re-packaging.