openscad/openscad #3617: Customiser sliders seem to have broken

nophead, 2021-01-15:
[image]

```
X = 0; // [-1 : 0.1 : 1]
Y = 0; // [-1 : 0.1 : 1]
extruder = -1; //[-1 : 3]

/* [Hidden] */
```

The Y slider position in the picture corresponds to the minimum, which is -1, but the spinbox is -0.9 and that is value being used. Dragging it to 0.8 causes it to jump back to the 0.9 position with the spinbox showing 0.8.

Sometimes moving slider does not update the preview and sometimes it does. Sometime it matches the spinbox and sometimes it doesn't. 

Clicking on the track of the slider makes it jump and update the spinbox but doesn't update the preview. 

I am pretty sure it used to work until recently. Adjusting via the spinbox seems reliable.

OpenSCAD 2021.01-RC6 Win7 / 64bit

thehans, 2021-05-18:
@nophead I just created PR #3778 which should address the issues you list here.
I also changed the automatic preview behavior a bit (details in PR comment), and would appreciate any testing/feedback.  
I've only tested on Linux Mint 20, so it would help to verify on other platforms.

nophead, 2021-05-18:
Seems a bit broken to me, but so is a recent snapshot: OpenSCAD 2021.05.11.ci7927 I was using, I must just have not used the customiser with it until now. 

It now shows all the variables included from other scad files in the customiser whereas previous versions only showed variables defined in the main file. It also doesn't stop at my `/* [Hidden] */` line and shows too many variables in my main file. Even if I move my customiser definitions to the top of the file it still shows the includes after them.

It also seemed to create a new set of values called "New set 1" of its own accord and changed some array values to 0 breaking my code as they are constants that should never be 0. 

It also marked the file as modified but I didn't have the editor visible, so how can it be modified? Changing a customiser setting now seems to make it look the file is modified.

nophead, 2021-05-19:
Also $variables seem to be broken now. Setting them on the command line no longer seems to work.

redlizard, 2021-05-19:
As you say, these issues are part of an unrelated recent customizer development at #3755.

Showing variables that should not be parameters as part of the customizer is a bug, proposed fix above.

The "New set 1" system is a deliberate change introduced in #3755. The "<design default>" set is now immutable, allowing you to always reset things to what the source file says; attempting to modify the design default set will instead create a New Set.

Modification markers are also deliberate. Customizer sets are saved when you save the main file, and not before; so making changes there sets the modification marker.

Array values of zero sounds like a bug -- can you elaborate on that and make a separate issue?

thehans, 2021-05-19:
@nophead The vector issue might be something from my changes, I just noticed another issue related to vector parameters in my PR, so I'm looking into that now.
I haven't been able to reproduce specifically what you mentioned about them getting set to 0 though, so a test case for that would be helpful.

I also am not seeing hidden variables visible as parameters, but I have already applied changes from #3780 locally, so maybe that's why?
However @redlizard it looks like params from `[Hidden]` group are saved into the json file now, which is differs from behavior of 2021.01 release.

nophead, 2021-05-19:
I don't like the fact that changing a value forces a new set and marks the file as modified. It then prompts me to save it, which I never do because I use an external editor. I never save customiser changes anyway, as I only use it to pose assemblies. For example moving the axes of a 3D printer. 

When I close and open my file without saving the new set is still there and the file is shown as modified but I can't see a json file.

[image]
It fails to compile because the small_leg's 4 element list has the third element set to zero. I also notice parent_build_volume has its third element set to 0. I didn't intentionally change these. They seem to be a tiny subset of all my thousands of global constants. It doesn't show 99% of them because they aren't simple literals. The small leg comes from this file:

```
//
// Microswitches
//
small_leg  = [0.9,  3.3, 0.4, 0];
medium_leg = [0.5,  3.9, 3.2, 1.6, [0, -0.5]];
large_leg  = [11.4, 0.8, 6.3, 1.8, [1.7, 0]];
//                                                               t    w     l     r  h     h                               b     b    b             o     f     l                                                l           b         b
//                                                               h    i     e     a  o     o                               u     u    u             p     p     e                                                e           o         u
//                                                               i    d     n     d  l     l                               t     t    t                         g                                                g           d         t
//                                                               c    t     g     i  e     e                               t     t    t             t     m                                                                  y         t
//                                                               k    h     t     u                                        o     o    o             o     a     p                                                t                     o
//                                                               n          h     s  d     p                               n     n    n             l     x     o                                                y           c         n
//                                                               e                   i     o                                                                    s                                                p           l
//                                                               s                   a     s                               w     t    p                         n                                                e           r         c
//                                                               n                         n                                          o                         s                                                                      l
//                                                                                         s                                          s                                                                                                r
small_microswitch  = ["small_microswitch", "DM1-00P-110-3",      5.8, 6.5,  12.8, 0, 2,    [[-3.25, -1.65], [3.25, -1.65]], 2.9, 1.2, [-1.95, 3.75], 0.2, 0.55, [[-5.08, -4.95], [0,    -4.9],  [5.08, -4.9]  ], small_leg,  grey(20), "white"  ];
medium_microswitch = ["medium_microswitch","SS-01 or SS-5GL",    6.4, 10.2, 19.8, 1, 2.35, [[-4.8,  -2.6 ], [4.7,  -2.6 ]], 3.2, 2,   [-2.8,  5.8 ], 0.5, 1.00, [[-8.05, -7.05], [0.75, -7.05], [8.05, -7.05] ], medium_leg, grey(20), "burlywood" ];
large_microswitch  = ["large_microswitch", "Saia G3 low force", 10.4, 15.9, 28.0, 2, 3.1,  [[-11.1, -5.15], [11.2,  5.15]], 4,   2.75,[-9.1,  9.55], 0.3, 1.2,  [[19.7,   2.19], [19.7, -3.45], [8.3, -10.45] ], large_leg,  "ivory",  "white" ]; //G3M1T1PUL

microswitches = [small_microswitch, medium_microswitch, large_microswitch];

use <microswitch.scad>
```

thehans, 2021-05-19:
@nophead I had a bug in my changes where existence of vector parameter acted as if there was a change to values, which would force preview on load (even when "Automatic Reload and Preview" was OFF), and instantly created a "New set".

I just pushed the fix for that so please try a build from this latest revision.  

It will still create a New set and mark source as modified upon an *actual* value change.  I understand this is a concern given your workflow, but I'd like to keep that separate from this PR, as the behavior was not introduced here, and may require more discussion to reach a solution.  So a separate issue for that would be good.

There have also been a number of other fixes applied since my initial PR including two from redlizard, which I think should address the other issues you have brought up so far.

nophead, 2021-05-19:
Yes that seems to have fixed the regression and the sliders work a lot better.

I think it is definitely wrong to mark the file changed when it isn't. The star should be on the name in the drop down, i.e. `New set 1` as that is the thing that has changed and potentially needs saving. Even if I delete it with the minus button and go back to `<design default>` it still shows the file modified and prompts me to save it but there is nothing to save.

thehans, 2021-05-20:
Fixed by #3778
Please open a separate issue for marking of changed json vs source script.

closed 2021-05-20 by thehans