openscad/openscad #3619: Updates to OpenGL rendering

t-paul, 2021-01-17:
Follow-up on #1935 which implements VBO based OpenGL rendering. This is meant as list for collecting future improvement which can (and probably should) be handled via separate PRs.

- [ ] Automatically select renderer based on OpenGL version
- [ ] Add VAO, and enable if OpenGL supported
- [ ] Convert axis and other OpenGL code outside of renderers to use VAO/VBO
- [ ] Take action on OpenGL error instead of just logging it

thehans, 2021-01-19:
For the 4th item, I'm curious what actions were meant by @obliviand?

obliviand, 2021-01-31:
There are some errors that need to be handled and retry the operation, or return an error. They are just logged right now.