openscad/openscad #3621: Color ignored by linear_extrude

TheAndr0id, 2021-01-18:
Using colors is a nice way to check complex shapes and where they connect/intersect.  If doing this with a linear_extrude, the color command is ignored.

eg:

`color("Red", 1.0) linear_extrude(height=5){ ...`

will create a red shape.

`linear_extrude(height=5){ color("Red", 1.0) ...`

will generate a yellow shape.

thehans, 2021-01-18:
The problem is that linear_extrude creates a single object (with a single color), but may be composed of multiple child objects (with multiple colors), so choosing a color would be ambiguous.

```
linear_extrude(height=5) {
  color("red") square(5, center=true);
  color("green") square(5);
}
```

Maybe possible to support as a part of lazy-union feature #350 though.

TheAndr0id, 2021-01-20:
> The problem is that linear_extrude creates a single object (with a single color), but may be composed of multiple child objects (with multiple colors), so choosing a color would be ambiguous.
>
> ``` [...]

I can accept this.  For the sake of debugging it would be nice if linear_extrude retained the color attribute for the elements it is extruding, but I can totally understand this would be a low priority feature.  For now I just remove the extrude, figure out the alignment and then do the extrusion.

This should be updated on the wiki in any case.  I wasted a bunch of time trying to figure out what I was doing wrong with the color command.  (I'm still learning)

Oh, and I tested your example above non-extruded.  It Z-fights between red and green in the overlap. My expectations would be that the extruded part to do the same thing.  How would the linear_extrusion function do that?  Exception handling?  If multiple colors were defined at the same level, do nothing?  Crash?  Set random fires?  I'm maybe not the best person to decide that...  :)

If you want to keep this open as a enhancement request, that's fine, if you want to just close this issue I'm fine with that too.

TheAndr0id, 2022-08-22:
(Sorry for the close and reopen - Who the hell designed this interface?)

Just leaving a further comment for anyone searching for solutions.

While the color() command doesn't work, the debug tools (#, %) do, so you may do something like:

```
linear_extrude(height=10)
  union(){
    square([5,5]);
    #circle(d=7);
  }
```
Which will result in the 2D circle being highlighted in transparent red on the bottom of the resulting 3D shape.  Not exactly what I was looking for, but is still very useful for debugging.

kintel, 2024-07-20:
Closing as duplicate of #366

closed 2024-07-20 by kintel