openscad/openscad #3625: Dependency issues with AppImage

argentum2f, 2021-01-20:
I'm trying to run OpenSCAD-2019.05-x86_64.AppImage on RHEL7 and running into dependency problems.

Can more dependencies be included in the AppImage to make it less distro dependent? I thought that was the purpose in the first place.

In the meantime - can anyone recommend some workarounds to the following problems? The hard part is I can't make system changes (e.g. anything requiring root).

The first problem I had was a bunch of errors from (I guess?) an out of date `libstdc`, e.g. about 15 errors similar to:

    ./OpenSCAD-2019.05-x86_64.AppImage: /lib64/libstdc++.so.6: version `CXXABI_1.3.8' not found

I got past those errors based on a suggestion I found elsewhere using `export LD_LIBRARY_PATH=` to point to another version of `libstdc` I installed with anaconda (I can install core conda packages - but not conda-forge packages).

Now I'm stuck on on some `Qt_5` errors:

        $ export LD_LIBRARY_PATH=/home/***/anaconda3/envs/sfm/lib/:/home/***/anaconda3/lib/:/usr/lib64/:/usr/lib/
        $ ./OpenSCAD-2019.05-x86_64.AppImage 
        ./OpenSCAD-2019.05-x86_64.AppImage: /home/***/anaconda3/envs/sfm/lib/libQt5Core.so.5: version `Qt_5.11' not found (required by ./OpenSCAD-2019.05-x86_64.AppImage)
        ./OpenSCAD-2019.05-x86_64.AppImage: /home/***/anaconda3/envs/sfm/lib/libuuid.so.1: no version information available (required by /tmp/.mount_OpenSCqzwOLp/usr/bin/../lib/lib3MF.so.1)

thehans, 2021-01-21:
We are on the verge of a new release.  I don't know if any of these dependency issues would be solved by it, but could you try the latest release candidate appimage and see?  

Links are on the download page here:  http://www.openscad.org/downloads.html#rc
Direct link to RC6:  https://files.openscad.org/rc/OpenSCAD-2021.01-RC6-x86_64.AppImage

argentum2f, 2021-01-22:
Nope, same issues:

        $ ./OpenSCAD-2021.01-RC6-x86_64.AppImage 
        /tmp/.mount_OpenSCsnHTmR/AppRun.wrapped: /home/***/anaconda3/envs/sfm/lib/libQt5Core.so.5: version `Qt_5.14' not found (required by /tmp/.mount_OpenSCsnHTmR/AppRun.wrapped)
        /tmp/.mount_OpenSCsnHTmR/AppRun.wrapped: /home/***/anaconda3/envs/sfm/lib/libuuid.so.1: no version information available (required by /tmp/.mount_OpenSCsnHTmR/usr/bin/../lib/lib3MF.so.1)
        $ ./OpenSCAD-2019.05-x86_64.AppImage 
        ./OpenSCAD-2019.05-x86_64.AppImage: /home/***/anaconda3/envs/sfm/lib/libQt5Core.so.5: version `Qt_5.11' not found (required by ./OpenSCAD-2019.05-x86_64.AppImage)
        ./OpenSCAD-2019.05-x86_64.AppImage: /home/***/anaconda3/envs/sfm/lib/libuuid.so.1: no version information available (required by /tmp/.mount_OpenSCmXfuI1/usr/bin/../lib/lib3MF.so.1)