openscad/openscad #3627: Different Normalized CSG Tree Windows/Debian?

MichaelAtOz, 2021-01-22:
I started this worried about missing geometry:

I exported a complex design as .csg from 2019.05 -> 2019.csg
[image]

Loaded in 2021.01-RC6 it is missing some geometry.
[image]

Comparing the same .scad exported to .csg by 2021.01-RC6 (->2021.csg), it is missing empty group()'s.

e.g. 
[image]

I presume this is from the difference() & intersection() with 'empty' nodes, as discussed a few times recently. The .scad does have some
```
difference() {
  union() {
    add_something();
  ...
  }
  if () 
    remove_something();
}
```
Export .scad to .csg with RC6, load it into 2019.05, everything is there. 
So OK.

But I noticed something dodgy is happening.

1. 2019.05 .scad file on W7 - Normalized CSG tree has 601 elements. Base case.
2. 2019.05 2019.csg file on W7 - Normalized CSG tree has 601 elements. tick
3. 2019.05 2021.csg file on W7 - Normalized CSG tree has 601 elements. tick
4. 2021.01-RC6 2019.csg file on W7 - Normalized tree has 589 elements!. Blame the change mentioned?
5. 2021.01-RC6 2019.csg file on **Debian** - Normalized tree has 585 elements!. What?

6. 2021.01-RC6 2021.csg file on W7 - Normalized tree has 601 elements! tick.
7. 2021.01-RC6 2021.csg file on **Debian** - Normalized tree has 597 elements! Out by 4, same as 5. above.
8. 2021.01-RC6 .scad file on **Debian** - Normalized tree has 597 elements! What? Out by 4 v's 1. above.

Surely the same .scad & .csg file on RC6/W7 & RC6/Debian should have the same no. elements?

On **Debian** 2021.csg -> F5 -> Export to CSG, is identical to 2021.csg.

2019.csg & 2021.csg files: 
[For Dyson Cinetic - v7-03-dodgy.zip](https://github.com/openscad/openscad/files/5854049/For.Dyson.Cinetic.-.v7-03-dodgy.zip)

Aside - do we really need to change messages, ie missing 'CSG' & the added `!` on the end in RC. Makes it hard for matching with screen scrapers and accessibility.

thehans, 2021-01-22:
Builds since #3342 treat empty groups as empty geometry.  `difference { group();  ... }` means subtract "..." from nothing, which is nothing.  
Also, since #3555 `if (false)` will no longer generate an empty group.

So CSG export isn't necessarily transferable between versions because of those changes.

MichaelAtOz, 2021-01-22:
And W7 v's Debian having different Normalised element counts?

thehans, 2021-01-22:
OK, that does sound wrong.  Sorry, all that info about testing 2019 CSG export on 2021 RC was making it hard to see what problem you're reporting.

Can you do a diff between W7 RC6 vs Debian RC6 from "Design -> Display CSG Products..." ? (on CSG export originating from the same version only)  Specifically the "CSG after normalization:" section.   
I think that should have the corresponding number of terms/elements that are reported in the console log.  So a diff of those might pinpoint what exactly is happening.

MichaelAtOz, 2021-01-22:
There are 4 renders in the W7 CSG Products, missing in the Debian CSG Products.

[image]

Note there are render783 all over the place. It is probably text() differenced, ie a brand impressed into a plate.

Full data:
[CSG Products - 2021.csg RC6 W7.txt](https://github.com/openscad/openscad/files/5859170/CSG.Products.-.2021.csg.RC6.W7.txt)
[CSG Products - 2021.csg RC6 Debian.txt](https://github.com/openscad/openscad/files/5859171/CSG.Products.-.2021.csg.RC6.Debian.txt)

MichaelAtOz, 2021-01-23:
RC6 W7 v's Debian (Appimage) Library-info
[image]

Full info:
[RC6 W7 Library-info.txt](https://github.com/openscad/openscad/files/5859593/RC6.W7.Library-info.txt)
[RC6 Debian Library-info.txt](https://github.com/openscad/openscad/files/5859595/RC6.Debian.Library-info.txt)

I also just F6 rendered both the .csg & .scad, they both have the same console stats on the same platform,
but the two platforms have different stats, so the Normalized CSG difference makes it to the geometry.

W7
```
Rendering Polygon Mesh using CGAL...
Geometries in cache: 132
Geometry cache size in bytes: 5090640
CGAL Polyhedrons in cache: 87
CGAL cache size in bytes: 254776128
Total rendering time: 0:02:17.483
   Top level object is a 3D object:
   Simple:        yes
   Vertices:    12062
   Halfedges:   36438
   Edges:       18219
   Halffacets:  12420
   Facets:       6210
   Volumes:         2
Rendering finished.
```

Debian
```
Rendering Polygon Mesh using CGAL...
Geometries in cache: 132
Geometry cache size in bytes: 5480928
CGAL Polyhedrons in cache: 87
CGAL cache size in bytes: 266501824
Total rendering time: 0:02:04.069
   Top level object is a 3D object:
   Simple:        yes
   Vertices:    12856
   Halfedges:   38820
   Edges:       19410
   Halffacets:  13218
   Facets:       6609
   Volumes:         2
Rendering finished.
```
I'm about to check the exports.

MichaelAtOz, 2021-01-23:
Ignore the stats, W7 has a font, not on Debian, so different text shape.
But that does not explain the missing (or extra) render.
Just in case I will change both to Liberation and check the element counts again.

thehans, 2021-01-24:
Node 783 corresponds to the render containing the "MH" text, but I'm not sure yet why the CSG products aren't matching.

```
/*780*/     multmatrix([[1, 0, 0, 27.5], [0, 1, 0, -23.375], [0, 0, 1, -0.5], [0, 0, 0, 1]]) {
/*781*/      multmatrix([[1.1, 0, 0, 0], [0, 1.1, 0, 0], [0, 0, 1.1, 0], [0, 0, 0, 1]]) {
/*782*/       group() {
/*783*/        render(convexity = 13) {
/*784*/         linear_extrude(height = 10.1, center = false, convexity = 30, scale = [1, 1], $fn = 0, $fa = 0.05, $fs = 1) {
/*785*/          text(text = "MH", size = 10, spacing = 0.7, font = "Rockwell", direction = "ltr", language = "en", script = "Latn", halign = "center", valign = "baseline", $fn = 0, $fa = 0.05, $fs = 1);
                }
/*786*/         multmatrix([[1, 0, 0, -5.7], [0, 1, 0, 6.37], [0, 0, 1, 0], [0, 0, 0, 1]]) {
/*787*/          cube(size = [1.9, 1.9, 10.1], center = false);
                }
/*788*/         multmatrix([[1, 0, 0, 0.9], [0, 1, 0, 6.37], [0, 0, 1, 0], [0, 0, 0, 1]]) {
/*789*/          cube(size = [1.9, 1.9, 10.1], center = false);
                }
/*790*/         multmatrix([[1, 0, 0, -2.5], [0, 1, 0, 2.1], [0, 0, 1, 0], [0, 0, 0, 1]]) {
/*791*/          cube(size = [1.9, 1.9, 10.1], center = false);
                }
               }
              }
             }
            }
```

I get the same 597 elements as you on my Linux Mint box, which is unsurprising since its Debian based.  
The MH logo part looks ok preview, in that its properly subtracted from the part.  And its still part of some of the CSG products, just not *all* of the same ones.

My current theory, since you point out that the fonts aren't same on both systems, I would guess their bounding boxes end up slightly different.   There are optimizations performed in the CSG Tree based on bounding boxes of elements in the tree.  So if Debian's substitute font ends up a little smaller, it might not intersect as many nearby bounding boxes, and gets optimized away from those products.

Let me know if you can verify if CSG elements and rendered stats are same once they are using the same font.

MichaelAtOz, 2021-01-24:
Once Rockwell was installed the element counts match, 601.

MichaelAtOz, 2021-01-24:
@thehans Were those node numbers from `--debug`?

closed 2021-01-24 by MichaelAtOz

thehans, 2021-01-24:
No, it's a special build configuration that is rarely used, because it can interfere with cache.

It can be enabled with `qmake CONFIG+=idprefix ...`  or `cmake -DIDPREFIX=ON ...`