openscad/openscad #3628: 2021.01-RC6 - OpenCSG S/W rendering significantly slower

MichaelAtOz, 2021-01-22:
I use S/W rendering, accessing a W7 VM (on real W7 box via VMWare), from Debian via FreeRDP.
As without S/W rendering it only supports OpenGL V1 and preview is not good.

For simple models it works well, for complex models and higher resolution/preview-panel-size it is slow.

In 2019.05 in a 'normal' preview size a complex model took many seconds to render, as in paint, the preview, ie change front front view to top view, many seconds.
In 2021.01-RC6 it is unusable, it is as if it is continuing to repaint but not finishing, then repainting again, the GUI is very unresponsive, it gets the hourglass & 'Not Responding' repeatedly, typing a short-cut, eg Alt-F may sometimes work after many seconds, but often not.

To get a measure, I had to arrange a small preview size.
[image]

Note that this is not F5-preview time, that is 
```Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
```
The design as .csg is 
[For Dyson Cinetic - v7-03-dodgy-2021.zip](https://github.com/openscad/openscad/files/5854259/For.Dyson.Cinetic.-.v7-03-dodgy-2021.zip)

With 2019.05 after F5, it paints, clicking View-all or changing to other views takes ~4s to repaint.

With 2021.01-RC6 (`Total rendering time: 0:00:00.939`) after F5, it paints, View-all is slow

Here is a recording.
[clip0041.zip](https://github.com/openscad/openscad/files/5854275/clip0041.zip)
At 0:29 (18:29:09 on the clock at the top) you can see 2019.05 View-all takes ~4s.
With 2021.01-RC6, clicking View-all is at 1:12 (18:29:52), ... 4:59 (18:33:37) repaint finishes.
That is 10 frames/s video time & clock time seem out a bit.
So based on clock time, 4s v's 255s, ~64 times longer.
Time is proportional to preview window size, at a reasonable size it is ~14s, v's I had dinner and it didn't finish.

MichaelAtOz, 2021-02-01:
Could someone please have a look at this prior to release cut-off?

I think it is also repainting to often, such as associated with how it repaints when moving/resizing windows, it could benefit with deferred/skipped repaints if repaint time is a significant portion of a UI event-loop.

The S/W render I'm using (placed in program_files\openscad) is 
[opengl32.zip](https://github.com/openscad/openscad/files/5901772/opengl32.zip)

t-paul, 2021-02-01:
Release tag already happened, we are already way too close for comfort to make sure we have the new release in Debian/Ubuntu (next chance for Debian would likely be only in 2 years again). With things getting more automated it should be much easier to do a bugfix release soon-ish once we know how to fix.

t-paul, 2021-02-01:
I tried with newer version of Mesa OpenGL32.dll and it did get a bit more complicated, I want to update the wiki at some point, but here's what seems to be needed for that now:

Main lib: OPENGL32.DLL, needed are a number of dependencies from other packages
```
https://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-mesa-20.3.2-1-any.pkg.tar.zst

https://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-llvm-11.0.0-5-any.pkg.tar.zst
https://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-zlib-1.2.11-8-any.pkg.tar.zst
https://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-libwinpthread-git-9.0.0.6090.ad98746a-1-any.pkg.tar.zst
https://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-libffi-3.3-2-any.pkg.tar.zst
https://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-gcc-libs-10.2.0-6-any.pkg.tar.zst
```

List of libs placed in OpenSCAD folder to get it running:
```
opengl32.dll
libLLVM.dll
zlib1.dll
libglapi.dll
libstdc++-6.dll
libgcc_s_seh-1.dll
libwinpthread-1.dll
libffi-7.dll
```

MichaelAtOz, 2021-02-01:
> should be much easier to do a bugfix release soon-ish

OK.

This is a Windows S/W-rendering performance regression, 
ie OpenGL S/W* as driver on Windows presenting to FreeRDP client on Debian. 
\* GLEW version: 1.12.0 OpenGL Version: 3.1 Mesa 19.1.4
Note that IIRC it was not noticeably worse in snapshots until recent RC's. 
I tried a later version via [this](https://downloads.fdossena.com/Projects/Mesa3D/Builds/MesaForWindows-x64-20.1.8.7z), latest precompiled I could find, it was worse. 
Both practically unusable compared with 2019.05.

callahanp, 2021-09-29:
I'm wondering if the following code would help when examining this bug:
It places holes defined by an intersection in a flat panel.  As the number of holes increases, the preview's performance degrades until, at about 11 holes, it takes a long time to rotate the preview, with 100% CPU utilization.  Rendering works fine, and you can rotate the rendered object normally.  So this points to a redraw issue in Preview Mode as a possible cause.

Note that this is not a request to "fix" the code below, but a suggestion that it could be used as-is to explore rendering in preview issues.

```
// This module defines a panel for a row of circuit breakers.

module hole_punch(punch_thickness,punch_dia,flat_dia){
    //hole_punch creates a cylinder with two flat sides
    //work in progress 
    // - intersection cube is larger than it needs to be
    // - intersection cube should just fit.
    punch_height=punch_thickness*1.05;
    t=flat_dia+.04;
    intersection() {
    cylinder(h=punch_height, d=punch_dia, $fn=100);
    translate([-t,-t/2,-punch_height*.01]) 
        cube([t*2,t,punch_height*1.02]);
   }
}
module panel(breaker_count, 
             breaker_repeat, 
             breaker_dia, 
             breaker_flat_dia, 
             panel_height, 
             panel_width, 
             panel_thickness,
             behind_panel_depth) {
    color("blue") 
    {
        difference() {
            cube([ panel_width,  panel_height, panel_thickness]); 
            for (i=[0:breaker_count-1]){
            translate([0.75+i*breaker_repeat,panel_height/2,-panel_thickness*.02])
             hole_punch ( panel_thickness, breaker_dia, breaker_flat_dia );
            }
        }
    }
}
panel( breaker_count = 10, 
       breaker_repeat = 0.75, 
       breaker_dia =0.500, 
       breaker_flat_dia = 0.4190, 
       panel_height = 1.375+0.4, 
       panel_width = 12*0.75,
       panel_thickness = 0.2,
       behind_panel_depth= 1.5);
```