openscad/openscad #3631: Feature request: Position markers for later computations

jonnykl, 2021-01-25:
I want to to something like this:

```
translate([0, some_var, 0]) {
    rotate([90, 0, 0]) translate([0, 0, some_other_var]) {
        cylinder(h=cyl_height, d=cyl_dia, center=true);
        save_pos(cyl_center);  // save current position
    }
}

// use saved position
echo(cylinder_center=cyl_center);
translate([0, 0, cyl_center.z]) cube(5, center=true);
```

For example this could be very useful for calculating the dimensions of the whole object without having to manually calculate all the transformations (especially when they are nested more than 1-2 times).

Thanks!

thehans, 2021-01-25:
Sorry but I don't think this will ever be possible as it doesn't fit with the basic flow of the language.  

Expressions (eg variable lookup for `cyl_center`) are evaluated entirely before geometry statements (like transformations), so a simple variable can't know about those transformations.

The typical way to handle something like this would be to define a module, and use `children()` so that your "external" objects can be placed into particular positions/orientations relative to the module.

```OpenSCAD
module thing(cyl_dia, cyl_height, some_var, some_other_var) {
  translate([0, some_var, 0]) {
      rotate([90, 0, 0]) translate([0, 0, some_other_var]) {
          cylinder(h=cyl_height, d=cyl_dia, center=true,$fn=40);
          children(); // output children with current transformations
      }
  }
}

thing(2,10,5,3) cube(5, center=true);
```

closed 2021-01-25 by thehans

jonnykl, 2021-01-25:
Ok.
Could it be possible to add a special variable like `$vpt` but holding the value of the current position?

jordanbrown0, 2021-01-25:
The problem that Hans describes is real, but at least sort of solvable.  The transformation modules can pass transformation information down to their children.  I've demonstrated extracting absolute "world coordinate" values from inside deep transformation stacks, and reversing the transformation to allow a deep child to position something relative to the "real" origin and axes.

Another problem is that transformation isn't just translation and rotation.  There can be scaling and skewing in the picture too.  Trying to figure out what things like "current position" even mean can be tough.

But none of those are the really hard part.

Suppose you had a function or variable that would return the current origin and rotation relative to the "real" origin and axes, and you figured out what scaling and skewing mean to those results.  Whatever it means, congratulations, you've got them in this magic function or variable.  Now, what do you do with that value?  You have no way to get it out of the module it's in.  In the OpenSCAD language, it is *not possible* for a module to get information out, and that's a pretty fundamental aspect of the language.  You can't change a global variable, because there are no variables, no things that you can change.

MichaelAtOz, 2021-01-25:
> Expressions (eg variable lookup for cyl_center) are evaluated entirely **before** geometry statements (like transformations), so a simple variable can't know about those transformations.

$variables feed down the scope tree, not upwards to outside scopes.
They can go 'sideways'  to children, but if you use children, as Hans showed, you don't need this feature.

jonnykl, 2021-01-26:
Ok, thanks for the explanation!