openscad/openscad #3632: Feature Request: hull directly on vector of points

thehans, 2021-01-26:
In order to define an arbitrary convex polyhedron, a user currently has to specify all the faces, and get the winding orders correct.
If they know the result they want will be convex, it would be convenient to be able to simply provide the vector of points, and have OpenSCAD automatically apply convex hull.

I propose to add an optional parameter to hull so that these points can be passed in:
`hull(points);`

Alternatively, polyhedron could assume hull based construction when provided points and no faces:
`polyhderon(points);`

TLC123, 2021-01-26:
Good proposal .
I always do a work around.
I just make a monstrous invalid polygon and hull that mother.
As i know hull will throw it out without a glance..

`points=[for(i=[0:1000])rands(-10,10,3)];`
`hull(){`
`polyhedron( points , [[ each [ 0 : len(points) - 1 ] ] ] );`
`}`

[image]

t-paul, 2021-01-26:
I'm thinking if there's a way to make this a bit more general by extending what the 2D subsystem can do. There's already suggestions to add `line()` and #2796 introduces the idea of something like 2.5D objects (basically 2D shapes in 3D space). Combining those ideas could maybe produce an approach that looks more into composing existing things instead of adding more specific features to each of the primitives/modules.

Just sketching the idea, this may need some more thought...
```
hull() {
    point([0, 0, 10]);
    square(5, center = true);
}
```
```
hull() {
    square(10);
    translate([0, 0, 10]) polygon() {
        point([0, 0]);
        curve_to([10, 10], ...);
    }
}
```
The interesting part is that it would allow expanding things like SVG imports and Text inside of CSG filesa as 2D commands.

thehans, 2021-01-27:
I hadn't really considered having statements to create different dimensions of geometries like that.

It would definitely be neat to support all the possible combinations of nD geometries in 2D or 3D space for operations where they would be applicable.  It seems like our `Geometry::getDimension` maybe should return a pair of values (or possibly an enum as kintel suggested in the PR you linked)

So there could be 4 different dimensions of geometry:
 - 0D: points (or empty geometry i guess)
 - 1D: paths (i prefer the term "path" over "line", or maybe "polyline")
 - 2D: planar polygons (also maybe non-planar open [surfaces](https://en.wikipedia.org/wiki/Surface_(topology)) in 3D space)
 - 3D: polyhedra

And each of those could exist in either 2D or 3D space, except for polyhedra which can't exist in 2D space (unless you count projections).
So that's 7 total dimension-pair types: <0,2>, <0,3>, <1,2>, <1,3>, <2,2>, <2,3>, <3,3>

It might take some time to think through what operations would be potentially valid for each dimension pair, or even mixes of different dimension-pairs.

So, `hull()` could be fine with basically any of those, even mixing multiple types.  And if geometries in 2D space are mixed with those in 3D space then imply z=0.

Another example of mixing dimensionality in a useful way could be with minkowski.  I've sometimes wanted to be able to do something like:
```
minkowski() {
  square(100, center=true);
  sphere(d=10);
}
```
Which I feel should be perfectly valid, but currently it just gives a warning about mixed 2D/3D.

A couple other weird examples of userspace solutions to mixed dimension problems that I've developed over the years...
rotate_extrude of a 3D object:
```
module rotate_extrude3D(clearance, bounds=[1000,1000]) {
  rotate_extrude()
    intersection() { // clip to positive x axis before rotate_extrude
      translate([0,-bounds[1]/2]) square(bounds);
      offset(delta=clearance) // add clearance to profile
        union() for(a=[-90:$fa:90]) // union projections from multiple angles
          projection() rotate([-90,a]) children();
    }
}

difference() {
  translate([0,0,-10]) cube([100,100,100]);
  // assumes object is pointing along positive X axis before sweeping
  rotate_extrude3D(clearance=0.5, $fs=1, $fa=1) mything();
}

// View -> animate to show sweeping
%rotate($t*360) mything();

module mything() {
  union() {
    cube([50,20,2]);
    cube([50,2,10]);
  }
}
```

linear_extrude with twist of a 3D object:
```
$fs=0.25;
$fa=1;

d=2;
w=10;
h=6;
angle=45;

twist_slot(h,angle,48) 
  rotate([0,90,0]) cylinder(d=d,h=w,center=true);

module twist_slot(height,angle,steps) {
  union() for(i=[0:1:steps-1]) {
    hull() {
      translate([0,0,i/steps*height]) 
        rotate(i/steps*angle) 
          children();
      translate([0,0,(i+1)/steps*height]) 
        rotate((i+1)/steps*angle)
          children();
    }
  } 
}
```

navycuda, 2021-02-26:
> I'm thinking if there's a way to make this a bit more general by extending what the 2D subsystem can do. There's already suggestions to add `line()` and #2796 introduces the idea of something like [...]
...

I have been recently talking about the need for a proper sketching engine for openSCAD.  I think we need a proper way to produce parametric sketches within openSCAD.  I find importing .dxfs are unreliable and I lose the parametric aspect.

RonaldoCMP, 2021-09-25:
> So, `hull()` could be fine with basically any of those, even mixing multiple types. And if geometries in 2D space are mixed with those in 3D space then imply z=0.

I would take the other option, that is `hull()` of a mixing multiple types would have the greatest dimension. That would be consistent with the mathematical notion of convex hull.

thehans, 2021-09-25:
> I would take the other option, that is `hull()` of a mixing multiple types would have the greatest dimension. That would be consistent with the mathematical notion of convex hull.

I think we're on the same page, I just wasn't 100% clear in my explanation.  I meant the 2D geometries would be promoted to 3D, with *their* z=0.  And no change to already-3D geometries.

jordanbrown0, 2023-02-01:
See also
https://github.com/openscad/openscad/wiki/OEP7:-Mixed-Dimension-Geometry-Support

jordanbrown0, 2023-02-01:
See also PR #1168.

stefano2734, 2026-09-20:
Is this done with PR #1168 ?
CGAL 6.2.1 is also with hull functions.
https://doc.cgal.org/latest/Convex_hull_2/index.html
https://doc.cgal.org/latest/Convex_hull_3/index.html