openscad/openscad #3633: Crash on rendering a specific font (TeXGyreHeros)

DarkRuby25, 2021-01-26:
Ways to reproduce:
1. open the openSCAD example for logo + text
2. switch font to TeXGyreHeros
3. save and watch it crash immediately.

t-paul, 2021-01-26:
Please add details listed in this FAQ entry: [How do I report bugs?](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?)

t-paul, 2021-01-26:
Is that the Snap package?

No crash with 2021.01.24 nightly on Debian/Testing. Changing line 26 of the example "logo_and_text.scad" to
```
font = str("TeX Gyre Heros", style),
```
All 8 font variants installed from https://www.1001fonts.com/texgyreheros-font.html

```
OpenSCAD Version: 2021.01.24.nightly (git 9125c12)
System information: Linux 5.10.0-1-amd64 #1 SMP Debian 5.10.4-1 (2020-12-31) x86_64 Debian GNU/Linux bullseye/sid 8 CPUs 15.34 GB RAM

GLEW version: 2.1.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 20.3.3
GL Renderer: Mesa Intel(R) UHD Graphics 620 (KBL GT2)
GL Vendor: Intel
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)
```

DarkRuby25, 2021-01-26:
it was a snap package.
Not at at the office anymore so i tried with another laptop and it works correctly, no crash. Seems like the problem is with the installation of the fonts themselves (not sure what installed them), because when i installed them manually on the other laptop and copied to clipboard from the font library it copied with spacing between the words, while on the office computer when copying to the clipboard it copies with no spaces.

DarkRuby25, 2021-01-27:
Hey,
Working with the original computer with the problem, i have found another font that causes a crash: LMMonoLt10
The only connection between those fonts that they are all in the same path `/usr/share/fonts/X11/Type1` and with a file extension `.pfb`, but not all fonts from that path and extension cause a crash for example : `Courier 10 Pitch` doesn't crash.

lf94, 2021-04-19:
Potentially low hanging.

closed 2026-04-28 by github-actions[bot]