openscad/openscad #3638: Provide a simple measurement tool

west1012, 2021-01-30:
$500 to $1000 bounty depending on final capabilities of the tool.  Let me know how to set this up so you can verify the money is available and/or in some sort of escrow account.

I love OpenSCAD, but I do enough stuff where I need to I need to measure dimensions to make sure everything is OK that I can't really use OpenSCAD for what I want to do.

I am looking for a tool that does the following:
- Measure distances between two point
- Measure angles between three points (so essentially you create a triangle without the hypotenuse side and the angle between the two lines created by the three points is measured)
- Snap to edges, corners, and midpoints

I have not dug through OpenSCAD's source code to see what is possible, but if this is not infeasible due to the underlying structure of OpenSCAD, I would love to have this tool.

Let me know what you think, and/or if I'm publishing this feature request here correctly.  It has been almost ten years since I was a full time software engineer, and GitHub was in its infancy back then.  And, also, this is probably a dupe of this feature request, but I didn't want to bump an almost five year old feature request.

https://github.com/openscad/openscad/issues/1768

MichaelAtOz, 2021-01-30:
> $500 to $1000 bounty depending on final capabilities of the tool. Let me know how to set this up

Best to wait on the bounty until initial discussions on viability of the request.

TLC123, 2021-01-30:
@west1012  
Am i getting your proposal right. You are not suggesting any language extension but rather some graphical measuring tool to use in the view port on the final result?

west1012, 2021-01-30:
I am open to different approaches.  OpenSCAD seems like an incredible tool, but the fact that you can't (to my knowledge) measure distances and angles between different points to check your work changes OpenSCAD from being something very powerful, to something not very useful.  If it could be done easily in the language, and it was easy for a newbie user to use and comprehend what s/he was doing, (and preferably with some sort of graphics on the scene so you could visually verify you are measuring the things you think you are measuring) I would be OK with that.

For what it's worth, my goal here is sort of anti-selfish one.  I'm looking for something so all people (who can understand the OpenSCAD language) can use it, easily.  It would be nice to have a cheap/free alternative for hobbyists and students, and a measuring tool seems like the "missing piece" in OpenSCAD.  I'm sure there are others, which we can definitely talk about in regards to bounties going forward.  I am OK funding OpenSCAD work.

t-paul, 2021-01-30:
@west1012 The only way available to post bounties is via BountySource (link was just added to the issue by automation). With the other issue having a bounty already but a simpler focus, I suppose we will need to keep both open. If this one here is solved, it should cover #1768 too.

So my suggestions for such a big bounty would be to flesh out the description a bit more and then see how it goes. There's a reasonable chance it's going to happen, we had some discussions about how it could be implemented on IRC already and there's someone showing interest in taking on the topic.

@TLC123 has a good point, we should make it clear we are talking about GUI interaction. From our discussion about technical details, it sounds like we may need to restrict measurements to fully rendered mesh view (F6) - if it's possible for preview remains to be seen.

Do you maybe have some references to other tools which could serve as reference for how it may work from user perspective? I only know the Blender protractor which seems powerful but also quite complicated to use.

west1012, 2021-01-30:
I am happy to post $500 to Bountysource if it's confirmed this is possible and someone wants to go through with it.

west1012, 2021-01-30:
t-paul, my last comment was posted without me seeing your most recent comment, so apologies if it looks like I didn't reply to your statements properly.  We replied at roughly the same time, I didn't see them :)

GUI or via the language is fine, as long as it is easy-ish to use.  I trust that you guys aren't going to just implement something to get the bounty and call it a day, but will think about how to implement things so it's both not impossible to code, but also not impossible to use, if that makes sense.  So I really leave it up to you.  If it's via the language, I'm not sure snap to vertex or, say, midway point of a line is even possible?  Maybe it's not necessary if it's easy to refer to a part of a line via the language.

For my background: I'm an old C++ guy who went deep into the language when I used it (metaprogramming for the win!  Or at least it was fun).  And I have a degree in CS.  So feel free to discuss technical stuff with me / I'm not just some random joe on the internet who wants a feature, but can actually talk about technical specifics, if desired.  Mostly, because of busy-ness in my life, I would prefer you guys come up with an idea and implementation without my input thought :).  

I will go ahead and post a $250 bounty for now to show I'm really interested.  To my knowledge, I have no takebacks.  So either the money stays within the BountySource system, or it goes to you guys.  I will bump to $500 or more once we have more ideas on how we think it can be done.

west1012, 2021-01-30:
I have added the $250.  I don't know how long the details on the change take to populate to github:

https://www.bountysource.com/issues/96259516-provide-a-simple-measurement-tool

TLC123, 2021-01-30:
**The trouble with midpoints.**
Once rendered, basic overall metrics like volume surface area, bounding box etc is not to hard to find.
 
Selecting two or three or more vertices (either by clicking or just read of graphical labels in the view port) and enter into a simple measures dialog.  From - To or From - Via - To . From there just report all imaginable combinations of measurements. length, angle, normal, area, center point etc in a console like block.

Corners are essentially always represented by vertices.
 
**Midpoints of edges and faces is more troublesome.**
 
What looks like an edge visually  in geometry could be made up of several mesh edges.
However if  the the previous two point measurement **also reported coordinates of midpoint** and if the measure tool dialog  in addition to indexed vertices **also accepted any cartesian point** then it would be a matter of cut and paste from one  measure to the subsequent .
  
**In other words this proposed measuring tool could be utilized to construct  its own intermediate measuring points.** 
  
[image]
 
 [image]

__In a more ambitious path graphical rulers could be stretched out by mouse in the view port. Those in turn could have midpoints for click and construct measures._

Tomatower, 2021-01-30:
I have looked into this issue after implementing the mouse selection renderer and we already had a discussion on IRC.

The Mouse Selection currently requires for every Leaf Node of the document tree to have a unique 24-bit identifier, which can be encoded into the R/G/B color of the object during the rendering into a framebuffer. The clicked pixel can then be read from the framebuffer, and its R/G/B color is the 24 bit ID. Since all "fancy" things like lightnin, shadows, ... are disabled for this render pass, it is quite fast (not that openscad is any strain on modern graphics hardware - it is CPU bound) but more importantly it does not change the color of the object, and therefore its ID (see `MouseSelector::select` in https://github.com/openscad/openscad/blob/master/src/mouseselector.cc#L152) 
The ID assignment is done as a shader parameter, since there are few objects compared to vertices.

This approach can be extended by assigning a different color / ID to each vertex, edge and face, with a deterministic mapping to the location and associativity of each object.
A single triangle would need up to 7 identifiers: 1 face, 3 edges, 3 vertices (the vertices are shared, but might be duplicated for implementation reasons)

Reducing the number of triangles from 2^24 (== 16.77million) to 2^17 (==131k) can be limiting - but more modern graphics hardware capable of coping with these numbers should be able to implement `GL_RGB10_A2` with 2^30 bits of ID space (resulting in 2^23 (~8million) triangles (https://www.khronos.org/opengl/wiki/Image_Format - but I don't know about the portability!)

The hard question to be answered is therefore: how can we generate a deterministic mapping between vertices on the graphics card and their position in 3D space - the answer is the Vertex Buffer Object Index - prepared by obliviand but not yet merged  (https://github.com/openscad/openscad/pull/1935).

With the VBO in place one can create a Geometry shader to convert the VBO index (+ some offset) into a vertex ID, and use a similar method as the edge shader (https://github.com/openscad/openscad/blob/8a905133a2f27e23db07bb424599a242c5d7176d/src/GLView.cc#L207) to encode the ID into the edges and vertices.

To finally calculate the distances between all combinations of face, edge, point with given vertices is highschool math :)

t-paul, 2021-01-30:
The coloring logic would only be needed for picking in preview mode, right? I believe when having the full mesh available the traditional "intersect with ray from point of mouse click" should work?

TLC123, 2021-01-30:
Maybe you have already discussed this aspect in other places.
I imagine the ability to move the camera between selecting individually features is important if you need to measure between things that is not simultaneously visible in the same  view.

Tomatower, 2021-01-30:
The coloring logic is basically GPU assisted ray intersection - it is up to the test which of the two is easier to implement and faster.
If the selection is fast enough it might even be possible to highlight a hovered element in order to visually differentiate between edges, vertices and faces.

@TLC123 : The rendering pass is made at the time of the click, then the ID you clicked on can be remembered and highlighted, you can move the view around, rotate it and then click on the second point - triggering another "Render-To-Click" in a background buffer, just like in the mouse selection logic.
Afterwards you have two entities and can decide what you want to measure. With a logic in place to select vertices/edges/faces in 3D space the logic what you want to do with this information is "just" UI.

west1012, 2021-01-30:
Thanks guys for all the fast work on this.  For what it's worth, I was not hoping to get the function of "midpoint of a face" in my feature request, just "midpoint of an edge, or point between two points".  If the "midpoint of a face" is easy to do though, great!  Otherwise it is definitely not necessary as I could see how it could be a lot harder to do :)

Tomatower, 2021-01-30:
I was discussing a *UI Only* modification - so mostly a useful tool for *verification*, not so much for designing (at least not parametrically), since you would not be able to read these positions in code.

The general architecture of OpenSCAD does not support getting any information about geometry (be it boundingbox, barycenter, or even vertices) from the pipeline, and that would include the midpoint of an edge or face.

west1012, 2021-01-30:
I will be honest that I am a little lost on the graphics discussion here as I was always a systems / network guy, and never got into graphics.  So I apologize if I'm not understanding that discussion or the implications of it.  However, if the tool is possible, I understand that this would be a tool only for verification.  OpenSCAD is not FreeCAD / Fusion360, and I understand (or at least I believe) everything has to be designed / modeled through OpenSCAD's language, and not through any UI elements.

TLC123, 2021-01-30:
@Tomatower Even so, a measure tool provides a important **workflow feedback loop** if not programmatically.
Something like the heavily requested . Import an model, take some dimensions, add or subtract stuff.
take some more dimensions, add or subtract. etc

Going forward we could imagine measure tools with shortcut powers like "copy point as translate()" or "copy vector as rotate()" "copy face normal as rotate()" 
Essentially pushing transformation snippets back to code.

I really appreciate the engaged feedback  , Thank you really.

Tomatower, 2021-01-30:
I like your Idea! - something like: 

- Enter Measurement mode.
- Select outer edge face of my box
- select outer edge of my cylinder
- See "1.73 units (x: 1, y: 1, z: 1)" with buttons underneath: "copy as vector" and "insert as vector" and "insert as translate([])"
- inserts the text "translate([1,1,1])" at cursor position into the scad file.

I might take a look into this - but any help or ideas are welcome!

MichaelAtOz, 2021-01-31:
Netfabb (if you run Windows) has a good measurement interface.
[image]

At 'A.' options are select point on surface, point on edge, corner point, point on cut, corner on cut.
Cut is the sliders below, which moves a cut-line, e.g. the red lines, showing the outline surface at that section.
At 'B.' options are measure distance, angle, radius.
At 'C.' for distance, wall thickness, point to point, point to line*, line to point, line to line, surface to point.
\* - line is an edge or a cut-line.
At 'C2.' for angle, select three points, select two edges.
At 'C3.' for radius, select a curved edge, select three point on circle, four points on sphere.

Food for thought.

west1012, 2021-01-31:
Thanks for the help.  I know you can export to OBJ or STL and measure things in NetFabb or Blender (free as in speech), but that gets a little tiresome when checking and re-checking things regularly.  One of the *big* pluses to OpenSCAD, in my mind, is the ability to model things based on a few variables, and then be able to change those variables and have the entire model completely change.  To the best of my knowledge, there's nothing else that does this to the level OpenSCAD can, even the mega expensive professional stuff.  However, if I want to check measurements after changing a variable, having to re-export every time really is a pain, especially if you are playing around with changing the variable to few different values to see how they compare, if that makes sense.

MichaelAtOz, 2021-01-31:
Sorry, I was not clear, I was feeding ideas for UI concepts for an OpenSCAD measurement solution.

Tomatower, 2021-01-31:
That looks like they had the same idea but thought it through to the end :) - like they (AutoCAD) were earning money with CAD tools!

I am thinking about how to transport this Idea into the OpenSCAD interface, so continue to spin this idea :)

- **Points / Edges / Faces**: "trivial", they are already rendered somehow and it is obvious for the user what we mean.
- **Centerpoints of faces/edges**: OpenSCAD is (for the most part) triangles - selecting the center of a cube is therefore not selecting the center of a face (yes - cubes are using quads, but they are triangulated in the first operation!) - Is "center of a face" useful in that case, and if so - how would we (as users) use it?
- **Radius**: When traversing the tree it is possible to detect "circular" operations - `sphere`, `circle`, `rotate_extrude` - which bring a defined centerpoint and radius at a given vertex. Handling them specially by allowing radius measurements would be nice - but would make that feature unusable for user-defined spheroids (like in BOSL2). Another option would be to allow to select centerpoint/circle point for measurement (making that a vertex-vertex distance operation), or a three-point circle select. Should 
- **Cut Plane**: a cut line creates arbitrary vertices / edges in the model one can reference to. What would be the best way to design it from our perspective? Like in https://github.com/openscad/openscad/pull/1424?
- Would the above benefit from introducing things like **construction lines / planes**? (thereby "real" 2D planes) - or do we suggest using the `%`-modifier and `cube([100, 100, 0.0001], center=true)`? It would be nice to have a simple interface to convert some variables into construction lines.

TLC123, 2021-01-31:
Inspiring ideas y'all.
 
 > * **Centerpoints of faces/edges**: 
  Yes visually unified faces often is made of several arbitrary co-planar mesh faces. 
  Getting the intended feature by a single click isn't that straight forward with out extra logic.
  I suggest some responsibility is laid  on the user.
  i.e. Select 2 or more points or features and calculate their average position center of gravity or some other suitable center functions. Little more cumbersome but the benefit is clarity of user intent.
 
> * **Radius**: 
As above, but in circle measure mode, circles and arcs could be indicated by two points as center and radius  or by three points on the circumference, a sphere by four points on its surface, combinations of points and tangential edges etc. It can be messy but powerful. The basic points retrieved is a excellent starting point and could easily be extended by user-defined expressions in a little scad script.
circumscribed, inscribed. arc length etc probably goes here as an exercise for the student.  

> * **Cut Plane** 
Interactive cut planes  would be amazing but  a lot simpler, but more crude, would be to  just do plane cuts in user-space.
 I have  example modules for crop_by_Z() , crop_by_plane(), slice_by_plane() and construction plane on [github ](https://github.com/TLC123/OpenSCAD_stuff/blob/master/2021/cropByPlane.scad)(not library quality though).
 
>* **Would the above benefit from introducing things like** 
Construction primitive s like point() and line(). Are we talking infinite lines or line segments here?  
I have examples of helper module that act as construction_plane on [github](https://github.com/TLC123/OpenSCAD_stuff/blob/master/2021/ConstructionPlane.scad)(not fully constrained plane or library quality).
 
>*  **The Netfabb example**  show graphical dimensions in the view port.
 I fail to see how that kind of feature could auto update as expected in OpenSCAD.
 _Is there consensus that such feature lies outside  **a simple measurement tool**  #3638?_
 I think that thing fits better in a future blueprint /drawings export.
 
 I hesitate to pile on even more ideas. 
 An analog to a laser range finder would be nifty.
 Query with ray origin and direction and get first (or all) intersection with range, coordinates, feature ID and what-not .
 Bloat features aside. 
 
 What is the minimum viable solution to simple measuring tool? Tape measure and protractor?

west1012, 2021-01-31:
I’m not sure if the question is directed at me, or if I’m understanding it right.  However, yes, I’m basically looking for a “tape measure” and “protractor”/“angle measurer” for the bounty.  More than that would be awesome, but I’ve learned from my software days it’s best just to keep things simple at first and get them out quicker than to start a big project that never gets anywhere just because it’s so big :)

west1012, 2021-01-31:
Or maybe I should just say “Point in space” measurer?  Like you could click on a vertex (or possibly other arbitrary points, maybe), and know that it was a (X, Y, Z) location.  With that, I think you could figure just about everything else out, from lengths of edges, to degrees of angle between edges.

Tomatower, 2021-01-31:
A vertex in "graphics-speak" is a xyz-float position - so that was exactly my point above: using color-encoding to find the entity you have clicked on, and as soon as you have this information it is mostly highschool math (plus a good UI/UX!!) to display more context information.

So the hard parts for this feature are:

- Get a sexy UI/UX on how a measurement is done (see Michaels screenshot above as a nice solution for that)
- Retrieve the coordinates and entity type from the viewport upon a click (see the mouse selection magic as an example)
- If we use the 30-bit color selector: check if it supported on _all_ target hardware
- Set up a testing environment that can verify these kind of interactions

TLC123, 2021-02-01:
Even simpler and minimal. no dialog or form. 

Noting but select and highlight. 
Click some point edge or face (shift for multi select)
- it lights up, 
- info is printed on the console, 
- ctrl-c puts the vertex on clipboard.
- click outside clears highlight.
And from there its up to the user to do what ever, 
I would paste values into a new scad file and get to work on a a geometry-measurements library.

Tomatower, 2021-02-01:
Thats all UI/UX for me :)

I would split the mouse selection (to select source-level objects) and measurement (selecting vertices and edges) in source code.
When we want to measure we only nee graphics primitives, which can be retrieved from the VBOs, without association to the drawn object.

using `gl_VertexID` (see https://www.khronos.org/registry/OpenGL-Refpages/gl4/html/gl_VertexID.xhtml) sounds like it could work?

I do not yet have a working Idea on how to generate the colors for the edges and faces that is non-ambiguous, but having a look at `gl_PrimitiveID` (see https://www.khronos.org/registry/OpenGL-Refpages/gl4/html/gl_PrimitiveIDIn.xhtml) in a geometry shader might be worth a shot.

west1012, 2021-02-06:
Just checking in.  Did people thing this was feasible / yes on a go ahead?  I only put down either $200 or $250 on the bounty and a one month time horizon to show that I was serious, but wasn't sure if it was feasible, otherwise I would have put down the whole $500 or more.  I'm guessing I will have to renew that bounty when the month is up, as maybe things won't be done by then?  Thanks.

donc42, 2021-02-06:
I am planning to take this up. I am currently looking at some of the other products for interface ideas.

mjparme, 2021-02-09:
> I am planning to take this up. I am currently looking at some of the other products for interface ideas.

MeshLab has a measuring tool you can take a look at for interface ideas.

ab-tools, 2021-02-28:
Just wanted to add my thoughts here as well:
I'm still very new to OpenSCAD and I love it! That's the first 3D CAD program which I (as being a developer) immediately understood how it works and could start building things productively within half an hour.

The only thing - and that's why I ended up here - I desperately miss is a simply UI tool to measure and thus verify what I have designed in code.

For me even a very simply tool like the one from the free Windows 10 "3D Builder" app would do if just the "snap" feature would work a bit better (and also snaps e. g. to the center of a cylinder) and you could enforce/restrict to vertical/horizontal axis or plane to get a "straight line".

Just like @west1012 wrote not having such a simple measure tool considerable reduces the usefulness of this great program and forces you to often export meshes to use external tools to verify your work which is pretty cumbersome.

Therefore I'm really glad to read here that this is something perhaps being implemented soon! 👍

bmuessig, 2021-05-18:
I've added a bit to the bounty. I'd prefer the intersection method, not the hacky color code ID.

callahanp, 2021-09-29:
Have you considered that the wireframe view already provides numerous points and line segments that could be selected?
[image]

Sergei-Udris, 2022-04-13:
OpenSCAD and MatterControl have a number of bounties which means users are interested in CAD and 3d printing
- i'm interested in  taking on the challenge of improving CAD software and collecting these bounties
- however, to  and give user best program-on-the-left visualize-on-the-right experience one needs to step out of current stuck into more powerful tools - we need a new program who takes the best from OpenSCAD and others and adds to it significantly
- https://github.com/Sergei-Udris/Calican will be that CAD program, taking step forward
  - first and most importantly - clojure: Calican will use full unabridged unlimited most powerful language giving user not a script but full a lisp language with REPL to work with 
  - and - that will simplify program as work already done, Calican will just give user the keys and high level functions like OpenSCAD
  - he will run on JVM  - having access to whole ecosystem of libraries for 3d rendering and geometric computation, as of today choice is JOGL and libraries showcased here https://jogamp.org/jogl/www/ - right in the CAD territory
  - will be built as processes and queues using unmatched clojure/core.async - bringing golang routines power and multiplying it within an REPL-able lisp that runs on JVM with full access to all the goodies

today Calican exposes clojure REPL and just added a 3d canvas on the right with JOGL - first step made
- next - implementing OpenSCAD amazing examples list and being able to export in at least one format - so Calican will be a born as  CAD program
- then - addressing bounty issues and implementing features like measurement tool
- work is plenty but exciting and rewarding - as technology of clojure lisp and jvm does not fight against us but is for us

Calican will not be under my namespace nor will i create an organization for him 
- goal is for him to be transferred under an exiting github organization
- i'm interested in programming him as a bounty-project for an organization  - so that i don't waste time on setting up opencollectives and sponsors but am fully focused on features

t-paul, 2022-04-13:
@Sergei-Udris so what is it? Spamming about other projects or actually discussing the OpenSCAD feature proposed here?

kwikius, 2022-09-28:
Not expecting any bounty, but here is a simple manual measuring tool. 
https://github.com/kwikius/scad_measurement
See the readme for example useage how to measure angles and distances

west1012, 2022-09-28:
Thanks!  I will take a look.

On the bounty:  Just for what it's worth, the cash has been sitting there
for two years... and I don't have any rights to take it back, I don't
think...

bmuessig, 2022-09-28:
> Not expecting any bounty, but here is a simple manual measuring tool.

Thanks for putting in the work.
Defining the measurements from the code is interesting, but not perhaps what I had initially in mind (I thought it would be an interactive measurement tool). I can't make up my mind if I like to work in this way.

kwikius, 2022-09-28:
With this one, you need to move the thing to position and then manipulate it, but its useful for getting rough measurements and maybe could be the basis of something more sophisticated.

kwikius, 2022-09-28:
Keep the bounty open and perhaps it will jolt someone to come up with something better!

west1012, 2022-09-28:
Benedikt, my issue is my money has just been sitting there at BountySource
now for almost two years and nothing really has happened here.  I don't
even know if I lose rights to my money after a certain point.

I'm happy to keep the bounty open, but maybe with some sort of
contingency like "If no one comes up with something better by the end of
the year, and you remind me at the beginning of 2023, it's all yours" or
something.  I'd rather disperse the money to someone in our community who
did something, anything, than forget about it, and whoever runs
BountySource gets it for free.

kwikius, 2022-09-29:
@west1012  If you feel the widget solves the problem to some extent, I won't stand in the way of you giving me a bounty eventually :)  I could do some more on it but am currently working on getting another feature ready to PR. https://github.com/openscad/openscad/issues/4349  , and I better get that done first. By 2023 _maybe_ I will have a more polished solution. It might be possible to embed something in the IDE.  If it could be reduced to a 2d problem ( say by rotating model, doing projection and outputting to dxf with measurement points on a different layer perhaps...) that might make life simpler! To a large extent depends on what you are measuring...

kwikius, 2022-09-29:
@west1012  Might want to look here ! Not sure quite what status is now

https://github.com/openscad/openscad/issues/3359#issue-640436244

west1012, 2022-09-29:
Ouch.  So after two years they keep it.

So right now we are more like 1.8 years or so, if I remember right, and I
just rounded up to keep things easy.  I would have to double check though

bmuessig, 2022-09-29:
> @west1012 Might want to look here ! Not sure quite what status is now

A few posts down it says they have reverted that change: https://github.com/openscad/openscad/issues/3359#issuecomment-645685158

MichaelAtOz, 2022-09-29:
You know we do have Admin people here. 
Myself ( @MichaelAtOz )  & @t-paul being the front line, amongst others. 
We do have normal lives too though...we like it to run on auto most of the time.

Bounties are not going anywhere soon.

Please do not scare the children if you do not know what is going on.
@west1012 & @bmuessig (& @TLC123 if you're listening) should you wish to redirect your bounty there are options, I need to refresh my understanding, but IIRC should you want to contribute to the pool and have us redirect it to the community there is no loss, or I think it can be 'claimed' back to you but you'd lose the Bountysource cut. 

The trouble with Bounties is there is not a large pool of people with the right mix of skill sets, who are also motivated by 'comparatively' small financial gain. OpenSCAD developers are either concerned with core support, or things that peak their interests.

west1012, 2022-09-29:
My apologies, but my bounty is for the thing I asked for and I don’t have
the intent to redirect it anywhere.  I was even willing to up the bounty to
even more, possibly a lot more, as was described in the original bounty.

I figure “bounties” are to direct action towards a certain goal, whereas
“donations” are for the cause in general.

bmuessig, 2022-09-29:
It's indeed going slow and the amounts are not enough. I don't care about this as much anymore, as I did when I posted my bounties, as I have switched to Alibre Atom by now. So I'm happy to follow with what the backers with more bounty are willing to do.

west1012, 2022-09-29:
> I was not soliciting, but you expressed concern that your money was 'just
> sitting there'.
>
> You may not have seen the $1,200 bounty, which has attracted only a couple
> of attempts, but (with apologies) no solid commitment to a full resolution.
> I doubt you would attract a developer with more until one has the same
> frustration that you have. Trouble is many people eventually get over the
> need for a ruler.

I was thinking more, potentially significantly more, than $1,200 but I
thought I’d sort of put the “$250 to $1000” number out there to test the
waters.  Unfortunately it’s hard to say “Oh yeah, I’ll totally pay you $5k+
or maybe even five figures if you do this” without committing a decent
amount of money upfront to a bounty to show people I’m serious about it.
But then the money can just sit there.

Maybe bounties aren’t the right way to handle bigger potential payments
like that, but in my experience it’s hard to show up to a community out of
nowhere and say “Hey guys, I want to pay you bigger sums of money to get
this thing done.  Can we have a serious talk about it and commit to
something?”.  I think there’s a level of disbelief that a request like that
is real and thinking that “Oh, this guy is going to ask for the world for
that amount, and not just what he’s asking for, so it’s probably not worth
the hassle to engage”…

So I suppose that’s why bounties seem to be the best option?

west1012, 2022-09-29:
I also moved to Alibre.  I like OpenSCAD for its
programmability/parametric-ness, but clearly it’s better as tool where you
already have something designed and thought out in a different software
package, and then you want to add some flexibility / programmability around
that basic design.

MichaelAtOz, 2022-09-29:
Perhaps I was being too subtle. 
While you could have piqued an interest with your recent five figure reference, you never know, but I doubt it. 

Our experience is there a few people with the right experience for _in depth_ OpenSCAD development.
Bounties do not attract them much. They already have well paid intense jobs. 
Most do it as a side interest as OpenSCAD is useful to them, but generally they commit to things of direct interest.

Less detailed changes like UI tweaks can sometimes get someone interested for beer money.

Bountysource has not been much value to date.

west1012, 2022-10-11:
I was kind of wondering if BountySource would actually be useful when I
committed to it at that small sum, but that’s why I kept it a small sum.

You are definitely welcome to reach out to me about some sort of bigger job
and, if you are in the US, we can put it in some sort of contract or
something.  (Outside of the US, an informal contract?  It’s hard to have a
serious cross-border contract that’s enforceable.)  My old career was in
systems level software, so you would be talking to someone who actually
understands what’s realistic and what isn’t.

In my mind, OpenSCAD has always been soooo close to being the ultimate CAD
tool for power users.  But not having that measurement tool really killed
my interest in the crib, unfortunately.

Sheridan-Tech, 2022-10-22:
I would be interested in producing this for the bounty. From what I have read, you need to be able to place points somewhere on the model and display the distance between the points.

Paulwhy2, 2022-12-06:
I have been working on an a distance measurement function and have a implemented a rough solution that sorta works.  You can Shift-double-click at 2 points on the model and get an approximate distance in the Log. It works in the preview, before a render.

How it works: The existing QGLView::mouseDoubleClickEvent(QMouseEvent *event) function provides the screen x and y coordinates of any doubleclick event and then finds the z cordinate with glReadPixels().  These x, y,and z are in screen coordinates which are no good for measuring, but that same event function converts these to model coordinates with gluUnProject().  You can use these model coordinates to measure, but they are only approximate, +/- a pixel width I think.  If you zoom in and click, you get higher accuracy.

The existing DoubleClickEvent function then moves the center of rotation of the model to the doubleclicked location.  I added an if/else at this point: if the shift key is held down while doubleclicking, it calls a new function: void measureDist(GLdouble px, GLdouble py, GLdouble pz), else it continues with the existing code.

The new measureDist() saves the coordinates on the first shift-doubleclick and puts a crosshairs on the location, then if you shift-doubleclick again, it subtracts the old coordinates from the new, does the pythagorean calculation and prints the result to the log.

It still needs a lot of work, but at this point I'd like some advice on if I am on a good track, or if anyone else is working on this or a better solution, or if anyone wants to help with this development.  

Attached is the code.  I can do a pull request if that helps.
[openscad measurement draft 0.txt](https://github.com/openscad/openscad/files/10170056/openscad.measurement.draft.0.txt)

jordanbrown0, 2022-12-07:
Clever idea, but I'm not sure that it works in all cases.  OpenSCAD is too clever for its own good in how it (ab)uses OpenGL, and I suspect that the pixel depth is sometimes not what you think it is.

Test against a point that's behind a difference.

```
color("blue") translate([6,3,3]) cube(1, center=true);
color("blue") translate([6,3,-3]) cube(1, center=true);
color("red") difference() {
    cube(10, center=true);
    translate([11,0,0]) cube(10);
}
```

Put the camera along the X axis, with a distance of at least 30 or so, so that you see the two blue cubes in front of the red cube.  Then measure the distance between the corners of the red cube that are near those blue cubes.

If you get 10, you're a winner.  But I suspect that you will get about 18.

If you do, this expanded version will explain why:

```
color("blue") translate([6,3,3]) cube(1, center=true);
color("blue") translate([6,3,-3]) cube(1, center=true);
color("red") difference() {
    cube(10, center=true);
    translate([10,0,0]) cube(10);
}

translate([5,5,-5]) cube(1,center=true);
translate([20,5,5]) cube(1,center=true);
echo(norm([5,5,-5] - [20,5,5]));
```

It may matter whether you're in perspective or orthogonal mode.

jordanbrown0, 2022-12-07:
Oh, I should check:  were you working from an F5 preview or an F6 render?  The issue I was concerned about would only apply to an F5 render.

Alas, I do not fully understand how OpenSCAD uses OpenGL.  Unions are straightforward.  I have at times at least partially understood differences, and I thought they operated by giving the pixels behind the subtracted object the depth of the *front* of the subtracted object, but maybe I do not understand.  Intersections I do not understand at all.

Probably the best thing for you to do is to inspect the logic that implements the various operations, and understand what it's doing.  It's probably best if you do that, to make sure that your scheme always works.

I will be happy to be shown to be wrong... I just wouldn't want people to get all excited only to find that the scheme didn't work in all cases.

Paulwhy2, 2022-12-07:
I was in preview, but I just tested it again in render.  I got the same result,  glReadPixels() returns the correct 'z', not the subtracted cube 'z', and the calculation works.
I agree, though,  a full understanding of the logic behind these operations is needed to be sure this scheme will work.

ndbroadbent, 2023-08-13:
I was looking for a measuring tool as well. In case anyone finds this thread, I found this web-based tool called Meshy: https://0x00019913.github.io/meshy/

You can click "Measure" => "Length", and then click on the object to set points. It will show the length between the two points.

[image]

Very handy to make sure everything is correct before sending it to the 3D printer.

JaDogg, 2023-11-05:
For those who cannot wait I wonder if this what you are looking for? - https://www.thingiverse.com/thing:280469 I have not tried it out yet

SR-G, 2023-11-05:
Well, it seems (if i'm not wrong) that this is a library allowing to "manually enter measurements inside the .scad file, in order to display the entered values in the rendering".

So this is definitely not what we want : we want OpenSCAD to automatically display the computed values of the rendered object, as known by OpenSCAD after the rendering (at least its outbound box, but of course an easy measurement of any sub-part would be really nice), in order to easily validate that the script that we are writing is correct once rendered (yes, this is programmatic, we are theorically supposed to "know" the results of any measures ... but with complex objects, modules, curves, ... + due to the fact that OpenSCAD does not allow to reassign variables, it's really not always so easy, and an online / on-the-fly display of all measures would really help). And we need this feature : 
- Without having to export the .STL in another tool ...
- Without having to manually enter anything anywhere ...

However, the provided example attached to that library may somehow be a bit similar to what is needed (even if here it seems a bit to "2D" oriented).

I understand that, per the current internal architecture of OpenSCAD, it may not be easy to develop this feature, but it's really missing. Each time i'm using OpenSCAD, i rant about this feature being missing.

[12 more replies]

PR #4813 (2023-12-27): Adding interactive measurement abilities to OpenSCAD
This PR enables PR to measure distances and angle measurement features to OpenSCAD, which is a long awaited feature.
For distance you need to enter 2 points, For angle you enter 3 points. 
Selecting a Point in the model works like raytracing. for each eye ray, the closest point is selected.
Key function for this ability is gluUnproject which basically does the reverse of displaying Points

See https://www.youtube.com/watch?v=xunM6tBu-E0 for demonstration.

Cheers

Fixes #3638.
closed 2023-12-27 by t-paul