openscad/openscad #3640: Next Release

t-paul, 2021-01-31:
Checklist for next release (no timeline yet)

ToDo
* [ ] Update copyright year in all place visible within the app
* [ ] Update `src/AboutDialog.html` with changes to libraries, people etc.
* [ ] Update `doc/openscad.1` with new cmd-line arguments
* [ ] Update `openscad.appdata.xml`
* [ ] Update `RELEASE_NOTES.md`
* [ ] Cut a RC branch and build RC binaries
* [ ] Update translation files
* [ ] Update changelog
* [ ] Prepare news entry

MichaelAtOz, 2021-01-31:
Is it worth looking at older deprications and killing some with this release? 
`import_dxf()` & `import_stl()` perhaps & command line options `-s` & `-x`?

A search of the wiki also gets:
`assign()` - candidate.
`triangles=` in `polyhedron()` - prob. too soon?

fancsali, 2023-04-30:
Don't get me wrong, I fully appreciate this is a FOSS project – but is there any chance, this is happening soon?

OpenSCAD is rather awesome, and the dev snapshots seem to have superior performance. I'd argue, if nothing else, this could warrant a new release...

pca006132, 2023-05-13:
@fancsali it seems that #3781 is the main blocker.

pca006132, 2023-11-19:
#3781 can probably be fixed soon. Wonder if there are major blockers to a new release that are not addressed yet.

Ivo-tje, 2024-10-11:
Any change we will ever get a new release? I still see commits in the repo, but last release is from 2021...

kintel, 2024-10-12:
We're working on the major blockers, but still significant work to do. Once the known blockers are resolved, we'll start triaging issues and decide what's needed to cut a release. It takes time as we don't have any full-time developers, so for now the best thing you can do is to actively use the development snapshot and report issues, or if you have relevant skills, find something that needs doing and help :)

tdeck, 2025-03-05:
First I want to thank everybody who has been working on OpenSCAD. It's really a wonderful tool that opens up a lot of great possibilities. I'm building [a tool called 3DMake](https://github.com/tdeck/3dmake) for Blind people to do 3D printing and design based on OpenSCAD, and I'd love to have a stable release with the faster build process to ship to users.

Based on the discussion in #5615, it seems like the options considered so far are:

1. Finish up all the pending features and their tests on master
2. Create a branch based on the last stable build from 2019 and backport the fixes

I understand that option 2 seems like a huge amount of work, I used to work on a multi-branch project that was set up like this, and it resulted in a tremendous amount of toil.

Instead, have the developers considered disabling some of the new features (in the stable build) in order to get a release out. For example, #5516 seems to be about testing a new function called `textmetrics()` that isn't in the stable build. Same with #5519. #5520 adds new parameters to linear_extrude that weren't there before. These sound great and I'd love to have them, But nobody depending on the stable build is using these features, so they wouldn't miss them. And they'd get a bunch of performance and other improvements with a new release.

Does it seem feasible to add an IFDEF to disable some of the new language features that aren't thoroughly tested? Would you be willing to accept a PR that does this? Or are all these features vital to the next release.

It's possible that adopting a "feature flag" type strategy for these new features would help you get off of this treadmill where every time one WIP feature gets closed out there's been half a new (awesome) feature developed in the interim.

Since my mission is to make 3D modeling more accessible to Blind people, and I'm depending on OpenSCAD, I'm very motivated to help but I lack experience in the codebase. If you are interested in chatting about how I could contribute, let me know. I'm happy to work on what I can, in the time that I have.

t-paul, 2025-03-06:
> Does it seem feasible to add an IFDEF to disable some of the new language features that aren't thoroughly tested?

That exists already. Everything listed in the Features Preferences Tab or the `--enable` command line parameter is excluded from a release build.

https://github.com/openscad/openscad/blob/ce6161cc0b20bb517f556d17ff958bf010df0ccc/src/Feature.h#L15-L24

> Since my mission is to make 3D modeling more accessible to Blind people, and I'm depending on OpenSCAD, I'm very motivated to help but I lack experience in the codebase. If you are interested in [...]

That would be great. The most important input would be feedback on what works well and what does not. Ideally starting with some low hanging fruit as free time is not exactly plenty for everyone. Based on reading comments and posts on social media I'm trying to consider accessibility on GUI changes, but I lack experience in using the associated tools. So even just discussing the topic would be very valuable, if you want to dive into the code that's nice too of course. Does IRC work for you? The main discussion channel with people doing development is [`#openscad`](https://web.libera.chat/?channel=#openscad) on Libera.Chat.

tdeck, 2025-03-06:
Thanks for your quick reply; GitHub associated these issues somehow so i thought they were blockers for the release and was trying to understand why:

[image]

> That would be great. The most important input would be feedback on what works well and what does not.

Apologies if bringing this up muddied the issue, but I'm essentially building an alternative UI for OpenSCAD so for my particular purpose I am not blocked by any accessibility issues per. se. in OpenSCAD. However, I'm not comfortable bundling it with a dev / nightly release of software so I'm basing it on the latest OpenSCAD release. That's the biggest shortcoming for me - it's very easy with the stable OpenSCAD to make a relatively simple model that takes 20 GB of RAM and spends 30 minutes rendering, then gets OOM killed by the operating system. The rendering time is even more critical for this use case because these users don't rely on the visual preview; instead they render the model and have it described by AI or print it out. I know that this has been greatly improved in the latest builds.

Folks using 3DMake are "makers" but are not necessarily software engineers, so I want to minimize the number of bugs they need to report and work around, which is why I haven't released the software based on a development build. So the number one thing I want to contribute to and drive toward is releasing a stable build of the software. However, it's not obvious to me how I can do that most effectively or what the main blockers are. I'm happy to write documentation or fix bugs or whatever, although I lack a background in computer geometry.

kintel, 2025-03-06:
For reference, release planning happens here: https://github.com/orgs/openscad/projects/2
As we progress towards a release, we can certainly discuss the boundary between "Blockers" and "Wishlist".

tdeck, 2025-03-06:
Thanks folks for walking me through that, I'll take a look at the tasks and see where I can make a contribution.

eric-s-raymond, 2025-05-26:
Comment on "Update copyright year in all place visible within the app".

Not necessary. In fact, for ease of maintainence, you can and should remove all years from copyrights. They are not in general required for the copyright assertion to be valid, and especially not when the file it's in has metadata (such as a commit  stamp in a repository) establishing the date on which copyright was asserted.

This is now commonly known at the FAANGs, who are quietly dropping copyright year from their source files. The reason they're doing it quietly is that opening your mouth in public about the law always carries a small risk that you will be attacked by a litigious asshole. Thus, corporate legal departments hate taking a public position on a practice like this unless they have to to defend the company's interests.

Because of this, my friends at the FAANGs have asked me not to tell people what the authority is for saying it's OK to drop copyright years.  You'll have to take my word for it as a founder of the Open Source Inititative.

pca006132, 2025-05-26:
Interesting. This makes sense. Found a blog for curl talking about this: https://daniel.haxx.se/blog/2023/01/08/copyright-without-years/

jordanbrown0, 2025-05-27:
IANAL, but two data points:

- My Really Big Corporate Employer (Oracle) wants a copyright on every source file, with the year the file was created and the year the file was last updated.
- The handout that I got from my IP lawyer (revised 1991, post Berne) says "We recommend that all works continue to be marked with copyright notices, since there are still a number of advantages from doing so".  They don't list (at least in the page that I read) those advantages.  They mention omitting the year "in special cases" that do not seem to apply here.

and three opinions:

- As I read the [Code section](https://www.copyright.gov/title17/92chap4.html) cited in one of the comments on the curl blog post, a year is a mandatory part of a copyright notice.  A copyright notice without a year is not a copyright notice; it's just text.  However, since copyright notices are not mandatory, you can get rid of them entirely if you like.  (Related tidbit:  "©", "Copyright", and "Copr." mark a copyright notice.  "(c)" does not.)
- If there's a copyright notice with a year, it should be maintained.  Maintain it or get rid of it; don't let it get stale.
- Many people will (erroneously) assume that anything that doesn't have a copyright notice is public domain.  They're wrong, but giving them the hint seems like a good idea.  (But a "notice" without a year probably addresses that need.)

jordanbrown0, 2025-05-28:
[ Summarizing material that I have hidden. ]
@eric-s-raymond points out that some believe that copyright years are no longer required, backed up by (uncited) advice from the US copyright office.  I pointed out that the letter of the law requires them, but that the entire notice is no longer required so removing the year is mostly harmless; it yields informative text that is not formally a copyright notice, with little legal change in the status.  Net, we could probably remove the years, or the entire notice, without significant adverse effect.

coryrc, 2025-10-15:
Do any of the experimental features need to be made official and/or completely removed before cutting a stable release?

kintel, 2025-10-15:
The experimental features we hope to make "official" are mentioned in https://github.com/orgs/openscad/projects/2.
Any remaining experimental features simply don't make it into the release build as it's built with `-DEXPERIMENTAL=OFF`

t-paul, 2026-02-04:
Hopeful? See linked project (right sidebar).

hasselmm, 2026-03-08:
No release for more than five years. Maybe time for a hard feature freeze and to ship?

jordanbrown0, 2026-09-13:
Back in 2021, @MichaelAtOz asked about deprecations:
* import_dxf()
* import_stl()
* command line option -s
* command line option -x
* assign()
* polyhedron(triangles=...)

All of those appear to have been dropped.

kalj, 2026-11-07:
I'll take the man page item.

Alexb715, 2026-11-12:
Have any of these been completed recently but the issue not updated?

kintel, 2026-11-13:
Challenge is that they are moving targets, but now that we have a release candidate out, we can start ticking them off, and be careful to update as needed between now and release.