openscad/openscad #3644: Generated STL producing bad results when being sliced (via PrusaSlicer)

antfurn, 2021-02-06:
Hi, The generated STL from the new versions of OpenSCAD seems to be producing bad results when I slice it for 3D printing. I'm using PrusaSlicer 2.3.0-beta2 and all this is on Windows 10.

An older version [OpenSCAD version 2020.06.14.ci5408 (git be16bd85) ] produces usable STL output which I can slice using exactly the same settings (.3mf file also included).

Here's all the files:
[Broken STL output (2021.01).zip](https://github.com/openscad/openscad/files/5937521/Broken.STL.output.2021.01.zip)

I'm sure I've hit this on other recent OpenSCAD files I've created for 3D printing too, but don't have all the output from those to hand.

Regards, Antony

MichaelAtOz, 2021-02-06:
Which version?
It should have RCn, ie:
[image]

I exported it via 2019.05, and 2021.01.RC6 and get the same Self-Intersections as the two STLs in the ZIP.

2019.05
[image]

2021.02 - from ZIP
[image]

A different version is likely to have slightly different triangulations, so I think it is luck that one sliced and the other didn't.

I'll have a look at your .scad code later.

MichaelAtOz, 2021-02-07:
Fixed
[image]

OpenSCAD has some difficulties (bug) with two curves joining sometimes, PITA, it has had this problem for a long time, and it is trial and error to work-around.

I moved the negative drawing_pin_ears() bits up to the difference() in chrgrMount(). You needed that anyway to get rid of the flat bit here:
[image]

The changed section
```
module chrgrMount() {
  rot = 31.1;
  difference() {
    union() {
      translate([-mnt_w/2,-ph_case_L/2,-mnt_floor])
      cube([mnt_w,mnt_h,mnt_t]);
      translate([0,0,-mnt_floor])
      cylinder(r=mag_chgr_r+16,h=4);
      
      translate([-mnt_w/2,
          -mnt_lip_w-ph_case_L/2,
          -mnt_floor])
      cube([mnt_w,mnt_lip_w,mnt_lip_h+mnt_t]);
      
      //drawing pin mnts
      rotate([0,0,rot])
      translate([0,drwpn_offset+mag_chgr_r,-mnt_floor])
              drawing_pin_ears();
      rotate([0,0,-rot])
      translate([0,drwpn_offset+mag_chgr_r,-mnt_floor])
              drawing_pin_ears();
      
      //cable clips
      translate([-0.01+mnt_w/2,-mnt_lip_w-ph_case_L/2,,-mnt_floor])
      cable_clips();
      mirror([1,0,0])
      translate([-0.01+mnt_w/2,-mnt_lip_w-ph_case_L/2,,-mnt_floor])
      cable_clips();
    }
      // cut ear holes
      rotate([0,0,rot])
          translate([0,drwpn_offset+mag_chgr_r,-mnt_floor]) 
              drawing_pin_ear_holes();
      // cut ear holes
      rotate([0,0,-rot])
          translate([0,drwpn_offset+mag_chgr_r,-mnt_floor]) 
              drawing_pin_ear_holes();
    
    rotate([-mnt_angle,0,0])
    phnNchgr();
    // cable gap
    translate([-ph_lght_gap_w/2,
        -mnt_lip_w-ph_case_L/2-0.1,
         -mnt_floor-0.1])
    cube([ph_lght_gap_w,ph_case_L/2,mnt_t+mnt_lip_h+2]);
  }
}

module drawing_pin_ears() {
    cylinder(d=drwpn_d+3.2,h=6);
}

module drawing_pin_ear_holes() {
    translate([0,0,drwpn_bt])
    cylinder(d=drwpn_d,h=6);
    
    translate([0,0,drwpn_bt+drwpn_t])
    mirror([0,0,1])
    drawing_pin();
}
```

antfurn, 2021-02-07:
Thanks for looking at this issue and providing an explanation.

Is there anyway for OpenSCAD to highlight these sort of intersection 'problems' ?
Unless I'm missing something obvious, there is nothing giving the user any indication that there is even an problem, let alone where that is coming from\how to fix it. Do different slicers perhaps give better indications when that are problems?

I don't really want to add an extra step to my work flow by having to check all my STLs in netfabb, managed to avoid that so far for 8+ years :-)

In this model I was happy to have that flat area at the bottom of the ear. That seems like a perfectly reasonable design choice to make. But I know what to looks for - thanks.

FYI: I was using the very latest _non_-RCx version downloaded from here: https://files.openscad.org/OpenSCAD-2021.01-x86-64-Installer.exe

StephenRC, 2021-02-07:
PrusaSlicer 2.3.0 final has been released.   Upgrade to that and see
if it help.  It should, the betas have various slicing bugs.

On Sat, 06 Feb 2021 11:45:12 -0800, antfurn <notifications@github.com>
had a flock of green cheek conures squawk out:

> Hi, The generated STL from the new versions of OpenSCAD seems to be producing bad results when I slice it for 3D printing. I'm using PrusaSlicer 2.3.0-beta2 and all this is on Windows 10.
>
> An older [...]

antfurn, 2021-02-07:
One thing I did notice was that the 'working' STL had a significantly larger file size. I'm guessing but does that mean there were a lot more triangulations in it?

antfurn, 2021-02-07:
@StephenRC nope - same problem in the latest PrusaSlicer 2.3.0.
[image]

t-paul, 2021-02-07:
The file size difference comes from ascii vs. binary STL. `admesh` flags 11 normal vectors as invalid which probably is unrelated to the binary STL, but you could try changing to ASCII STL in preferences and see if that still produces issues.

antfurn, 2021-02-07:
Well thanks @t-paul , that fixes it. My original model exported as **ASCII** STL format from 2021.01 and in the latest PursaSlicer **works just fine**. 
[image]

t-paul, 2021-02-07:
Oh, interesting. So maybe we have a bug in the (relatively new) binary export. Could you please also attach that working ASCII STL for comparison?

antfurn, 2021-02-07:
Working ASCII format STL file - same original model, I just selected the _non-default_ option to export the STL in ASCII format in the preferences.

[Wall_mag_changer_ASCII.zip](https://github.com/openscad/openscad/files/5939262/Wall_mag_changer_ASCII.zip)

MichaelAtOz, 2021-02-07:
> Is there anyway for OpenSCAD to highlight these sort of intersection 'problems' ?
Unless I'm missing something obvious, there is nothing giving the user any indication that there is even an problem

Unfortunately that is the nature of bugs, if we knew how to detect it, it would probably be fixable.
I'm condensing your code down to a minimum to try to see where it breaks.

Also note that due to floating point issues, there is always the possibility of small anomalies getting through, this is rare. And coincident faces/edges/points will cause problems (non-manifolds), that is a mathematical uncertainty problem, not a OpenSCAD issue.

thehans, 2021-02-08:
Here is a minimized script which reproduces the problem.

If you use F5, then F6, a CGAL error will be generated.
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418
```

If F6 is done by itself then no error will occur.
So this has something to do with the conversions between `Polyset` and `CGAL_Nef_Polyhedron`, likely the grid snapping involved with that.

```
$fs=1;
$fa=2;
mag_chgr_d = 58.0 + 0.2;
mag_chgr_t = 5.8;
ph_case_L = 136.0;
ph_case_W = 68.0;
ph_case_T = 10.0;
mnt_h = ph_case_L - 30.0;
mnt_w = mag_chgr_d + 6;
mnt_t = mag_chgr_t + 7.0;
mnt_lip_w = 3.0;
mnt_floor = mag_chgr_t/2 + 3.0;
drwpn_d = 12.0;
drwpn_bt = 0.85;
clip_board_h = 20;
clip_board_w = 39;
clip_board_t = 2.5;

render() {
  union() {
    translate([-mnt_w/2,-ph_case_L/2,-mnt_floor]) 
      cube([mnt_w,mnt_h,mnt_t]);
    translate([-0.01+mnt_w/2,-mnt_lip_w-ph_case_L/2,-mnt_floor]) 
    difference() {
      cube([clip_board_w,clip_board_h,clip_board_t]);
      translate([10,clip_board_h/2,drwpn_bt]) cylinder(d=drwpn_d,h=6);
    }
  }
}

cube(); // extra cube to trigger implicit union on result.
```

A quick fix is to remove the `-0.01` from the 2nd translate.

MichaelAtOz, 2021-02-08:
That doesn't generate the Self-Intersection.
This does:
```
$fs=1;
$fa=2;

mag_chgr_d = 58.0 + 0.2;
mag_chgr_r = mag_chgr_d/2.0;
mag_chgr_t = 5.8;
ph_case_L = 136.0;
ph_case_T = 10.0;

mnt_h = ph_case_L - 30.0;
mnt_w = mag_chgr_d + 6;
mnt_t = mag_chgr_t + 7.0;
mnt_floor = mag_chgr_t/2 + 3.0;

drwpn_d = 12.0;
drwpn_pd = 1.5;
drwpn_h = 10.0;
drwpn_t = 1.7;
drwpn_bt = 0.85;
drwpn_offset = 22;

chrgrMount();

module chrgrMount() {
    union() {
      translate([-32.1,30,-6])
        cube([64.2,8,12]);
      translate([0,0,-mnt_floor])
        cylinder(r=mag_chgr_r+16,h=4);            
      //drawing pin mnts
      rot = 31.1;
      rotate([0,0,rot])
        translate([0,drwpn_offset+mag_chgr_r,-mnt_floor])
          difference() {
            cylinder(d=drwpn_d+3.2,h=6);
            translate([0,0,drwpn_bt])
              cylinder(d=drwpn_d,h=6);
            translate([0,0,drwpn_bt+drwpn_t])
              mirror([0,0,1]) {
                cylinder(d=drwpn_d,h=drwpn_t);
                cylinder(d=drwpn_pd,h=drwpn_h);
              }
          }    
    }
}
```

The SI appears to be a stray edge, here:
[image]

When deleting the two triangles it leave an edge at the SI, unlike the other two deleted triangles:
[image]

Note that Netfabb does not flag it as degenerate, so I don't know what it is meant to be part of.

I have seen this elsewhere with similar unions of differenced curves.

MichaelAtOz, 2021-02-08:
Note that my exports were all ASCII.

MichaelAtOz, 2021-02-10:
Actually I was 0.1 off in my mental arithmetic.

    union() {
      translate([-32.1,30,-5.9])  // <===
        cube([64.2,8,12]);

But both produce the SI. Interestingly on opposite sides:
[image]

As one SI join is with a cube side, the other with the cylinder side, to me it implicates the pin_mts contributes.

Without the large cylinder there is no SI, but the stray edge is still present.
A bare one has no stray edge:
[image]

I dismantled the bare one, there was no stray edge in it.

I'll keep reducing this.
[posting this interim report - further developments shortly, hint teeny tiny triangle]

MichaelAtOz, 2021-02-11:
That 'stray edge' is a sliver of a triangle.
[image]

At `A` to the right of the pointer is a tiny triangle, shaped like the red one in `5.`, shown in `3.`.
That garbled text says `v1:835 (-31.3339 38 -5.9)` & `f280: (662 835 660)`. 

It is:
```
  facet normal 0 0 -1
    outer loop
      vertex -30.9048 37.6381 -5.9    //v0 662
      vertex -31.3339 38 -5.9         //v1 835
      vertex -31.6647 38.2791 -5.9    //v2 660
    endloop
```

In `1.`  the top half of `v0-v2` (ie v835-v660 in `4.`) has to angle around the circumference, hence the sliver.
I suspect this triangulation 'style' is the cause of many SI's of unions of gradual curves.

Unless I'm missing something, a better triangulation could be split the triangle in `1.` at `A`.
Is there a way to influence such triangulation?

This is simplified code, small cube and the round bit:
```
$fs=1;
$fa=2;
mag_chgr_d = 58.0 + 0.2;
mag_chgr_r = mag_chgr_d/2.0;
mag_chgr_t = 5.8;
ph_case_L = 136.0;
ph_case_T = 10.0;

mnt_h = ph_case_L - 30.0;
mnt_w = mag_chgr_d + 6;
mnt_t = mag_chgr_t + 7.0;
mnt_floor = mag_chgr_t/2 + 3.0;

drwpn_d = 12.0;
drwpn_pd = 1.5;
drwpn_h = 10.0;
drwpn_t = 1.7;
drwpn_bt = 0.85;
drwpn_offset = 22;

//translate([31.345,-38,0])
chrgrMount();
echo(mnt_floor=mnt_floor);
module chrgrMount() {
    union() {
      translate([-32.1,30,-mnt_floor])
        cube([12,8,12]);
      *translate([-mnt_w/2,-ph_case_L/2,])
        cube([mnt_w,mnt_h,mnt_t]);
      *translate([0,0,-mnt_floor])
        cylinder(r=mag_chgr_r+16,h=4);
      //drawing pin mnts
      rot = 31.1;
      rotate([0,0,rot])
        translate([0,drwpn_offset+mag_chgr_r,-mnt_floor])
          difference() {
            cylinder(d=drwpn_d+3.2,h=6);
            translate([0,0,drwpn_bt])
              cylinder(d=drwpn_d,h=6);
            translate([0,0,drwpn_bt+drwpn_t])
              mirror([0,0,1]) {
                cylinder(d=drwpn_d,h=drwpn_t);
                cylinder(d=drwpn_pd,h=drwpn_h);
              }
          }    
    }
}
```
When exported as Binary shows as SI in Netfabb*, but not in ASCII. (*but not in MeshLab)
Meshlab displays the Binary vertices to 4 decimal places, which match the numbers in the ASCII STL,
but it is unclear if the display is just like echo() hiding smaller differences (ie -31.33388999 v's -31.3339).

So can we influence such triangulation?

MichaelAtOz, 2021-02-11:
This visualises that triangle

```
v0=[ -30.9048, 37.6381];    //v0 662
v1=[ -31.3339, 38];         //v1 835
v2=[ -31.6647, 38.2791];    //v2 660
polygon( [v0, v1, v2]);
$fn=12;
translate(v0) color("red") circle(0.002);
translate(v1) color("green") circle(0.002);
translate(v2) color("blue") circle(0.002);
```
[image]

kintel, 2024-03-31:
We've improved STL export quite a bit lately. Are anyone able to verify if this is still a problem with the latest development snapshot?

kintel, 2024-12-22:
Closing as fixed due to no further reports

closed 2024-12-22 by kintel