openscad/openscad #3650: svg: (option to) import only visible Layers

3Dietrich, 2021-02-08:
*.svg has layers which have 2 states: visible and invisible. If you save, of course all layers are saved. And OpenSCAD always imports and uses everything, even the invisible layers. This limits the use of svg quite a bit or you have to do weird workarounds (delete all invisible, save for OpenSCAD, undo..).

So my request would be: Wouldn't it make sense to import only the visible layers?

Possible counter question "why use invisible Layers if you only use the visible?".
A: for design I often use auxiliary objects, or sometimes you want to use only one part for OpenSCAD others for lasercut, or you have several versions of a part which are all relevant, or or or....

t-paul, 2021-02-08:
No, SVG has no layer. See #2383. The discussion goes a bit into the direction to maybe support some extra features provided by Inkscape, but there's no intention to support special cases for all/proprietary tools.

3Dietrich, 2021-02-08:
ah! I see.
I also was thinking it should be an Inkscape request..
allright! close..

closed 2021-02-08 by 3Dietrich