openscad/openscad #3651: linear_extrude() with twist bug in 2021.01

nyponen, 2021-02-08:
linear_extrude() with the twist parameter no longer works as it did in 2019.05. This breaks much of my code that uses linear_extrude(). 

`linear_extrude(20, slices=2, twist=60)
    circle(r=10, $fn=6);`

The left side shows how this code worked in 2019.05. The right side shows how it works in 2021.01.

[image]

Lib & Build Info:
OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 8 CPUs 39.87 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD2021.01
Documents Path: C:\Users\nypon\Documents
User Documents Path: C:\Users\nypon\Documents
Resource Path: C:/Program Files/OpenSCAD2021.01
User Library Path: C:/Users/nypon/Documents/OpenSCAD/libraries
User Config Path: C:\Users\nypon\AppData\Local/OpenSCAD
Backup Path: C:/Users/nypon/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/nypon/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD2021.01\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/nypon/.local/share/fonts
C:/Users/nypon/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.6.0 - Build 26.20.100.8142
GL Renderer: Intel(R) HD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_post_depth_coverage
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_robustness_isolation
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_polygon_offset_clamp
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_framebuffer_fetch
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_conservative_rasterization
GL_INTEL_fragment_shader_ordering
GL_INTEL_framebuffer_CMAA
GL_INTEL_map_texture
GL_INTEL_multi_rate_fragment_shader
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_texture_compression_astc_hdr
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

t-paul, 2021-02-08:
Yes, the tesselation is different via #3351, but why would that be a bug?

nyponen, 2021-02-08:
It is a bug because twist gives a result that looks wrong. I use twist to make something look twisted. A left twist should look different from a right twist. In some cases in 2021.01, it no longer looks twisted, and left twist gives the same result as right twist. For example,
`linear_extrude(20, slices=2, twist=60)
    circle(r=10, $fn=6);`

[image]

jordanbrown0, 2021-02-08:
While indeed the one on the left looks more twisty than the one on the right, an important factor is the "slices" parameter.  It's basically a resolution problem:  we're taking a hexagon, and rotating it 20 degrees for each of three steps.  There's more than one way to connect the resulting points, and it happens that it's now choosing a way that looks less twisty.  In particular, for the four-sided figure that results from connecting two points on one hexagon to the same two points on the next hexagon, the difference is that to create triangles the old way connects the lower-right point to the upper-left point, and the new way connects the lower-left point to the upper-right point.  That seems more right, because it's shorter.  (In fact, it seems like that's the key difference:  the old way selects the long diagonal, while the new way selects the short diagonal.)

Exactly what's happening is made pretty obvious if you set slices=1 and turn on edges, and then watch what happens to the edges as you set twist to various angles from 0 to 60.  Try this inside an animation:

```
linear_extrude(20, slices=1, twist=$t*60) circle(r=10, $fn=6);
```

It's hard to claim that either is "right" or "wrong".  A given point on the bottom layer of the old way is connected to the two adjoining points on the bottom layer, to its "matching" point on the layer above, and to the point that is receding from it on the layer above.  In the new way it is the same, except that the last connection is to the point on the layer above that is approaching it.

The objects produced are always an approximation to the mathematical objects.  The slices parameter says how precise the approximation is.  When you set it to a very low number, you are saying "not very precise", and that leaves a lot of latitude for getting different results.

thehans, 2021-02-08:
The reason for those changes is that it results in overall smoother objects with less possibilities for self-intersection at a given twist angle.  Those self intersections were the source of many bugs, so the change to behavior is a "necessary evil" IMO.

But if you prefer the aesthetics of the old behavior, then the good news is that you can still replicate it by rotating a full extra angle-between-sides, for every slice.  And with an added version check (based on the merge date of those changes), it should even produce the same results across releases:
```
slices = 3;
sides = 6;
angle = 60;
linear_extrude(20, slices=slices, twist=(version_num() < 20200821 ? angle : angle + sign(angle) * 360 / sides * slices) )
  circle(r=10, $fn=sides);
```
Note: this only works for regular polygons centered at the origin, as in the two examples you have shown.

closed 2021-02-08 by thehans