openscad/openscad #3657: 2021.02.11.nightly - editor syntax colouring turned off, cannot be enabled

cebderby, 2021-02-11:
Editor syntax colouring seems to be turned off/broken in the current nightly.
Visiting the editor preferences, I see a 'Color syntax highlighting' option with a droplist with "Off" the only choice,
 and having touched this I see a new? line item syntaxhighlight=Off in the OpenSCAD.conf file
Tried deleting this and restarting, also changing this to "On", or consistent with other options "true", and then, even more consistent with other option lines also tried syntaxHighlight=true, but the syntax colouring stays off when restarting with any of these.

t-paul, 2021-02-11:
Which build is that specifically? It's caused by #3652, should be fixed shortly...

cebderby, 2021-02-11:
it's: 2021.02.11nightly (git 18dfdee)

t-paul, 2021-02-11:
Sorry, I meant which distribution / installed from what source exactly?

cebderby, 2021-02-11:
Ah, I'm running Linux Mint 19, so the Debian/Ubuntu openscad-nightly package from
deb http://download.opensuse.org/repositories/home:/t-paul/xUbuntu_18.04/ ./

t-paul, 2021-02-11:
Ah, I see, it's trying to read the files from `/usr/share/openscad` instead of `/usr/share/openscad-nightly`

cebderby, 2021-02-11:
ok, that's absent for me, but making a temporary symbolic link so /usr/share/openscad points across to openscad-nightly immediately restores the syntax highlighting and gives the preferences dropdown about 9 choices.  Seems to be it.  So no problem for the main package, feel free to close this out when you're happy, thanks.

t-paul, 2021-02-11:
Thanks for the confirmation, I'll close once the path issue is fixed for the nightlies.

t-paul, 2021-02-12:
I believe the issue is now fixed for the OBS builds (and also the correct nightly icons should be shown again).

closed 2021-02-12 by t-paul

cebderby, 2021-02-12:
Can confirm the next nightly looked good, thanks