openscad/openscad #3663: Rendering bug in preview mode only - Z-buffer or blend mode?

WMays287, 2021-02-12:
I'm working on a small library for my robotics projects which adds rendering subroutines for simple metal hardware (screws, extrusions, etc.). While working on a T-nut, I noticed what I suspect to be a bug involving difference() linear_extrude() and polygon(): When looking at a preview (F5) some of the faces turn invisible, which doesn't happen with render (F6). ...Yes I know I have an outdated version of OpenSCAD, and this issue may have already been solved, but I'm still posting this just in case.
[image]
[image]

t-paul, 2021-02-12:
Looks like you need to specify `convexity` of the polyhedron (see [faq entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?)).

WMays287, 2021-02-12:
You're right, the default convexity is too low for shapes like this.

closed 2021-02-12 by WMays287