openscad/openscad #3664: Jump to source code from 3D preview make selected line more visible

gingerbeardman, 2021-02-12:
## Steps
1. Right click a model
2. Choose a shape
3. Look at source window

## Expected
- line should be centred in source window for maximum accessibility/visibility

## Actual
- selected line can appear anywhere in the source window, but commonly it is at top/bottom depending on whether the line was before/after where you were in the source before using the feature

## Screenshots

- target line was later in source file than current line before calling jump to source
  [image]

- target line was earlier in source file than current line before calling jump to source
  [image]

- target line on same screen of source as current line before calling jump to source
  [image]

Tomatower, 2021-02-12:
First of all - yay someone is using the feature :) 

Second, this "Move Cursor" codepath is shared with several other Cursor movement thingies. So I would fix that by making sure to have +- 3 lines in view.

So when you have 100 lines in view, and your search jumps *down* to line 200, you would see the window 103-203.
Jumping *up* would bring lines 197-297 into view. 
This is how VIM works in its default config, and IMO the least intrusive.

t-paul, 2021-02-12:
Yes, I agree it's probably better to have that kind of threshold so the code does not jump around if the target is already visible near the center. We'll have to check what qscintilla supports.

I would like to see an additional highlight too if we can find a way, depending on the color scheme it might be not too obvious where the cursor ended up. I don't know what's possible, some kind of graphical effect like flashing the line or the line number.

Tomatower, 2021-02-12:
QScintilla has 

> `void QsciScintilla::ensureLineVisible(int line)`
> Ensures that the line number line is visible. 

[Source](http://gnu-darwin.org/www001/ports-1.5a-CURRENT/devel/qscintilla2/work/QScintilla-1.73-gpl-2.1/doc/html-Qt4/classQsciScintilla.html#1d0ab66e8831b88f6a876642c896b6ef)

So some boundaries check and testing at the edge cases (going past the end/beginning) sound feasible.

gingerbeardman, 2021-02-13:
Thanks all!

This is a great update.

Tomatower, 2021-02-17:
See linked PR, this works except for the first and last line - I personally do not like overscrolling.

PR #3676 (2021-02-23): Make jumped-to lines more visible by sliding +-3 lines into view
see #3664 for motivation.

closes #3664
commit b568c4ea (2021-02-23): Make jumped-to lines more visible by sliding +-3 lines into view

closes #3664
closed 2021-02-23 by t-paul