openscad/openscad #3665: [CMake] Adding various options?

waebbl, 2021-02-13:
For source code based linux distributions like Gentoo, there are issues with automagic dependencies, i.e. optional dependencies, which are included in the build, if they are found at configuration time. If the user later decides to remove such dependencies, the application fails to execute properly.
In OpenSCAD there are optional dependencies on cairo, qtdbus and qtgamepad. I've added options to the cmake file for the build on Gentoo. If such is wanted, I can provide a pull request for those.
In addition, I've added options to request using ccache and building and running the tests. The first has caused various issues in the past, when used with the package manager, the latter is mostly used by devs, but not casual users. Making it optional avoids to pull in imagemagick and python.

Let me know, if you want these options added.

t-paul, 2021-02-14:
CMake ist not yet the official build system, but we are slowly moving now to make it the official and only one. So help with improving behavior would be very welcome. I would like to see small steps though, as there's quite a number of platforms we are building for.

What would be the behavior of those options? Would that mean every option needs to be flagged for inclusion? If that's the case we may need to find some way to maybe setup the default for the optional dependencies (like `-DMINIMAL_DEPENDENCIES=ON`?).

List of topics to check: https://github.com/openscad/openscad/issues/2518
Progress on moving the builds to cmake: https://github.com/openscad/openscad/issues/3652

waebbl, 2021-02-14:
> CMake ist not yet the official build system, but we are slowly moving now to make it the official and only one. So help with improving behavior would be very welcome. I would like to see small steps though, as there's quite a number of platforms we are building for.
> 
Thanks for your answer. I made a live ebuild to build, i.e. using the master branch, and tested against the various options I've added. Only tests don't run properly (~30 tests failed). But I didn't do any runtime testing (beside test framework) yet.

Only found one issue, when `-DHEADLESS=ON -DNULLGL=ON` are both set, the `export_png.cc` file fails to build, because of using some undeclared OpenGL related integral types (GLByte, GLUint, IIRC), but I fixed this for now, by not allowing to build with this configuration. I wanted to verify this first before I open an issue for it, they're not related to the added options, but to the already available ones. This is possibly related to the ticks on fixing `-DNULLGL=ON` and `-DHEADLESS=OFF` in issue #2518.

> What would be the behavior of those options? Would that mean every option needs to be flagged for inclusion? If that's the case we may need to find some way to maybe setup the default for the optional dependencies (like `-DMINIMAL_DEPENDENCIES=ON`?).
> 
The options I have added default to ON, so the logic isn't changed if the options are not passed to cmake. But they give the possibility to calculate a correct dependency tree for the package manager.
Is there a need for minimal optional dependencies? I've tried them with enabling each single one and several combinations of them and all build fine. So the minimal set of optional dependencies would be to set cairo, qtdbus and qtgamepad all to OFF. I'm not counting ccache and testing here, as those are build time related and do not impose a runtime dependency. I currently don't see a need for such a minimal dependency option.

t-paul, 2021-02-14:
As you said, with default to ON it would match the current behavior, so we are ok without any extra MINIMIZE style option. It would at least cover `lib3mf`, `hidapi` and `spnav` too - we can look at that later, if the request comes up.

waebbl, 2021-02-14:
The components `spnav` and `lib3mf` are currently considered required by me. I haven't seen `hidapi`, although I had this dependency in the 2019.05 package.
I'm gonna prepare a PR and we can details within it.

waebbl, 2021-02-14:
> Only found one issue, when `-DHEADLESS=ON -DNULLGL=ON` are both set, the `export_png.cc` file fails to build, because of using some undeclared OpenGL related integral types (GLByte, GLUint, IIRC), [...]
Just noticed these is already on the list in #2518 and has been reported before in #3667

t-paul, 2021-02-14:
Hidapi is not hugely useful on Linux (as it requires elevated access rights), so it's ok to leave it off.

t-paul, 2021-02-17:
I believe you covered everything mentioned here in your PR. If I missed something, we can just open a new issue or add to the backlog in #2518.

closed 2021-02-17 by t-paul

waebbl, 2021-02-17:
Thanks for merging