openscad/openscad #3686: OpenSCAD freezes on launch

GeorgeFlorian, 2021-02-23:
- OpenSCAD version: 2019.05
- OS: Linux Mint 19.3

Description:
Haven't launched OpenSCAD in a while and I did today but it freezes on launch.

[image]

I've reinstalled OpenSCAD to try and fix the situation but to no avail.

This is all I have. It simply won't start and I have no idea why.

thehans, 2021-02-24:
A few random ideas to try:
1) start openscad from a terminal, with no scad file loaded.
2) Try running it from a different window manager.
3) Move your config file: `~/.config/OpenSCAD/OpenSCAD.conf` to `[...].bak`  and try running again.
4) run `openscad --debug=all` from terminal and see what gets logged.

GeorgeFlorian, 2021-02-24:
I've tried starting OpenSCAD from terminal and I got tons of errors:

This is what I get:
```
qt5ct: using qt5ct plugin
qt5ct: custom style sheet is disabled
qt5ct: D-Bus global menu: no
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QXcbConnection: XCB error: 3 (BadWindow), sequence: 555, resource id: 10556741, major code: 40 (TranslateCoords), minor code: 0
composeAndFlush: makeCurrent() failed
```
with `composeAndFlush: makeCurrent() failed` repeating itself forever.

This is the `openscad --info` :
```
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2019.05
System information: Linux 5.4.0-65-generic #73~18.04.1-Ubuntu SMP Tue Jan 19 09:02:24 UTC 2021 x86_64 Linux Mint 19.3 6 CPUs 15.32 GB RAM
User Agent: OpenSCAD/2019.05 (Linux x86_64; Linux Mint 19.3)
Compiler: GCC "7.5.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_65_1
Eigen version: 3.3.4
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.5
QScintilla version: 2.10.2
InputDrivers: 
GLib version: 2.56.4
lodepng version: 20180910
libzip version: 1.1.2
fontconfig version: 2.12.6
freetype version: 2.8.1
harfbuzz version: 1.7.2
lib3mf version: (not enabled)
Application Path: /usr/bin
Documents Path: /home/george/.local/share
User Documents Path: /home/george/Documents
Resource Path: /usr/share/openscad
User Library Path: /home/george/.local/share/OpenSCAD/libraries
User Config Path: /home/george/.config/OpenSCAD
Backup Path: /home/george/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/george/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/george/.local/share/fonts
  /home/george/.fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/mathjax
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/abyssinica
  /usr/share/fonts/truetype/ancient-scripts
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/kacst
  /usr/share/fonts/truetype/kacst-one
  /usr/share/fonts/truetype/lao
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/liberation2
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/nanum
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/openoffice
  /usr/share/fonts/truetype/padauk
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/sinhala
  /usr/share/fonts/truetype/takao-gothic
  /usr/share/fonts/truetype/tibetan-machine
  /usr/share/fonts/truetype/tlwg
  /usr/share/fonts/truetype/ttf-khmeros-core
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/truetype/wqy
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/X11/encodings/large

X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  20
  Current serial number in output stream:  21
```

t-paul, 2021-02-24:
What's the output of `glxinfo -B`. Does `glxgears` work? (On Debian those are in package `mesa-utils`)

GeorgeFlorian, 2021-02-24:
> glxgears

It seems that `glxgears` does not work. The output of `glxinfo -B`  and `glxgears` is identical:
```
name of display: :0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  37
  Current serial number in output stream:  38
```

thehans, 2021-02-24:
OpenSCAD 2019.05 package itself should not have had any updates in recent months, and since not even glxgears is running for you, I would conclude that something on your graphics stack must have introduced some regression since you last (successfully) ran OpenSCAD.  

To narrow down which might be the culprit, I would have a careful look at your update history for packages relating to:
Kernel -> GPU Driver -> Xorg (or Wayland?) -> Window Manager -> Qt

Note that we recently had a similar report on IRC of being unable to start OpenSCAD.  The user was running i3 window manager, and it appears you may also be using some kind of tiling window manager (is it also i3?)
This is why I asked to try running it from a different window manager, but my request was ignored both times.

There is likely nothing that can be done from OpenSCAD's side of things to get this running again for you, and will need to be reported upstream to the responsible parties.

thehans, 2021-02-25:
@GeorgeFlorian For what it's worth, I just encountered this same issue (on Linux Mint 20, Cinnamon Desktop).  I had run software updater and a lot of graphics packages (nvidia drivers and related components) were updated.  
The update manager told me a reboot was required but I ignored it because I wasn't ready to close all my apps just yet.  

A little while later I tried opening a .scad file and i got the same thing, OpenSCAD window was frozen and not repainting, couldn't start glxgears either, with the same "BadValue" message.  So I went ahead and rebooted and everything is working again. 

It seems the classic IT mantra "Have you tried turning it off and on again?" still applies ;-)

GeorgeFlorian, 2021-02-26:
I reinstalled Linux (went from 19.3 to 20.1) and now it works. It most definitely was a driver error.

closed 2021-02-26 by GeorgeFlorian