openscad/openscad #3689: Transparency leaks through on wayland

waweic, 2021-02-24:
Hi, there's a thing I just noticed, I didn't investigate it further. I'm using Openscad in Sway and whenever there's a transparent object, I can see my wallpaper in the background. Is this expected behaviour? I do find this mildly irritating / funny.

[image]

t-paul, 2021-02-24:
No, that's not supposed to happen and it does not when using Gnome on Wayland, so I'm not sure this is an OpenSCAD issue.

nobodyinperson, 2021-06-19:
Also happens for me on Manjaro XFCE with X, not Wayland.

ednisley, 2021-08-04:
Still happens with Manjaro XFCE with X as of #bb54b97, although it took me a while to recognize the Veil Nebula image recolored from the desktop background:
[image]
[Hubble Veil Nebula](https://www.nasa.gov/image-feature/goddard/2021/hubble-revisits-the-veil-nebula)

sephthir, 2022-08-20:
Also happening on Fedora 35 + Wayland, installed via `dnf` (in order to support dark theming...), it's picking up whatever window is behind it (in this case, this thread). This can be more than distracting, as it can make it really hard to tell what's going on if the window behind it has a lot going on.

[image]

pipatron, 2023-12-04:
And definitely happening for me on Gnome+Wayland in debian bookworm in 2023.12.01.nightly (git 31b56c7)

flaviut, 2024-05-10:
Related issue in another app with a proposed solution from a wayland developer: https://github.com/swaywm/sway/issues/6549

See also: [FreeCAD](https://github.com/FreeCAD/FreeCAD/issues/6177), [OpenSCAD](https://github.com/openscad/openscad/issues/3689), [Cura](https://github.com/Ultimaker/Cura/issues/8466)

frysztak, 2024-12-15:
This issue is driving me insane, so I did some digging.

On Wayland, Qt defaults to RGBA8888 ([[1]](https://github.com/qt/qtwayland/blob/c0471e22a0319afa9f8404b908619192ff0de06d/src/hardwareintegration/client/wayland-egl/qwaylandeglwindow.cpp#L117)) surface format. One way to disable that is to set env variable `QT_WAYLAND_DISABLE_WINDOWDECORATION=1` ([[2]](https://github.com/qt/qtwayland/blob/c0471e22a0319afa9f8404b908619192ff0de06d/src/client/qwaylanddisplay.cpp#L925)). That's obviously not a good solution, especially on GNOME (because it needs CSD). It seems like there's a bug somewhere in Qt, which causes this weird ghosting effect when rendering something with alpha. Weirdest thing is, it doesn't show up on screen captures or even `apitrace` traces. Anyways, I reported this to Qt team: https://bugreports.qt.io/browse/QTBUG-132197. Hopefully they'll figure this out.

kintel, 2025-02-18:
@thehans Do you think this has any implications on #1485 ? Given the state of that PR, perhaps it's not really a concern right now :)

closed 2025-02-19 by kintel