openscad/openscad #3702: OpenSCAD-2021.01-x86_64.AppImage crash when open "saving as" windows

clmwith, 2021-03-02:
(AppRun.wrapped:60345): Gtk-WARNING **: 15:12:52.485: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Humanity/status/16/image-missing.svg: Impossible de charger le module de chargement d’images : /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so : /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Humanity/status/16/image-missing.svg: Impossible de charger le module de chargement d’images : /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so : /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Abandon (core dumped)

Into Ubuntu Ubuntu-MATE 20.04.1 64bits distribution

Thank's

closed 2021-03-02 by t-paul

clmwith, 2021-03-02:
sorry, thank you !

t-paul, 2021-03-02:
No problem, have a look at the linked issue, it has some workarounds. Maybe this could help in your case too.

clmwith, 2021-03-02:
fixed by using "custom..." theme button and choosing "Menta" icons rather than Ambiant-MATE