openscad/openscad #3703: Openscad fails to render scenes with imported STLs

sissi-luaty, 2021-03-02:
Hi! Openscad shows the preview of a scene correctly, and exports a *.png correctly from the command line. However, when clicking the "render" button or exporting a STL, the STLs included in the scene disappear from the output.
I have tried with various STLs, fixed meshes, but the problem persists.

Here is an example of the problem with the well-known 3D benchy model:

        color([0.5882352941176471, 0.23529411764705882, 0.7843137254901961], 0.10){
        %%cube(size = [350, 300, 190], center=false);
        }

        color([0, 0, 0.6], 0.20){
        translate([0, 0, -10])
        {
        cube(size = [350, 300, 10], center=false);
        }
        }
                translate([175.0, 150.0, 0])
        {
        rotate([0.0, 0.0, 0.0])
        {
        scale([1.0, 1.0, 1.0])
        {
        translate([-0.5000005939156722, -2.2649765014648438e-06, 0])
        {
        import("./3D_benchy.stl");
        }
        }
        }
        }
        
        $vpt = [197.278, 92.6229, 75.1442];          //coordinates of the center of the camera's viewport
        $vpr = [70.0, 0, 352.1];        //rotation of the camera
        $vpd = 940.0;                   //distance of the camera

This is similar to the bug reported in https://stackoverflow.com/questions/47697717/openscad-render-is-empty, without solution, but here I give details in a different context.

I have observed this problem both with openscad-2021.01-x86-64 portable for windows, and in openscad-2021.01 for linux (at kubuntu 18.04).

In the following figures there are screenshots with:
i) preview working fine;
ii) rendering not as expected.

[image]
[image]

I have attached the full contents with my *.SCAD file (i.e. the code above) and the 3d benchy STL (i.e. from https://www.thingiverse.com/thing:763622) that is included in the scene, in which this error is observed, in proj.zip: [proj.zip](https://github.com/openscad/openscad/files/6070458/proj.zip).

There is an error in the log when rendering:
- "CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!..."

sissi-luaty, 2021-03-02:
I have noticed that if the line **"cube(size = [350, 300, 10], center=false);**" is removed, then the output does not miss the STL model, but does not contain the full project with the platform as needed.
I would like to render the full scene with all the elements as shown in the preview image (the transparent one can be ignored), to obtain a STL file as output.

MichaelAtOz, 2021-03-02:
The .stl is bad, it has self-intersections (red lines) and degenerate faces (white):
[image]
[image]

sissi-luaty, 2021-03-12:
Thank you for the answer. Then, I have fixed this STL with the netfabb software (full version), but the problem persists.

MichaelAtOz, 2021-03-14:
@sissi-luaty care to share the 'fixed' STL.

CGAL only (99.99%) does `CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!...` when there is a problem with the STL, the 'assertion' is specific tests on the internal logic of geometry.

sissi-luaty, 2023-03-27:
This problem still appears: sometimes, a file *.STL can not be rendered when there are multiple objects.
For instance, for this minimal code, with PREVIEW (F5) works, but with RENDER (F6) doesn't work:
```
group(){
    cube(size = [70, 70, 10], center=false);
    sphere(r=10);
    import("./3D_benchy.stl");
}
```
The STL disappears from the rendered scene and that CGAL error is displayed in the log. How can this be fixed?

nophead, 2023-03-27:
You need to fix the STL file.

sissi-luaty, 2023-03-27:
I've tried with various different STLs, and I've fixed them with various methods, such as with admesh. The problem remains.

sissi-luaty, 2023-03-27:
In addition, if the var VARIABLE in the following code takes values such as -10, -9, -8, 3, 52.1... it renders (top screenshot); but for -1, 0, 1, 2 it doesn't (bottom screenshot). This bug seems to be related to openscad or GCAL (or their communication). If the problem were the STL the behaviour would be different.

[image]
[image]

```
translate([0, 0, -10])
{
cube(size = [87, 130, 10], center=false);
}
translate([0, <<VARIABLE>>, 0])
{
rotate([90.0, 0.0, 0.0])
{
scale([0.30000002980217694, 0.30000002980217694, 0.30000002980217694])
{
import("./3d_wizard.stl");
}
}
}

```
[3d_wizard.zip](https://github.com/openscad/openscad/files/11080673/3d_wizard.zip)

nophead, 2023-03-27:
CGAL only accepts perfect geometry. Most free STL fixers won't fix self
intersections. I haven't found a free one that does. So a lot of models on
thingiverse are useless for me because l use a  slicer that also doesn't
like self intersections.

violapaul, 2023-11-27:
Grrr.  Same here.  

- If I import an STL file it will view, and render back to an STL.
- If I add any other geometry, it will drop the STL.

If free STL fixers will not work, what tool might?

violapaul, 2023-11-27:
For those that are stuck.  Meshlab (free) fixed the problem.

I followed these steps (I am not sure which are critical).  The steps came from Bing Chat.

Meshlab is a free and open source software for editing and repairing 3D meshes. You can use it to fix common errors in STL files, such as holes, self-intersections, and non-manifold edges. Here are some steps to repair an STL file with Meshlab:

- Open Meshlab and import the STL file you want to fix.
- Select Filters > Cleaning and Repairing > Select Self Intersecting Faces. This will highlight the faces that overlap with each other. Press Delete to remove them.
- Select Filters > Cleaning and Repairing > Remove Duplicate Faces. This will delete any faces that are duplicated in the mesh.
- Select Filters > Cleaning and Repairing > Remove Duplicate Vertices. This will merge any vertices that are too close to each other.
- Select Filters > Cleaning and Repairing > Remove Faces from Non Manifold Edges. This will remove any faces that are connected to more than two other faces, creating a non-manifold edge.
- Select Filters > Remeshing, Simplification and Reconstruction > Close Holes. This will fill any holes in the mesh with new faces. You can adjust the parameters to control the size and shape of the holes to be closed.
- Select Filters > Normals, Curvatures and Orientation > Re-Orient All Faces Coherently. This will make sure that all the faces have the same orientation, either inward or outward.
- Select File > Export Mesh As and save the repaired STL file.

[You can also use other tools such as Netfabb](https://www.whiteclouds.com/blog/how-to-repair-mesh/)¹ or Blender² to repair STL files. For more information, you can check out these guides³⁴⁵. I hope this helps. 😊

Source: Conversation with Bing, [1](https://www.whiteclouds.com/blog/how-to-repair-mesh/)1/[2](https://formlabs.com/blog/best-stl-file-repair-software-tools/)7/202[3](https://robotter13.blogspot.com/2016/02/how-to-repair-stl-file-with-meshlab-and.html)
(1) Repair Mesh in Blender, NetFabb, and MeshLab - WhiteClouds. https://www.whiteclouds.com/blog/how-to-repair-mesh/.
(2) How to Repair STL Files for 3D Printing With the 5 Best (Free) STL .... https://formlabs.com/blog/best-stl-file-repair-software-tools/.
(3) How to repair STL File with Meshlab and Netfabb - Blogger. https://robotter13.blogspot.com/2016/02/how-to-repair-stl-file-with-meshlab-and.html.
([4](https://43dprint.org/how-to-edit-stl-files/)) How to Edit STL Files: The Basics & 8 Best Editors. https://43dprint.org/how-to-edit-stl-files/.
([5](https://howto3dprint.net/how-to-use-meshlab/)) All You Need to know about Meshlab | howto3Dprint.net. https://howto3dprint.net/how-to-use-meshlab/.

tomek-szczesny, 2023-11-28:
Still didn't fix my mesh because apparently it wasn't closed. I suspect it had more than one part that were not connected to each other.

Here's a workaround for those who need their 3D printed goodies before Christmas:
- In OpenSCAD, export whatever was rendered even without the missing STL import,
- Merge it with original STL using Meshlab (keyword: Boolean Union)
- Export the product of merger.

The resulting STL was accepted by 3D printing house, that's good enough for me.
I'm not saying this fixes the problem, but if someone's under pressure I recommend this approach.

ilka-schulz, 2023-12-10:
I have the same issue. @violapaul's solution did not work for me... I would be happy to add 100 EUR to the **bounty** pot but the bounty link seems to be broken...

tomek-szczesny, 2023-12-10:
I should add, the 3D model I fixed with the aforementioned method was accepted by the JLC automated system, but then I got complaints from customer service.
I ended up learning Blender and fixing the model manually for a few hours, to make things proper.

t-paul, 2023-12-10:
@ilka-schulz Status of bountysource is unknown at this point, there's rumors the parent company is bankrupt, but it's hard to find any information at all unfortunately.

T365, 2023-12-13:
I had that problem too, I fixed in Bambu studio, just open the file and right click "simplify model / fix model"

kintel, 2024-03-31:
Status as of 2024-03-24: Works with Manifold, still broken with CGAL

Miq1, 2024-12-13:
I had the same and found that increasing the renderer's memory in settings from 100MB/100MB to 500MB/500MB solved the issue for me.