openscad/openscad #3706: "Save as..." to existing filename prompts twice to overwrite.

thehans, 2021-03-04:
Running nightly dev snapshot on Mint 20

1) Create 2 files: `foo.scad`, `foo2.scad`
2) Start OpenSCAD and open `foo.scad`, click "File -> Save As..."
3) Select foo2.scad, a confirmation appears:
> **Save File -- OpenSCAD**
> foo2.scad already exists.
> Do you want to replace it?
> No / Yes

4) Click yes, file dialog closes, and 2nd confirmation appears (only the dialog title bar differs slightly)
> **foo.scad -- OpenSCAD**
> foo2.scad already exists.
> Do you want to replace it?
> No / Yes

jbinvnt, 2022-04-13:
The first overwrite confirmation dialog box appears to be created automatically by the `QFileDialog` when the user selects a filename that already exists. But the second box is created manually in the TabManager.

Here is [the function](https://github.com/openscad/openscad/commit/b0c95e89f7#diff-726fe757a1709bd8dba1eb410e9d1474096db960dd87ba1d359983e2507e2fe3R485) when it was first added. Note that in that commit, this second dialog box appears only when the new filename suffix is empty. The comment might be referring to [this issue](https://bugreports.qt.io/plugins/servlet/mobile#issue/QTBUG-32902).

But in the [current version](https://github.com/openscad/openscad/blob/e62afcf5d524638aaf5014fbd32ff8ddeac9e29d/src/gui/TabManager.cc#L702), the logic suggests that this second dialog box appears whenever the file already exists, regardless of whether the extension is empty. So as a remedy I first tried deleting the second dialog box entirely. However, after having done this, in the very specific set of inputs as shown in the screenshot, the user will get no confirmation that they are about to overwrite an existing file.
[image]

jbinvnt, 2022-04-17:
To elaborate on my previous comment, I believe that the fix will be to move the line that opens the second dialog back into the nested conditions that check for only these exact circumstances. Then the user will only get the second confirmation in the scenario where the first confirmation is not given.

PR #4235 (2022-04-22): Show manual overwrite confirmation only in the correct context
Closes #3706
closed 2022-04-22 by t-paul